tianocore/edk2 #10299: Integer Overflow in CreateHob() could lead to HOB OOB R/W (Bugzilla Bug 4166)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4166
Date: 2022-11-23T04:13:54+00:00
From: Marc Beatove &lt;<mbeatove>&gt;
To:   @niruiyu
CC:   ghouse.adoni.mohammed, @jkmathews, @kraxel, ling.yu, louis.shen, @miki-intel-work, rcorreadeysasi, rodrigo.gonzalez.del.cueto, w.sheng

Last updated: 2024-01-09T14:07:58+00:00

tianocore-issues, 2024-12-20:
## Comment 20081
Date: 2022-11-23 04:13:54 +0000
From: Marc Beatove &lt;<mbeatove>&gt;

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): @niruiyu

The CreateHob() function aligns the requested size to 8 performing the following operation:
```
HobLength = (UINT16)((HobLength + 0x7) &amp; (~0x7));
```

No checks are performed to ensure this value doesn&apos;t overflow, and could lead to CreateHob() returning a smaller HOB than requested, which could lead to OOB HOB accesses.

tianocore-issues, 2024-12-20:
## Comment 20150
Date: 2022-11-30 10:37:03 +0000
From: @miki-intel-work

Assigning to Ray Ni

tianocore-issues, 2024-12-20:
## Comment 20250
Date: 2022-12-21 04:06:56 +0000
From: Marc Beatove &lt;<mbeatove>&gt;

My team has asked me that I share with you our disclosure timeline. It will take effect from today:

We are privately disclosing this vulnerability to 
you so that you can develop a fix and manage its rollout. We do not require 
you to keep any information of this report secret, but if you make it 
public then please let us know that you did. This advisory will be kept 
private by Google for 30 days after a fix is publicly available or after 90 
days if no fix is made. After this deadline we plan to disclose this 
advisory in full at: http://github.com/google/security-research/. Please 
read more details about this policy here: https://g.co/appsecurity

tianocore-issues, 2024-12-20:
## Comment 20919
Date: 2023-01-25 11:46:27 +0000
From: Marc Beatove &lt;<mbeatove>&gt;

Any updates?

tianocore-issues, 2024-12-20:
## Comment 21016
Date: 2023-02-09 21:36:27 +0000
From: @miki-intel-work

Confirmed this is a code issue

tianocore-issues, 2024-12-20:
## Comment 21023
Date: 2023-02-10 11:40:45 +0000
From: Marc Beatove &lt;<mbeatove>&gt;

Created attachment 1323
Suggested patch for bug

Attaching a suggested fix for this issue to try to speed up the process.

Attachment: [BZ-1323-file_4166.txt](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-1323-file_4166.txt)

tianocore-issues, 2024-12-20:
## Comment 21039
Date: 2023-02-13 12:26:32 +0000
From: @kraxel

Patch looks good to me.

Rai?

Given that HOBs are only allocated in early firmware phases (SEC+PEI), before the firmware starts processing potentially untrusted input from network or storage devices in DXE phase I think it is highly unlikely this problem is exploitable.

So, can we just commit the fix (ideally before the 2023-02 stable tag),
without going through embargo/disclosure dance?  We don&apos;t have much time
left anyway, we are already roughly half through the 90day period ...

tianocore-issues, 2024-12-20:
## Comment 21048
Date: 2023-02-13 18:00:55 +0000
From: @miki-intel-work

CVE-2022-36765 Reserved

tianocore-issues, 2024-12-20:
## Comment 21071
Date: 2023-02-15 07:46:57 +0000
From: @kraxel

Note: PeiCreateHob() in MdeModulePkg/Core/Pei/Hob/Hob.c already as a simliar check.

pierregondois, 2025-07-30:
V1: https://edk2.groups.io/g/devel/topic/103657270#msg113576
V2: https://edk2.groups.io/g/devel/topic/103658961#msg113607
V3: https://edk2.groups.io/g/devel/topic/103675959#msg114038

and merged at:
https://github.com/tianocore/edk2/commit/aeaee8944f0eaacbf4cdf39279785b9ba4836bb6