tianocore/edk2 #10365: EfiUnacceptedMemoryType can be allocated by gBS->AllocatePool (Bugzilla Bug 4315)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4315
Date: 2023-01-18T02:14:02+00:00
From: @lgao4
To:   @mxu9
CC:   @mxu9

Last updated: 2023-01-18T20:04:55+00:00

tianocore-issues, 2024-12-20:
## Comment 20715
Date: 2023-01-18 02:14:44 +0000
From: @lgao4

min.m.xu@intel.com: can you confirm whether this is an issue?

tianocore-issues, 2024-12-20:
## Comment 20714
Date: 2023-01-18 02:14:02 +0000
From: @lgao4

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): @mxu9

EfiUnacceptedMemoryType memory can&apos;t be allocated by gBS-&gt;AllocatePages. It should not be allocated by gBS-&gt;AllocatePool.

tianocore-issues, 2024-12-20:
## Comment 20716
Date: 2023-01-18 02:19:43 +0000
From: @mxu9

Yes, this is an issue. I will submit a patch to fix it.

tianocore-issues, 2024-12-20:
## Comment 20723
Date: 2023-01-18 20:04:55 +0000
From: @lgao4

Min works on it.

pierregondois, 2026-10-07:
Fixed by: https://github.com/tianocore/edk2/commit/96192ba5bd4f

closed 2026-10-07 by pierregondois