tianocore/edk2 #10374: Running windows 2k22 on OvmfXen got ASSERT /root/source-git/edk2/MdeModulePkg/Core/Dxe/Event/Tpl.c(66): ((BOOLEAN)(0==1)) (Bugzilla Bug 4340)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4340
Date: 2023-02-09T07:54:26+00:00
From: Joey Lee &lt;<jlee>&gt;
To:   @tperard
CC:   @tperard, jlee

Last updated: 2023-02-09T20:24:42+00:00

tianocore-issues, 2024-12-20:
## Comment 21009
Date: 2023-02-09 07:56:20 +0000
From: Joey Lee &lt;<jlee>&gt;

After bisecting, found this patch causes problem:

commit 71cdb91f313380152d7bf38cfeebe76f5b2d39ac
Author: Anthony PERARD &lt;anthony.perard@citrix.com&gt;
Date:   Mon Apr 12 14:30:03 2021 +0100

    OvmfPkg/OvmfXen: Set PcdFSBClock
    
    Update gEfiMdePkgTokenSpaceGuid.PcdFSBClock so it can have the correct
    value when SecPeiDxeTimerLibCpu start to use it for the APIC timer. 
...

I did not see Tpl.c(66) ASSERT after reverted 71cdb91f3 patch on edk2-stable202202 or edk2-stable202208. It confirmed that the patch causes problem.

tianocore-issues, 2024-12-20:
## Comment 21010
Date: 2023-02-09 07:57:07 +0000
From: Joey Lee &lt;<jlee>&gt;

After debugging, I found that the following patch can workaround issue:

Index: edk2/MdePkg/Library/SecPeiDxeTimerLibCpu/X86TimerLib.c
===================================================================
--- edk2.orig/MdePkg/Library/SecPeiDxeTimerLibCpu/X86TimerLib.c
+++ edk2/MdePkg/Library/SecPeiDxeTimerLibCpu/X86TimerLib.c
@@ -91,7 +91,7 @@ InternalX86GetTimerFrequency (
   )
 {
   return
-    PcdGet32 (PcdFSBClock) /^M
+    100000000 /^M
     mTimerLibLocalApicDivisor[MmioBitFieldRead32 (ApicBase + APIC_TDCR, 0, 3)];
 }

But this is not a right solution. Currently I have no idea with using PcdGet32 to get the value of PcdFSBClock causes problem.
I have printed the value of PcdFSBClock in the above function. But it shows 100000000 which the same with the fixed value.

And, just add a redundant PcdGet32 (PcdFSBClock) statement in the above patch before return statement, it can trigger issue again. Like this:

Index: edk2/MdePkg/Library/SecPeiDxeTimerLibCpu/X86TimerLib.c
===================================================================
--- edk2.orig/MdePkg/Library/SecPeiDxeTimerLibCpu/X86TimerLib.c
+++ edk2/MdePkg/Library/SecPeiDxeTimerLibCpu/X86TimerLib.c
@@ -90,8 +90,9 @@ InternalX86GetTimerFrequency (
   IN      UINTN  ApicBase
   )
 {
+  PcdGet32 (PcdFSBClock);^M
   return
-    PcdGet32 (PcdFSBClock) /^M
+    100000000 /^M
     mTimerLibLocalApicDivisor[MmioBitFieldRead32 (ApicBase + APIC_TDCR, 0, 3)];
 }

Looks that the value of PcdFSBClock is not root cause. The point is that using PcdGet32 to access a dynamic PcdFSBClock value causes problem. Before 71cdb91f31 patch, PcdFSBClock is a fixed value in the PcdsFixedAtBuild section in  MdePkg/MdePkg.dec. After 71cdb91f31 patch be applied, PcdFSBClock be moved to the PcdsDynamic section. So the PcdGet32() behavior is changed.

tianocore-issues, 2024-12-20:
## Comment 21012
Date: 2023-02-09 10:18:55 +0000
From: Joey Lee &lt;<jlee>&gt;

Hi Anthony,

Thanks for your respone!

(In reply to anthony.perard from comment #3)
&gt; Hi, I did already send a patch for that:
&gt; https://edk2.groups.io/g/devel/message/96077
&gt; 

Thanks for your patch. I will test it on our issue machine.

&gt; But, I think that wasn&apos;t enough, so I&apos;ve got other patch to fix the issue. I
&gt; haven&apos;t posted them yet on the mailing list.
&gt; 

I will happy to test your new patch no our environment.

Thanks!

tianocore-issues, 2024-12-20:
## Comment 21008
Date: 2023-02-09 07:54:26 +0000
From: Joey Lee &lt;<jlee>&gt;

- Industry Specification: ---
- Release Observed: edk2-stable202105
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): @tperard

Running Windows 2k22 on OvmfXen got the following FATAL ERROR and hang on booting:

PartitionValidMbr: Bad MBR partition size EndingLBA(DEB56E8B) &gt; LastLBA(11CFFF)
FSOpen: Open &apos;\EFI\BOOT\BOOTX64.EFI&apos; Success
[Bds] Expand PciRoot(0x0)/Pci(0x1,0x1)/Ata(Primary,Master,0x0) -&gt; PciRoot(0x0)/Pci(0x1,0x1)/Ata(Primary,Master,0x0)/HD(1,GPT,D53CB690-4824-46B9-96B7-ABEDBA8EA173,0x800,0x32000)/\EFI\BOOT\BOOTX64.EFI
[Security] 3rd party image[0] can be loaded after EndOfDxe: PciRoot(0x0)/Pci(0x1,0x1)/Ata(Primary,Master,0x0)/HD(1,GPT,D53CB690-4824-46B9-96B7-ABEDBA8EA173,0x800,0x32000)/\EFI\BOOT\BOOTX64.EFI. 
InstallProtocolInterface: 5B1B31A1-9562-11D2-8E3F-00A0C969723B EEBAB040
Loading driver at 0x00010000000 EntryPoint=0x0001002C170 bootmgfw.efi
InstallProtocolInterface: BC62157E-3E33-4FEC-9920-2D3B36D750DF EF0B9618
ProtectUefiImageCommon - 0xEEBAB040
  - 0x0000000010000000 - 0x000000000022E000
ConvertPages: Incompatible memory types, the pages to allocate have been allocated
FATAL ERROR - RaiseTpl with OldTpl(0x1F) &gt; NewTpl(0x10)
ASSERT /root/source-git/edk2/MdeModulePkg/Core/Dxe/Event/Tpl.c(66): ((BOOLEAN)(0==1))

EDK2 version:
since edk2-stable202105 to now

Building command:
build -p OvmfPkg/OvmfXen.dsc -a X64 -b DEBUG -t GCC5 -D FD_SIZE_2MB

tianocore-issues, 2024-12-20:
## Comment 21011
Date: 2023-02-09 08:47:38 +0000
From: @tperard

Hi, I did already send a patch for that:
https://edk2.groups.io/g/devel/message/96077

But, I think that wasn&apos;t enough, so I&apos;ve got other patch to fix the issue. I haven&apos;t posted them yet on the mailing list.

Thanks.

tianocore-issues, 2024-12-20:
## Comment 21013
Date: 2023-02-09 10:20:48 +0000
From: @lersek

(Thanks for the CC but I&apos;m no longer active in upstream edk2.)

tianocore-issues, 2024-12-20:
## Comment 21015
Date: 2023-02-09 20:24:42 +0000
From: Joey Lee &lt;<jlee>&gt;

(In reply to anthony.perard from comment #3)
&gt; Hi, I did already send a patch for that:
&gt; https://edk2.groups.io/g/devel/message/96077
&gt; 
&gt; But, I think that wasn&apos;t enough, so I&apos;ve got other patch to fix the issue. I
&gt; haven&apos;t posted them yet on the mailing list.
&gt; 
&gt; Thanks.

I have tested this patch on edk2-stable202202 and edk2-stable202211. It works to me to fix problem.

r1chard-lyu, 2025-01-03:
Hi @tperard, do you have any updates on this issue? If not, could you please submit a PR with this patch?

tperard, 2025-01-08:
Hi, I think the updated patch not been posted that I mention earlier are probably those:
- https://github.com/xcp-ng-rpms/edk2/blob/0e80c49e047ea8af75698736ef38ea3fb0fd7778/SOURCES/MdePkg-SecPeiDxeTimerLibCpu-Support-for-dynamic-PcdF.patch
- https://github.com/xcp-ng-rpms/edk2/blob/0e80c49e047ea8af75698736ef38ea3fb0fd7778/SOURCES/OvmfPkg-OvmfXen-Use-RuntimeTimerLibCpu-for-DXE_DRIVER.patch

I don't know if I'll have time to do a proper submission.

r1chard-lyu, 2025-01-10:
Hi @tperard, I think these patches are worth submitting. You can feel free to submit them. By the way, if you're busy, I can help you with testing, submission, and include your sign-off, since I am following this bug. Please let me know if you need my assistance, or if you are able to submit it on your own.