tianocore/edk2 #10381: add verbose option to Shell dmem command (Bugzilla Bug 4352)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4352
Date: 2023-03-01T15:07:13+00:00
From: Stuart Yoder &lt;<stuart.yoder>&gt;
To:   Stuart Yoder &lt;<stuart.yoder>&gt;
CC:   @lgao4, Sam.Kaynor, samer.el-haj-mahmoud, Sunny.Wang, vincent.stehle

Last updated: 2023-03-21T22:06:48+00:00

tianocore-issues, 2024-12-20:
## Comment 21131
Date: 2023-03-01 15:07:13 +0000
From: Stuart Yoder &lt;<stuart.yoder>&gt;

- Industry Specification: UEFI Shell Specification
- Target OS: ---
- Bugzilla Assignee(s): Stuart Yoder &lt;<stuart.yoder>&gt;

When the dmem command is run in the UEFI Shell without any arguments it displays the name and address of EFI system table entries.

The proposal is to add a -verbose argument to the dmem command which will parse and provide a human readable display of the RT_PROPERTIES, CONFORMANCE_PROFILE, and IMAGE_ EXECUTION_INFO tables if they are present.

Example below:

fs0:\&gt; dmem -verbose

Memory Address 000000003FF7D808 200 Bytes 
3FF7D808: 49 42 49 20 53 59 53 54-02 00 01 00 78 00 00 00 *IBI SYST....x...*
...

Valid EFI Header at Address 000000003FF7D808 
-------------------------------------------- 
System: Table Structure size 00000078 revision 00010002 
ConIn (3F1AD388) ConOut (3F1AF288) StdErr (3F1ADD08) 
Runtime Services 000000003F3FC318
...

RT Properties Table
---------------------------
Version         0x1
Runtime Services Supported:
  GET_TIME        1
  SET_TIME        1
  WAKEUP_TIME     0
  ...        

Conformance Profile Table
---------------------------
Version     0x1
Profile GUIDs:
    UEFI_SPEC_GUID    523c91af-a195-4382-818d-295fe4006465

Image Execution Table
--------------------------------------------
EFI\BOOT\grubaa64.efi: SIG_NOT_FOUND

tianocore-issues, 2024-12-20:
## Comment 21132
Date: 2023-03-01 15:08:58 +0000
From: Stuart Yoder &lt;<stuart.yoder>&gt;

Created attachment 1333
ECR

Attachment: [BZ-1333-VerboseOptionDmem_v1.docx](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-1333-VerboseOptionDmem_v1.docx)

tianocore-issues, 2024-12-20:
## Comment 21239
Date: 2023-03-21 22:06:48 +0000
From: @lgao4

stuart.yoder@arm.com: please work it.

pierregondois, 2025-09-18:
This was implemented by:
- 497766f70975 ("ShellPkg: UefiShellDebug1CommandsLib: Conformance Profiles in Dmem.c")