tianocore/edk2 #10419: Introduce new status code to UEFI PI specification 1.8 (Bugzilla Bug 4483)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4483
Date: 2023-06-15T09:32:17+00:00
From: @nicklela
To:   @nicklela
CC:   @changab, @lgao4, @iworam, @vincent-j-zimmer

Last updated: 2024-04-25T19:44:47+00:00

tianocore-issues, 2024-12-20:
## Comment 21661
Date: 2023-06-15 09:32:17 +0000
From: @nicklela

- Industry Specification: UEFI Platform Initialization Specification Specification
- Target OS: ---
- Bugzilla Assignee(s): @nicklela

# Title: Introduce new status code to UEFI PI specification 1.8

# Status: Draft

[Status] must be one of the following:
* Draft
* Submitted to industry standard forum
* Accepted by industry standard forum
* Accepted by industry standard forum with modifications
* Rejected by industry standard forum

# Document: UEFI Platform Initialization Specification Version 1.

Here are some examples of [Title and Version]:
* UEFI Specification Version 2.8
* ACPI Specification Version 6.3
* UEFI Shell Specification Version 2.2
* UEFI Platform Initialization Specification Version 1.7
* UEFI Platform Initialization Distribution Packaging Specification Version 1.1

# License

SPDX-License-Identifier: CC-BY-4.0

# Submitter: Nickle Wang&lt;nicklew@nvidia.com&gt;

# Summary of the change

Introduce EFI_COMPUTING_UNIT_MANAGEABILITY status code for ManageabilityPkg and Redfish*Pkg

# Benefits of the change

EFI_COMPUTING_UNIT_MANAGEABILITY will be used in edk2 RedfishPkg and edk2-redfish-client RedfishClientPkg to report Redfish operation errors. It will also be used to report errors in edk2-platforms ManageabilityPkg.

# Impact of the change

This is newly introduced status code for reporting manageabilty status so I don&apos;t see impact to current system.

# Detailed description of the change [normative updates]

EFI_COMPUTING_UNIT_MANAGEABILITY is created as one of the subclasses in computing unit class.

#define EFI_COMPUTING_UNIT_CACHE               (EFI_COMPUTING_UNIT | 0x00040000)
#define EFI_COMPUTING_UNIT_MEMORY              (EFI_COMPUTING_UNIT | 0x00050000)
#define EFI_COMPUTING_UNIT_CHIPSET             (EFI_COMPUTING_UNIT | 0x00060000)
+ #define EFI_COMPUTING_UNIT_MANAGEABILITY       (EFI_COMPUTING_UNIT | 0x00070000)

Below operation values are defined to report failure in manageability related operations. I only provide the definitions for Redfish functions but the failure case like in MCTP, IPMI and KCS can be created in the future.

+///
+/// Computing Unit Manageability Subclass Error Code definitions.
+/// The detail information is reported by REPORT_STATUS_CODE_WITH_EXTENDED_DATA
+//  with ASCII string in EFI_STATUS_CODE_STRING_DATA.
+///@{
+#define EFI_MANAGEABILITY_EC_REDFISH_COMMUNICATION_ERROR        (EFI_SUBCLASS_SPECIFIC | 0x00000000)
+#define EFI_MANAGEABILITY_EC_REDFISH_HOST_INTERFACE_ERROR       (EFI_SUBCLASS_SPECIFIC | 0x00000001)
+#define EFI_MANAGEABILITY_EC_REDFISH_BOOTSTRAP_CREDENTIAL_ERROR (EFI_SUBCLASS_SPECIFIC | 0x00000002)

* EFI_MANAGEABILITY_EC_REDFISH_COMMUNICATION_ERROR will be used to report communication failure between host and Redfish service providers (or BMC).
* EFI_MANAGEABILITY_EC_REDFISH_HOST_INTERFACE_ERROR is reported when host system can not create Redfish host interface due to some errors.
* EFI_MANAGEABILITY_EC_REDFISH_BOOTSTRAP_CREDENTIAL_ERROR is reported when host system cannot get bootstrap credentials by following the Host Interface standard.

Detail reason will be provided in ASCII string by calling REPORT_STATUS_CODE_WITH_EXTENDED_DATA().

RFC: 
* https://edk2.groups.io/g/devel/message/105525
* https://edk2.groups.io/g/devel/message/105595
* https://edk2.groups.io/g/rfc/message/802

tianocore-issues, 2024-12-20:
## Comment 21662
Date: 2023-06-15 09:53:58 +0000
From: @nicklela

I follow the code-first-process and create a branch at edk2-staging: https://github.com/tianocore/edk2-staging/commits/BZ4483-new-manageablity-pi-status-code

tianocore-issues, 2024-12-20:
## Comment 21747
Date: 2023-07-18 22:40:03 +0000
From: @lgao4

nicklew@nvidia.com is working on it.

tianocore-issues, 2024-12-20:
## Comment 21793
Date: 2023-08-01 21:19:48 +0000
From: @nicklela

Update status:

M02395: Code first - Introduce new status code for ManageabilityPkg and Redfish*Pkg https://mantis.uefi.org/mantis/view.php?id=2395 (feedback)

approved as new content

Minutes of 7/20/2023 meeting: https://members.uefi.org/wg/piwg/mail/thread/9738

tianocore-issues, 2024-12-20:
## Comment 22985
Date: 2024-04-25 19:44:47 +0000
From: @nicklela

New status code is in PI 1.9 draft review now. 

1.9 2395 New status code for ManageabilityPkg and Redfish*Pkg 
Section III-6.7.1, Table 6.5, Section III-6.4.1.4.9

PI 1.9 draft: https://members.uefi.org/wg/piwg/document/1434

nicklela, 2025-01-03:
PI 1.9 is published and this is pull-request under review: https://github.com/tianocore/edk2/pull/6560