tianocore/edk2 #10423: Implement Delayed Dispatch PPI (Bugzilla Bug 4496)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4496
Date: 2023-06-30T13:04:31+00:00
From: @kuqin12
To:   @kuqin12
CC:   @lgao4

Last updated: 2023-07-24T13:27:46+00:00

tianocore-issues, 2024-12-20:
## Comment 21705
Date: 2023-06-30 13:04:31 +0000
From: @kuqin12

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): @kuqin12

The Delayed Dispatch PPI is (re)defined in PI specification v1.8. This ticket is created to track the implementation and upstream progress.

Project MU has this implemented and running for a while. I will integrate the change and upstream to the mailing list.

tianocore-issues, 2024-12-20:
## Comment 21735
Date: 2023-07-18 22:25:15 +0000
From: @lgao4

kun.qin@microsoft.com: please work on it.

tianocore-issues, 2024-12-20:
## Comment 21767
Date: 2023-07-24 13:27:46 +0000
From: @kuqin12

V1 patch sent as https://edk2.groups.io/g/devel/message/107111.

tianocore-issues, 2024-12-20:
## Comment 21757
Date: 2023-07-21 04:56:41 +0000
From: @lgao4

This is a new feature.

kuqin12, 2025-04-29:
This is resolved by d64d1e195ceb003c824786bfd817227c7ae81f06