tianocore/edk2 #10445: Add support for SMBIOS spec 3.7.0 to SmBios.h (Bugzilla Bug 4563)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4563
Date: 2023-10-09T08:21:26+00:00
From: shenbagadevir
To:   shenbagadevir
CC:   @lgao4, sainadhn, shenbagadevir, srinivasanm, sundaresans

Last updated: 2024-05-27T07:31:35+00:00

tianocore-issues, 2024-12-20:
## Comment 21968
Date: 2023-10-09 08:21:26 +0000
From: shenbagadevir

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): shenbagadevir

Created attachment 1423
Proposed Patch

Created an attachment
Proposed Patch

Please include support for Smbios Spec V 3.7.0 to IndustryStandard\SmBios.h under MdePkg. We will contribute the changes, attached the Patch for reference.

Attachment: [BZ-1423-Smbios-3.7-MdePkg-Changes-V1.zip](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-1423-Smbios-3.7-MdePkg-Changes-V1.zip)

tianocore-issues, 2024-12-20:
## Comment 21995
Date: 2023-10-18 22:42:31 +0000
From: @lgao4

 shenbagadevir@ami.com: please send the patch to edk2 mail list for code review.

tianocore-issues, 2024-12-20:
## Comment 22890
Date: 2024-04-09 09:23:35 +0000
From: shenbagadevir

Hi @gaoliming@byosoft.com.cn,

I have sent the patch edk2 mail list.
Please review and let me know your comments.

Thanks,
Shenba

tianocore-issues, 2024-12-20:
## Comment 22978
Date: 2024-04-25 08:47:13 +0000
From: shenbagadevir

Hi @gaoliming@byosoft.com.cn,

I have sent patch to update ShellPkg SmbiosView to support the updated TYPE17.
Please help to review [PATCH] MdePkg:Added support for Smbios 3.7.0 Spec as well and proceed to create pull request.

Thanks,
Shenba

tianocore-issues, 2024-12-20:
## Comment 23062
Date: 2024-05-27 07:31:35 +0000
From: shenbagadevir

Hi @gaoliming@byosoft.com.cn,

I have committed changes from only one file from MdePkg, but the PatchCheck is failed showing commits in multiple Packages error. When I run Patch check locally before pushing, the patch check was passed. Please review this and help to resolve this issue. I have also commented the same in pull request. 

https://github.com/tianocore/edk2/pull/5666

Thanks,
Shenba

pierregondois, 2026-10-07:
The PR was merged at https://github.com/tianocore/edk2/pull/5622 and https://github.com/tianocore/edk2/commit/dd58d1227c102ad4a799002f16b3a8c796d7d7cc

closed 2026-10-07 by pierregondois