tianocore/edk2 #10449: potentially invalid elapsed tick count calculation from commit 43dcf453fc15 (Bugzilla Bug 4578)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4578
Date: 2023-10-26T17:14:53+00:00
From: @lersek
To:   patrick.henz
CC:   @lersek

See also:     https://github.com/tianocore/edk2/pull/6534
Last updated: 2024-12-12T19:08:59+00:00

tianocore-issues, 2024-12-20:
## Comment 22076
Date: 2023-10-26 17:14:53 +0000
From: @lersek

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): patrick.henz

Commit 43dcf453fc15 (&quot;MdeModulePkg/XhciDxe: Use Performance Timer for
XHCI Timeouts&quot;, 2023-09-25) added TimerLib-based timeout calculation to
XhciDxe. The XhcGetElapsedTicks() function seeks to take performance
counter wrap-around into account, for both upwards counting and
downwards cointing counters.

However, the function does not consider such performance counters
(regardless of counting direction) whose numerically lowest value is not
zero (but a positive value).

For an upwards counting counter, the post-wrap CurrentTick value may not
be zero-based.

For a downwards counting counter, the pre-wrap (*PreviousTick) value may
not be zero-based.

Proposed fix (on top of master @ f945b72331d7):

&gt;| diff --git a/MdeModulePkg/Bus/Pci/XhciDxe/Xhci.c b/MdeModulePkg/Bus/Pci/XhciDxe/Xhci.c
&gt;| index 7a2e32a9dd09..c682bc41346c 100644
&gt;| --- a/MdeModulePkg/Bus/Pci/XhciDxe/Xhci.c
&gt;| +++ b/MdeModulePkg/Bus/Pci/XhciDxe/Xhci.c
&gt;| @@ -2387,20 +2387,22 @@ XhcGetElapsedTicks (
&gt;|    if (mPerformanceCounterStartValue &lt; mPerformanceCounterEndValue) {
&gt;|      //
&gt;|      // Counter counts upwards, check for an overflow condition
&gt;|      //
&gt;|      if (*PreviousTick &gt; CurrentTick) {
&gt;| -      Delta = (mPerformanceCounterEndValue - *PreviousTick) + CurrentTick;
&gt;| +      Delta = (mPerformanceCounterEndValue - *PreviousTick) +
&gt;| +              (CurrentTick - mPerformanceCounterStartValue);
&gt;|      } else {
&gt;|        Delta = CurrentTick - *PreviousTick;
&gt;|      }
&gt;|    } else {
&gt;|      //
&gt;|      // Counter counts downwards, check for an underflow condition
&gt;|      //
&gt;|      if (*PreviousTick &lt; CurrentTick) {
&gt;| -      Delta = (mPerformanceCounterStartValue - CurrentTick) + *PreviousTick;
&gt;| +      Delta = (mPerformanceCounterStartValue - CurrentTick) +
&gt;| +              (*PreviousTick - mPerformanceCounterEndValue);
&gt;|      } else {
&gt;|        Delta = *PreviousTick - CurrentTick;
&gt;|      }
&gt;|    }
&gt;|

tianocore-issues, 2024-12-20:
## Comment 22108
Date: 2023-10-31 16:47:42 +0000
From: @lersek

Patch on list:

[edk2-devel] [PATCH] MdeModulePkg/XhciDxe: Non-zero start/stop values in XhcGetElapsedTicks

https://edk2.groups.io/g/devel/message/110434
http://mid.mail-archive.com/c3038878c4d30c54e60cce7192cf1aa60c30ad2e.1698770394.git.patrick.henz@hpe.com

tianocore-issues, 2024-12-20:
## Comment 23564
Date: 2024-12-12 19:08:59 +0000
From: @lersek

https://github.com/tianocore/edk2/pull/6534

lersek, 2024-12-22:
Resolved with commit fbbf4206c13e, via #6534.

closed 2024-12-22 by lersek