tianocore/edk2 #10451: Consider removing SignedCapsulePkg from edk2 repository (Bugzilla Bug 4589)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4589
Date: 2023-11-04T21:17:39+00:00
From: @mdkinney
To:   unassigned &lt;<unassigned>&gt;
CC:   @ardbiesheuvel, @lgao4, @leiflindholm

Last updated: 2024-01-30T22:12:26+00:00

tianocore-issues, 2024-12-20:
## Comment 22129
Date: 2023-11-04 21:17:39 +0000
From: @mdkinney

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): unassigned &lt;<unassigned>&gt;

The FmpDevicePkg is the replacement for SignedCapsulePkg.

A few platforms in edk2-platforms depend on SignedCapsulePkg.

If those platforms are no longer maintained and are removed from
edk2-platforms, then SignedCapsulePkg can be safely removed from 
the edk2 repository.

The other option is to move SignedCapulePkg into the edk2-plaforms
repository.

tianocore-issues, 2024-12-20:
## Comment 22474
Date: 2024-01-30 22:12:26 +0000
From: @lgao4

Add Ard and Leif for disucssion first.

os-d, 2025-01-10:
@mdkinney @leiflindholm @ardbiesheuvel looks like in this PR: https://github.com/tianocore/edk2/pull/6475, there was agreement to archive the edk2-platforms platforms that depend on SignedCapsulePkg and then removing SignedCapsulePkg per this issue. Anything else needed to discuss or just have the removals done?

ardbiesheuvel, 2025-01-13:
So does this mean we'll get an `archived-after-edk2-stable-XXXX` branch in `edk2-platforms`, and the platforms are removed from `master`? That sounds fine to me.

mdkinney, 2025-01-13:
A tag in edk2-platforms may be better choice than a branch if there are no new commits expected.