tianocore/edk2 #10481: Buffer Underflow when performing shell command lookup (Bugzilla Bug 4684)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4684
Date: 2024-02-14T20:46:14+00:00
From: cglosner
To:   unassigned &lt;<unassigned>&gt;
CC:   @lersek

Last updated: 2024-02-15T05:34:17+00:00

tianocore-issues, 2024-12-20:
## Comment 22596
Date: 2024-02-14 20:46:14 +0000
From: cglosner

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): unassigned &lt;<unassigned>&gt;

Inside the function ShellFileHandleReadLine, right before the return there is a while loop that doesn&apos;t do a check if on the StrLen(Buffer) &gt; 0 because if the Buffer has 1 element then after one iteration you attempt to access element -1 which is invalid:

  while (Buffer[StrLen (Buffer)-1] == L&apos;\r&apos;) {
    Buffer[StrLen (Buffer)-1] = CHAR_NULL;
  }

This doesn&apos;t result in a crash, but when running it ASan it would throw an assert.

https://github.com/tianocore/edk2/blob/edc6681206c1a8791981a2f911d2fb8b3d2f5768/ShellPkg/Library/UefiShellLib/UefiShellLib.c#L4446

tianocore-issues, 2024-12-20:
## Comment 22602
Date: 2024-02-15 05:34:17 +0000
From: @lersek

The loop quoted in comment#0 is poor in general. Calling StrLen() for every character of the string makes for quadratic work.

If we assume that at most one \r may be at the end of the buffer, then a loop is unjustified, a simple &quot;if&quot; should do. And if we want to strip any final sequence of \r characters, then this is better:

  CHAR16  *CrStart;
  CHAR16  *Pos;

  CrStart = NULL;
  for (Pos = Buffer; *Pos != L&apos;\0&apos;; Pos++) {
    if (*Pos != L&apos;\r&apos;) {
      CrStart = NULL;
      continue;
    }
    if (CrStart == NULL) {
      CrStart = Pos;
    }
  }
  if (CrStart != NULL) {
    *CrStart = L&apos;\0&apos;;
  }

But, even this logic might be better to integrate into the prior read loop.

pierregondois, 2025-06-16:
Closing as fixed.

closed 2025-06-16 by pierregondois