tianocore/edk2 #10516: Add ACPI FADT table generator for X64 architecture (Bugzilla Bug 4781)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4781
Date: 2024-05-31T00:48:11+00:00
From: @abdattar
To:   @abdattar
CC:   @lgao4, @pierregondois, @samimujawar

Last updated: 2024-06-18T03:16:56+00:00

tianocore-issues, 2024-12-20:
## Comment 23083
Date: 2024-05-31 00:48:11 +0000
From: @abdattar

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): @abdattar

Add ACPI FADT table generator for X64 architecture

tianocore-issues, 2024-12-20:
## Comment 23126
Date: 2024-06-18 01:42:51 +0000
From: @lgao4

please work on it.

tianocore-issues, 2024-12-20:
## Comment 23133
Date: 2024-06-18 03:16:56 +0000
From: @pierregondois

Hello,
the x86 FADT generator has a dependence on the DynamicTablesPkg reorg
(https://edk2.groups.io/g/devel/message/116330)

I should send the patches Abdul depends on during this week,
Regards,
Pierre

abdattar, 2025-01-19:
Changes has been merged.
bc0fc75637 DynamicTablesPkg/AcpiFadtLib: Adds FADT X64 generator

closed 2025-01-19 by abdattar