tianocore/edk2 #10529: Code First - PI Specification - Add RNG PPI (Bugzilla Bug 4814)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4814
Date: 2024-07-17T20:32:31+00:00
From: @makubacki
To:   @makubacki
CC:   @lgao4

Last updated: 2024-09-04T20:49:38+00:00

tianocore-issues, 2024-12-20:
## Comment 23241
Date: 2024-07-17 20:32:31 +0000
From: @makubacki

- Industry Specification: UEFI Platform Initialization Specification Specification
- Target OS: ---
- Bugzilla Assignee(s): @makubacki

The proposed addition to the PI Specification is a Random Number Generator (RNG) PPI to provide a standard binary interface for a platform PEIM to provide RNG services to other PEIMs. Similar to the RNG Protocol defined in the UEFI 2.10 Specification, the RNG PPI returns random numbers and can serve as an entropy source for seeding other cryptographic services. The RNG PPI interface is proposed to use the same interface as the protocol for API consistency and implementation reuse.

Security has evolved in modern firmware implementations to necessitate supporting technologies that depend on strong entropy sources. More content is expected to be secure at rest, in transit, and in use. In addition, random numbers strengthen recent security mitigations being adopted in PEI phase modules such as dynamically generated stack cookie values.

At the same time, secure software supply chain practices are gaining adoption with an emphasis on software inventory management and software bill of materials (SBOM) generation. This lends to more binary distribution alongside an accompanying SBOM especially for security sensitive components such as the platform RNG provider. RNG technologies also vary across architecture, vendor, and computing segments. A platform may need to provide RNG services in PEI across binary boundaries and that can be accomplished with the RNG PPI.

Because this change is introducing an entirely new API, there will not be any impact on existing implementations. An instance of the RngLib will be provided that instead of statically linking RNG code directly will locate and use the RNG PPI. As is always the case for dynamic interfaces like PPIs, platform integrators will need to account for dependencies between the PEIM that produces the RNG PPI and the PEIMs that consume it.

tianocore-issues, 2024-12-20:
## Comment 23242
Date: 2024-07-17 21:23:00 +0000
From: @makubacki

Code first branch is posted in edk-staging:

https://github.com/tianocore/edk2-staging/tree/BZ4814-AddRngPpi

Some small refactoring may be made to the PeiRngLib and RngPei implementations, but they are representative of their responsibilities.

The document with additional details is here:

https://github.com/tianocore/edk2-staging/blob/BZ4814-AddRngPpi/CodeFirst/BZ4814-SpecChange.md

tianocore-issues, 2024-12-20:
## Comment 23371
Date: 2024-09-04 20:49:38 +0000
From: @lgao4

michael.kubacki@microsoft.com has worked on it.

sachinami, 2025-01-28:
Hi @makubacki 

As RNG PPI has been added to the PI 1.9 Specification, would it be possible now to merge the changes to the edk2 master.

makubacki, 2025-01-28:
> Hi [@makubacki](https://github.com/makubacki)
> 
> As RNG PPI has been added to the PI 1.9 Specification, would it be possible now to merge the changes to the edk2 master.

Hi @sachinami, I created the linked pull request in edk2 with the change.

sachinami, 2025-01-29:
> > Hi [@makubacki](https://github.com/makubacki)
> > As RNG PPI has been added to the PI 1.9 Specification, would it be possible now to merge the changes to the edk2 master.
> 
> Hi [@sachinami](https://github.com/sachinami), I created the linked pull request in edk2 with the change.

Thank you @makubacki !

closed 2025-02-06 by mergify[bot]