tianocore/edk2 #10533: index of array out of bound when debug (Bugzilla Bug 4834)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4834
Date: 2024-08-21T04:41:18+00:00
From: Wenyi, Xie &lt;<xiewenyi0201>&gt;
To:   Wenyi, Xie &lt;<xiewenyi0201>&gt;
CC:   @lgao4

Last updated: 2024-09-04T20:38:06+00:00

tianocore-issues, 2024-12-20:
## Comment 23334
Date: 2024-08-21 04:41:18 +0000
From: Wenyi, Xie &lt;<xiewenyi0201>&gt;

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): Wenyi, Xie &lt;<xiewenyi0201>&gt;

the debug in function Tpm12TisTpmCommand using TpmOutSize as the bound.
for (Index = 0; Index &lt; TpmOutSize; Index++) {
  DEBUG ((DEBUG_VERBOSE, &quot;%02x &quot;, BufferOut[Index]));
}
As the TpmOutSize is read from tpm, when something wrong happened in tpm, the value read from tpm maybe large and it lead to array access out of bound. Is it necessary to check the value of TpmOutSize before using it.

tianocore-issues, 2024-12-20:
## Comment 23365
Date: 2024-09-04 20:38:06 +0000
From: @lgao4

Please propose the code change for this issue.

ghbaccount, 2025-01-19:
Dear Maintainers,

please review a trivial fix for this issue.
Use original buffer value provided by caller, that is updated only after sanity check.

closed 2025-03-03 by mergify[bot]