tianocore/edk2 #10535: Add Reset Reason structure added in ACPI 6.5 (Bugzilla Bug 4839)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4839
Date: 2024-08-28T17:13:17+00:00
From: @SaloniKasbekar
To:   @SaloniKasbekar
CC:   

Last updated: 2024-08-28T19:29:40+00:00

tianocore-issues, 2024-12-20:
## Comment 23351
Date: 2024-08-28 17:13:17 +0000
From: @SaloniKasbekar

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): @SaloniKasbekar

Add definitions for Reset Reason Health Record Header, Reset Reason Health Record Structure and Reset Reason Health Record Vendor Data Entry which are part of ACPI Spec v6.5

tianocore-issues, 2024-12-20:
## Comment 23352
Date: 2024-08-28 17:14:03 +0000
From: @SaloniKasbekar

Assigned to me.

tianocore-issues, 2024-12-20:
## Comment 23353
Date: 2024-08-28 19:29:40 +0000
From: @SaloniKasbekar

PR link - https://github.com/tianocore/edk2/pull/6127

pierregondois, 2025-06-19:
PR was merged