tianocore/edk2 #10538: Exception in Intel GCC platform with OpensslFullAccel library (Bugzilla Bug 4846)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4846
Date: 2024-09-10T08:43:28+00:00
From: prarthanasv
To:   unassigned &lt;<unassigned>&gt;
CC:   kanagavels, sountharyan, srinivasanm, yi1.li

Last updated: 2024-10-25T02:35:31+00:00

tianocore-issues, 2024-12-20:
## Comment 23391
Date: 2024-09-10 08:43:28 +0000
From: prarthanasv

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): unassigned &lt;<unassigned>&gt;

Facing exception with OpensslLibFullAccel library during normal boot in Intel GCC platforms. Below exception occurs in SHA256_Update(). 

FmpDxe.Entry(691203DC)
!!!! X64 Exception Type - 0E(#PF - Page-Fault)  CPU Apic ID - 00000000 !!!!
ExceptionData - 0000000000000000  I:0 R:0 U:0 W:0 P:0 PK:0 SS:0 SGX:0
RIP  - 00000000691C09B1, CS  - 0000000000000038, RFLAGS - 0000000000210206
RAX  - 000000006F584898, RCX - 00000000693B6098, RDX - 0000001A47A44CDE
RBX  - 00000000693B6098, RSP - 000000006F584800, RBP - 00000000000003DE
RSI  - 00000000000003DE, RDI - 0000000000000000
R8   - 000000000000000F, R9  - 0000000040000000, R10 - 0000000040000200
R11  - 0000000000000000, R12 - 00000000691E9124, R13 - 00000000691E9502
R14  - 00000000691E9124, R15 - 00000000691E9504
DS   - 0000000000000030, ES  - 0000000000000030, FS  - 0000000000000030
GS   - 0000000000000030, SS  - 0000000000000030
CR0  - 0000000080010013, CR2 - 0000000000000000, CR3 - 000000006F001000
CR4  - 0000000000000668, CR8 - 0000000000000000
DR0  - 0000000000000000, DR1 - 0000000000000000, DR2 - 0000000000000000
DR3  - 0000000000000000, DR6 - 00000000FFFF0FF0, DR7 - 0000000000000400
GDTR - 0000000069BC8020 0000000000000057, LDTR - 0000000000000000
IDTR - 0000000069BF4018 0000000000000FFF,   TR - 0000000000000048
FXSAVE_STATE - 0000000069BC7470
!!!! Find image based on IP(0x691C09B1) /builds/adl/Build/AlderLake/DEBUG_GCC/X64/FmpDevicePkg/FmpDxe/2fe2cbfc-b9aa-4a93-ab5b-40173b581c42/DEBUG/FmpDxe.dll (ImageBase=000000006911B000, EntryPoint=00000000691203DC) !!!!

tianocore-issues, 2024-12-20:
## Comment 23410
Date: 2024-09-24 05:09:27 +0000
From: sountharyan

Hi @emcmillan@microsoft.com, 
@mikuback@linux.microsoft.com,
@guomin.jiang@intel.com,
@jian.j.wang@intel.com,
@jiewen.yao@intel.com,
@xiaoyu1.lu@intel.com,
@yi1.li@intel.com

Kindly provide your comments ASAP.

Thanks,
Sountharya

tianocore-issues, 2024-12-20:
## Comment 23451
Date: 2024-10-15 01:07:16 +0000
From: sountharyan

gaoliming@byosoft.com.cn

tianocore-issues, 2024-12-20:
## Comment 23452
Date: 2024-10-15 01:58:14 +0000
From: @liyi77

Hi, could you enabled AVX2 and try again?

tianocore-issues, 2024-12-20:
## Comment 23454
Date: 2024-10-15 02:35:14 +0000
From: @liyi77

(In reply to sountharyan from comment #4)
&gt; (In reply to yi1.li@intel.com from comment #3)
&gt; &gt; Hi, could you enabled AVX2 and try again?
&gt; 
&gt; Do we need to enable OPENSSL_NO_EC_NISTP_64_GCC_128 or DEFINE
&gt; OPENSSL_FLAGS_X64    = -mavx2 to enable AVX2?

I mean enable avx2 in BIOS, there should be a AVX2 Knob in CPU Configuration of BIOS SETUP.

tianocore-issues, 2024-12-20:
## Comment 23468
Date: 2024-10-22 02:20:35 +0000
From: sountharyan

Hi @yi1.li@intel.com,

AVX is enabled by default in BIOS SETUP. But still exception occurred.

Attached the Bios setup image for reference. Kindly provide your comments.

Thanks,
Sountharya

tianocore-issues, 2024-12-20:
## Comment 23453
Date: 2024-10-15 02:29:59 +0000
From: sountharyan

(In reply to yi1.li@intel.com from comment #3)
&gt; Hi, could you enabled AVX2 and try again?

Do we need to enable OPENSSL_NO_EC_NISTP_64_GCC_128 or DEFINE OPENSSL_FLAGS_X64    = -mavx2 to enable AVX2?

tianocore-issues, 2024-12-20:
## Comment 23462
Date: 2024-10-18 04:52:56 +0000
From: sountharyan

Created attachment 1549
Attached the Bios setup image.

Attachment: [BZ-1549-image.png](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-1549-image.png)

tianocore-issues, 2024-12-20:
## Comment 23463
Date: 2024-10-18 04:54:01 +0000
From: sountharyan

Hi @yi1.li@intel.com,

AVX is enabled by default. still, we are facing exception.
Attached the image for reference. 

Thanks,
Sountharya

tianocore-issues, 2024-12-20:
## Comment 23499
Date: 2024-10-25 02:35:31 +0000
From: @liyi77

I can not reproduce it on my board.
Please provide more debugging information, such as
1. Is it reproduced under VS build
2. Add more debug logs, track which line has the problem, and what is the input.

arvinrajp, 2025-02-06:
Hi @yi1.li@intel.com,

When OpensslLibFullAccel is enabled, an exception occurs only during the DXE phase while executing OpensslLibConstructor().

After analysis, it was concluded that the same function (OPENSSL_ia32_cpuid()) is used in both the PEI and DXE phases but is sourced from different .s (assembly) files. This issue occurs when consuming the OPENSSL_ia32_cpuid() function in x86_64cpuid.S in the DXE phase only.

Thanks,
Arvin Raj

arvinrajp, 2025-02-17:
Hi @liyi77, 

Kindly provide your comments on the above comment.

Thanks,
Arvin Raj

cc: @kanagavels97

mdkinney, 2025-02-17:
Can you provide more details on the build command options used?  The VS20xx compilers should use the .nasm files and not the .S files.

liyi77, 2025-02-19:
Hi, I still cannot reproduce this issue on my env (GCC 11.4.0).
Could you provide more information:
1. Is it reproduced under VS build
2. Disassembly code of the addr where PF occurs.

arvinrajp, 2025-02-24:
Hi @liyi77, 

Boots fine with the VS compiler with the OpensslLibFullAccel library enabled. Also, our environment version is GCC 20.04.6. 
The CR2 value is not NULL. Please refer to the log below.

NvramDxe.Entry(6E9D8B29)
!!!! X64 Exception Type - 0E(#PF - Page-Fault)  CPU Apic ID - 00000000 !!!!
ExceptionData - 0000000000000002  I:0 R:0 U:0 W:1 P:0 PK:0 SS:0 SGX:0
RIP  - 000000006E9E4709, CS  - 0000000000000038, RFLAGS - 0000000000210046
RAX  - 0000000000000000, RCX - 000000006E9EB880, RDX - 000000006F092080
RBX  - 0000000000000001, RSP - 000000006F068798, RBP - 5FB72E88732E755C
RSI  - 000000006E965018, RDI - 0000000000000008
R8   - 0000000000000001, R9  - 0000000060000202, R10 - 000000006E9EB120
R11  - 0000000000000080, R12 - 0000000000000007, R13 - 000000006A77B90A
R14  - 000000006F068830, R15 - 000000006F08A5D0
DS   - 0000000000000018, ES  - 0000000000000018, FS  - 0000000000000018
GS   - 0000000000000018, SS  - 0000000000000018
CR0  - 0000000080010013, CR2 - 0000000000000010, CR3 - 000000006EA01000
CR4  - 0000000000000668, CR8 - 0000000000000000
DR0  - 0000000000000000, DR1 - 0000000000000000, DR2 - 0000000000000000
DR3  - 0000000000000000, DR6 - 00000000FFFF0FF0, DR7 - 0000000000000400
GDTR - 000000006F09EBC0 0000000000000057, LDTR - 0000000000000000
IDTR - 0000000071886004 000000000000010F,   TR - 0000000000000048
FXSAVE_STATE - 000000006F09E020
!!!! Find image based on IP(0x6E9E4709) /builds/adl/Build/AlderLake/DEBUG_GCC/X64/AmiModulePkg/NVRAM/NvramDxe/DEBUG/NvramDxe.dll (ImageBase=000000006E9D2000, EntryPoint=000000006E9D8B29) !!!!

Thanks,
Arvin Raj

cc: @kanagavels97

liyi77, 2025-02-24:
CR2=0x10 is also also an illegal address.
Please send me DEBUG_GCC/X64/AmiModulePkg/NVRAM/NvramDxe folder.

arvinrajp, 2025-05-05:
Hi @liyi77 ,

Sorry for the delayed response.

The Nvram module is part of our OEM package, so unfortunately, we are unable to share its source.

We are currently encountering a GCC build error related to the use of extern in the cpuid.c file. To resolve the issue temporarily, we removed the extern declaration. Could you please advise if removing extern could lead to any potential issues or exceptions during runtime?

[image]

Additionally, have you experienced similar GCC build errors related to extern usage? Could you please provide your comments on this.

Thanks,
Arvin Raj

liyi77, 2025-05-06:
> Hi [@liyi77](https://github.com/liyi77) ,
>
> Sorry for the delayed response.
>
> The Nvram module is part of our OEM package, so unfortunately, we are unable to share its source.
>
> We are currently [...]

hi, did you modify the files in OpensslLib? OPENSSL_ia32cap_P should be defined in x86_64cpuid.s.

arvinrajp, 2025-05-06:
Hi @liyi77 , 

Yes, we **removed the extern keyword** to resolve the **GCC build error**. The build was successful on Windows, but we encountered an error with extern during the GCC build. Attached the changes for reference.

[image]

I have attached the build error we encountered for your reference. Kindly share your suggestions.

`OPENSSL_ia32cap_P' referenced in section `.text.OPENSSL_cpuid_setup' of /tmp/FmpCapsuleUpdate.dll.FYPIBD.ltrans0.ltrans.o: defined in discarded section `COMMON' of /home/BCTG/Desktop/Arvin/adl-38/adl/Build/AlderLake/RELEASE_GCC/X64/CryptoPkg/Library/OpensslLib/openssl/OpensslLibFullAccel/OUTPUT/OpensslLibFullAccel.lib(x86_64cpuid.obj)
`OPENSSL_ia32cap_P' referenced in section `.text.OPENSSL_cpuid_setup' of /tmp/FmpCapsuleUpdate.dll.FYPIBD.ltrans0.ltrans.o: defined in discarded section `COMMON' of /home/BCTG/Desktop/Arvin/adl-38/adl/Build/AlderLake/RELEASE_GCC/X64/CryptoPkg/Library/OpensslLib/openssl/OpensslLibFullAccel/OUTPUT/OpensslLibFullAccel.lib(x86_64cpuid.obj)
`OPENSSL_ia32cap_P' referenced in section `.text.OPENSSL_cpuid_setup' of /tmp/FmpCapsuleUpdate.dll.FYPIBD.ltrans0.ltrans.o: defined in discarded section `COMMON' of /home/BCTG/Desktop/Arvin/adl-38/adl/Build/AlderLake/RELEASE_GCC/X64/CryptoPkg/Library/OpensslLib/openssl/OpensslLibFullAccel/OUTPUT/OpensslLibFullAccel.lib(x86_64cpuid.obj)
collect2: error: ld returned 1 exit status

mdkinney, 2025-05-06:
Are you using that latest edk2 sources?  The COMMON section should not be discarded on GCC builds.  There have been some recent changes to BaseTools/Scripts .lds files. https://github.com/tianocore/edk2/commit/4d28a1c2fdf3673650018cf16f76cfc9796e78af

I recommend using that latest linker scripts and do not modify OpenSLL files.

liyi77, 2025-05-09:
Agree with Mike, this does not seem to be a problem with the openssl code, please try to update the edk2 code.

I do not recommend **removing the extern keyword** because this will cause the asm code cannot detect the avx capability of cpu correctly.

arvinrajp, 2025-05-09:
Hi @mdkinney / @liyi77 

I included the changes in the commit message and attempted to build, but I am still encountering the same GCC build error.

[image]

Is there anything we need to do to resolve this error?

liyi77, 2025-05-10:
> Hi [@mdkinney](https://github.com/mdkinney) / [@liyi77](https://github.com/liyi77)
> 
> I included the changes in the commit message and attempted to build, but I am still encountering the same GCC build error.
> 
> [image]
> 
> Is there anything we need to do to resolve this error?

have you re-make BaseTools after apply https://github.com/tianocore/edk2/commit/4d28a1c2fdf3673650018cf16f76cfc9796e78af?

remove BaseTools folder (remove old files)
git checkout . (restore clean BaseTools)
apply https://github.com/tianocore/edk2/commit/4d28a1c2fdf3673650018cf16f76cfc9796e78af
cd BaseTools
make

arvinrajp, 2025-06-18:
Hi @mdkinney / @liyi77 , 

Please ignore the previous comment. Once we remade the base tool, the build succeeded with the extern keyword. However, we are encountering the same exception during boot. We are currently debugging the issue and will get back to you once we identify the root cause.

Thanks,
Arvin Raj

cc: @kanagavels97 @PrarthanaSagarV

arvinrajp, 2025-07-21:
Hi @mdkinney / @liyi77 , 

We found that an exception occurs after the **pushq %rbx** instruction is executed. We confirmed this by adding a JMP instruction. Could you please share your commands?

[image]

Thanks,
Arvin Raj

mdkinney, 2025-09-10:
I suggest we add a GoogleTest unit test with the use case, so we can use good debuggers to root cause.  Based on the behavior, I am wondering of there is a calling convention mismatch.

What OS/Compiler combinations have you reproduces this exception with?