tianocore/edk2 #10542: MdeModulePkg: SataController binding driver can check for DevicePath Protocol before starting (Bugzilla Bug 4858)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4858
Date: 2024-10-01T12:07:46+00:00
From: Paul Chang &lt;<PaulChang>&gt;
To:   Paul Chang &lt;<PaulChang>&gt;
CC:   @lgao4

Last updated: 2024-10-24T04:38:12+00:00

tianocore-issues, 2024-12-20:
## Comment 23415
Date: 2024-10-01 12:07:46 +0000
From: Paul Chang &lt;<PaulChang>&gt;

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): Paul Chang &lt;<PaulChang>&gt;

Created attachment 1541
Propsed patch for reference

Ph52a PCIE to SATA card inserted on Intel MTL/ARL causes system hanging with AMI code base. The issue is not there on previous Intel platform RPL because Intel uses SataControl code in its platform package with the fix attached.

Root cause of this issue is because Ph52a&apos;s driver only uses DevicePath protocol alone and EDK2 driver only uses PciIo protocol alone. Both drivers start and try to manage SATA controller. 

To avoid seeing same issue with difference card vendor in the future, we are hoping EDK2 group can take this fix generically.

Provide AMI Internal ticket EIP803584 for reference.

Attachment: [BZ-1541-SataController_Patch_20241001.7z](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-1541-SataController_Patch_20241001.7z)

tianocore-issues, 2024-12-20:
## Comment 23484
Date: 2024-10-23 20:54:15 +0000
From: @lgao4

Paul Chang: please create edk2 pull request to make this change.

tianocore-issues, 2024-12-20:
## Comment 23495
Date: 2024-10-24 04:38:12 +0000
From: Paul Chang &lt;<PaulChang>&gt;

Hi gaoliming,

I have created pull request below.
https://github.com/tianocore/edk2/pull/6371

Thanks
Paul

pierregondois, 2026-10-07:
Fixed by: https://github.com/tianocore/edk2/pull/6385

closed 2026-10-07 by pierregondois