tianocore/edk2 #10547: Revert "BaseTools: Add Stack Cookie Support to MSVC and GCC IA32/X64/… (Bugzilla Bug 4870)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4870
Date: 2024-10-30T05:11:08+00:00
From: tabassum.yasmin
To:   unassigned &lt;<unassigned>&gt;
CC:   

Last updated: 2024-10-30T05:11:08+00:00

tianocore-issues, 2024-12-20:
## Comment 23505
Date: 2024-10-30 05:11:08 +0000
From: tabassum.yasmin

- Industry Specification: ---
- Release Observed: edk2-stable202208
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): unassigned &lt;<unassigned>&gt;

Build error as below 
Fsp24SecCoreM.lib(AutoGen.obj) : error LNK2001: unresolved external symbol __security_check_cookie
Fsp24SecCoreM.lib(SecFsp.obj) : error LNK2001: unresolved external symbol __report_rangecheckfailure

Build error is caused by checkin : https://github.com/intel-restricted/firmware.boot.uefi.iafw.externalmirror.edk2/commit/ae83c6b7fd83a5906e016a32027c1bcd792a624e

r1chard-lyu, 2024-12-25:
The page is not found.

os-d, 2024-12-25:
This commit should not be reverted. The error pointed to is likely an issue with the issue author’s setup. Please reply here with more details (build log, DSC) if you would like help.