tianocore/edk2 #10548: unresolved external symbol gettimeofday (Bugzilla Bug 4871)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4871
Date: 2024-10-30T05:13:27+00:00
From: tabassum.yasmin
To:   unassigned &lt;<unassigned>&gt;
CC:   

Last updated: 2024-10-30T05:13:27+00:00

tianocore-issues, 2024-12-20:
## Comment 23506
Date: 2024-10-30 05:13:27 +0000
From: tabassum.yasmin

- Industry Specification: ---
- Release Observed: edk2-stable202305
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): unassigned &lt;<unassigned>&gt;

Build error as below

K:\Build\NovaLakeFspPkg\DEBUG_VS2019\X64\NovaLakeFspPkg\FspSecCore\FspSecCoreO\OUTPUT\X64\FspSecEntry.iii K:\Build\NovaLakeFspPkg\DEBUG_VS2019\X64\NovaLakeFspPkg\FspSecCore\FspSecCoreO\OUTPUT\X64\FspSecEntry.ii
OpensslLib.lib(d1_lib.obj) : error LNK2001: unresolved external symbol gettimeofday

Link https://github.com/intel-restricted/fi... by Kalavakolanu, Hema Anmisha
Kalavakolanu, 

Issue has been caused by checkin as below

https://github.com/intel-restricted/firmware.boot.uefi.iafw.externalmirror.edk2/commit/6f74054bdcc9b85e1d07dfd28067df4917d415c9

r1chard-lyu, 2024-12-25:
Could you please provide your OS and OpenSSL versions?