tianocore/edk2 #10552: FMMT PEIM rebased FV feature Support (Bugzilla Bug 4879)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4879
Date: 2024-12-13T13:13:38+00:00
From: @AshrafAliS
To:   unassigned &lt;<unassigned>&gt;
CC:   

Last updated: 2024-12-13T13:13:49+00:00

tianocore-issues, 2024-12-20:
## Comment 23565
Date: 2024-12-13 13:13:38 +0000
From: @AshrafAliS

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): unassigned &lt;<unassigned>&gt;

https://github.com/tianocore/edk2/tree/master/BaseTools/Source/Python/FMMT#known-issue

FFS Replace is not working when the FVs are rebased.
C based FMMT is working, Python Based FMMT is not working.

tianocore-issues, 2024-12-20:
## Comment 23566
Date: 2024-12-13 13:13:49 +0000
From: @AshrafAliS

https://github.com/tianocore/edk2/discussions/6485

lgao4, 2025-03-13:
@YuweiChen1110 , please check it.

AshrafAliS, 2025-03-27:
@YuweiChen1110  Need your help on this issue.