tianocore/edk2 #10553: After openssl is upgraded facing __GNUC__  safe_math.h not found issue (Bugzilla Bug 4880)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4880
Date: 2024-12-16T09:30:50+00:00
From: hema.anmisha.kalavakolanu
To:   unassigned &lt;<unassigned>&gt;
CC:   

Last updated: 2024-12-16T09:30:50+00:00

tianocore-issues, 2024-12-20:
## Comment 23571
Date: 2024-12-16 09:30:50 +0000
From: hema.anmisha.kalavakolanu

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): unassigned &lt;<unassigned>&gt;

Facing the below issue after openssl is upgraded

Edk2\CryptoPkg\Library\OpensslLib\openssl\include\internal/safe_math.h(19): warning C4668: &apos;__GNUC__&apos; is not defined as a preprocessor macro, replacing with &apos;0&apos; for &apos;#if/#elif&apos;

r1chard-lyu, 2024-12-25:
A PR has been merged into the OpenSSL master, 3.2, 3.3, and 3.4 branches. 
https://github.com/openssl/openssl/pull/26204

liyi77, 2025-02-28:
fixed by : https://github.com/tianocore/edk2/pull/10750

closed 2025-02-28 by liyi77