tianocore/edk2 #7841: SdMmcPciHcDxe driver fails to detect SD card on OVMF platform with SD card enabled (Bugzilla Bug 1632)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 1632
Date: 2019-03-19T03:03:46+00:00
From: Shi Steven &lt;<steven.shi>&gt;
To:   Shi Steven &lt;<steven.shi>&gt;
CC:   f4bug, hao.a.wu, steven.shi, yonghong.zhu

Last updated: 2022-02-10T21:50:45+00:00

tianocore-issues, 2024-12-20:
## Comment 6749
Date: 2019-03-19 03:03:46 +0000
From: Shi Steven &lt;<steven.shi>&gt;

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): Shi Steven &lt;<steven.shi>&gt;

SdMmcPciHcDxe driver fails to start on OVMF platform with SD card enabled. Below are the reproduce steps in my side:

$ build -p OvmfPkg/OvmfPkgIa32X64.dsc -t CLANG38 -a IA32 -a X64 -b DEBUG -DDEBUG_ON_SERIAL_PORT
$ dd if=/dev/null of=/home/jshi19/wksp_efi/edk2/Build/rootfs/sd.img bs=1M seek=16
$ sudo qemu-system-x86_64 -m 5120 -smp 1 -pflash ~/wksp_efi/edk2-fork/Build/Ovmf3264/DEBUG_CLANG38/FV/OVMF.fd -global e1000.romfile=&quot;&quot; -machine q35 -serial mon:stdio -display none --net none -drive if=none,id=mydrive,file=/home/jshi19/wksp_efi/edk2/Build/rootfs/sd.img -device sdhci-pci,id=sdhci0 -device sd-card,id=sdcard0,drive=mydrive

... ...
Slot[0] Base Clock Frequency: 52MHz
BaseClkFreq 52MHz Divisor 65 ClockFreq 400Khz
SdCardIdentification: Found a SD device at slot [0]
SdCardSetBusMode: Switch to AccessMode 1 ClockFreq 50 BusWidth 4 fails! The Switch response is 0x0
... ...

From the above boot log, the SdCardSetBusMode() function fails.

tianocore-issues, 2024-12-20:
## Comment 6750
Date: 2019-03-19 03:13:53 +0000
From: Shi Steven &lt;<steven.shi>&gt;

Hello Laszlo,
I&apos;m trying to test the SD Card on Qemu with OVMF firmware. I&apos;m using above qemu command to enable the SD card. Could you help to take a look whether my &quot; -device sdhci-pci,id=sdhci0 -device sd-card...&quot; command is correct?

tianocore-issues, 2024-12-20:
## Comment 6751
Date: 2019-03-19 03:19:08 +0000
From: Shi Steven &lt;<steven.shi>&gt;

Forgot to mention that the OvmfPkgIa32X64 need add the SD related modules before build. Below is my example:

diff --git a/OvmfPkg/OvmfPkgIa32X64.dsc b/OvmfPkg/OvmfPkgIa32X64.dsc
@@ -947,3 +947,7 @@
       NULL|SecurityPkg/Library/HashInstanceLibSha512/HashInstanceLibSha512.inf
   }
 !endif
+
+  MdeModulePkg/Bus/Pci/SdMmcPciHcDxe/SdMmcPciHcDxe.inf
+  MdeModulePkg/Bus/Sd/SdDxe/SdDxe.inf
+  MdeModulePkg/Bus/Sd/EmmcDxe/EmmcDxe.inf
diff --git a/OvmfPkg/OvmfPkgIa32X64.fdf b/OvmfPkg/OvmfPkgIa32X64.fdf
@@ -393,6 +393,10 @@ INF  SecurityPkg/Tcg/Tcg2Config/Tcg2ConfigDxe.inf
 !endif
 !endif

+INF  MdeModulePkg/Bus/Pci/SdMmcPciHcDxe/SdMmcPciHcDxe.inf
+INF  MdeModulePkg/Bus/Sd/SdDxe/SdDxe.inf
+INF  MdeModulePkg/Bus/Sd/EmmcDxe/EmmcDxe.inf
+

tianocore-issues, 2024-12-20:
## Comment 6772
Date: 2019-03-20 07:04:05 +0000
From: @lersek

Hi Steven,

unfortunately I won&apos;t be able to help much here. I&apos;ve never used the emulated SD card of QEMU in the past, and I don&apos;t know what we should expect of it / how we should construct the command line for it. Your comment 2 also shows that OVMF has never tried to run the edk2 SD drivers on QEMU&apos;s device model.

From the MAINTAINERS file of QEMU, it seems like Phil offers &quot;Odd Fixes&quot; for the device model(s):

SD (Secure Card)
M: Philippe Mathieu-Daudé &lt;f4bug@amsat.org&gt;
S: Odd Fixes
F: include/hw/sd/sd*
F: hw/sd/core.c
F: hw/sd/sd*
F: tests/sd*

I&apos;ve added Phil to the CC list now (using his RH email); perhaps he can help. (I think he might need some time before he gets to this BZ comment though.) If it&apos;s urgent, please try to ask on the qemu-devel mailing list. Thanks.

tianocore-issues, 2024-12-20:
## Comment 6849
Date: 2019-03-21 20:13:20 +0000
From: Yonghong Zhu &lt;<yonghong.zhu>&gt;

Steven, could you help to work on it. thanks.

tianocore-issues, 2024-12-20:
## Comment 18322
Date: 2022-02-10 21:50:45 +0000
From: Shi Steven &lt;<steven.shi>&gt;

No needs to fix it. Pls reopen it if there is new need.

bam80, 2025-05-07:
Hi, apparently I faced the same issue:
https://gitlab.com/qemu-project/qemu/-/issues/2954
https://github.com/tianocore/tianocore.github.io/issues/88

Could we reopen, or I missing something?