tianocore/edk2 #7848: Need to boot into UEFI Internal Shell directly when run the emulator (Bugzilla Bug 1649)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 1649
Date: 2019-03-21T21:23:45+00:00
From: haiyan.li &lt;<haiyanx.a.li>&gt;
To:   @LiuZhiguang001
CC:   yonghong.zhu, @LiuZhiguang001

Last updated: 2019-05-13T00:55:06+00:00

tianocore-issues, 2024-12-20:
## Comment 6869
Date: 2019-03-21 21:23:45 +0000
From: haiyan.li &lt;<haiyanx.a.li>&gt;

- Industry Specification: ---
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): @LiuZhiguang001

Build EmulatorPkg and then boot. The emulator will thrown information &quot;F2 to enter Setup.\n F7 to enter Boot Manager Menu.\n Enter to boot directly.&quot; It will boot into &quot;Boot Manager Menu&quot; even if nothing is done, just like we have already pressed F7. 
Expected result: When we start the emulator, can boot into the first boot option without manual intervention. If the first boot option is UEFI Internal Shell, it will be convenient to user.

tianocore-issues, 2024-12-20:
## Comment 6887
Date: 2019-03-22 02:29:56 +0000
From: Yonghong Zhu &lt;<yonghong.zhu>&gt;

Ray, please help to check. thanks.

tianocore-issues, 2024-12-20:
## Comment 7204
Date: 2019-04-18 20:32:46 +0000
From: Yonghong Zhu &lt;<yonghong.zhu>&gt;

Ray please help to investigate . thanks.

tianocore-issues, 2024-12-20:
## Comment 7333
Date: 2019-04-29 01:23:16 +0000
From: @LiuZhiguang001

take this one.

tianocore-issues, 2024-12-20:
## Comment 7482
Date: 2019-05-09 23:05:23 +0000
From: @LiuZhiguang001

bug fixed
https://github.com/tianocore/edk2/commit/0e92957eaa50be08ebbd8f0fd4e989b380704466

tianocore-issues, 2024-12-20:
## Comment 7498
Date: 2019-05-13 00:55:06 +0000
From: haiyan.li &lt;<haiyanx.a.li>&gt;

Boot directly to Shell without pressing any buttons.
Enter Setup if press F2.
Enter Boot Manager Menu if press F7.
Boot directly if press ENTER.

edk2: 4ff689d98277c7cb15859e9e922ad5d67e05122e

pierregondois, 2025-06-19:
Closed as the bug was fixed