tianocore/edk2 #7916: ECC reports 4 errors in NetworkPkg (Bugzilla Bug 1741)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 1741
Date: 2019-04-22T23:48:37+00:00
From: haiyan.li &lt;<haiyanx.a.li>&gt;
To:   xueshengfeng &lt;<shengfengx.xue>&gt;
CC:   @dandanbi, xuemin.yuan, yonghong.zhu

Last updated: 2019-05-04T22:22:36+00:00

tianocore-issues, 2024-12-20:
## Comment 7291
Date: 2019-04-27 22:04:12 +0000
From: Yonghong Zhu &lt;<yonghong.zhu>&gt;

shengfeng, could you help to fix this regression issue ?

tianocore-issues, 2024-12-20:
## Comment 7232
Date: 2019-04-22 23:48:37 +0000
From: haiyan.li &lt;<haiyanx.a.li>&gt;

- Industry Specification: ---
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): xueshengfeng &lt;<shengfengx.xue>&gt;

ECC reports 4 errors in NetworkPkg. 
Error Code: 1010
Error Message: UNI file that is associated by INF or DEC file need define the prompt and help information.
File: C:\ECC\Edk2R9\NetworkPkg\NetworkPkg.uni
LineNo: 1
Other Error Message: 
1. STR_gEfiNetworkPkgTokenSpaceGuid_PcdIPv4PXESupport_PROMPT is missing in the NetworkPkg.uni file.
2. STR_gEfiNetworkPkgTokenSpaceGuid_PcdIPv4PXESupport_HELP is missing in the NetworkPkg.uni file.
3. STR_gEfiNetworkPkgTokenSpaceGuid_PcdIPv6PXESupport_PROMPT is missing in the NetworkPkg.uni file.
4. STR_gEfiNetworkPkgTokenSpaceGuid_PcdIPv6PXESupport_HELP is missing in the NetworkPkg.uni file.

SHA-1: b29e6365c37f60c1e3757bc1fde448102e98854c

tianocore-issues, 2024-12-20:
## Comment 7312
Date: 2019-04-28 00:01:08 +0000
From: xueshengfeng &lt;<shengfengx.xue>&gt;

(In reply to Yonghong Zhu from comment #1)
&gt; shengfeng, could you help to fix this regression issue ?

i will check it soon.

tianocore-issues, 2024-12-20:
## Comment 7359
Date: 2019-04-30 03:34:27 +0000
From: xueshengfeng &lt;<shengfengx.xue>&gt;

Already committed

SHA-1: 679906d9bd4189f2575efed2a46d23ad7027f4cf

* NetworkPkg: Add missing string token in NetworkPkg.uni

UNI file that is associated by INF or DEC file need define the prompt and help information in NetworkPkg.uni

tianocore-issues, 2024-12-20:
## Comment 7368
Date: 2019-05-04 22:22:36 +0000
From: haiyan.li &lt;<haiyanx.a.li>&gt;

Run pass with edk2:fbb0ec7ea4c0d1e9e397fe898fb7d69de0779084

kirkchoutw, 2026-11-15:
Hi @antklein ,

I've been studying edk2 in my free time, and while going through open issues (using keyword searches together with the `gh` CLI),
I noticed that this one appears to have already been fixed by the following commit:
https://github.com/tianocore/edk2/commit/679906d9bd

If so, would it be okay to close this issue?
Just a friendly reminder :)