tianocore/edk2 #9497: TLS HandShake Error in KMIP server. (Bugzilla Bug 4739)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4739
Date: 2024-03-27T06:12:03+00:00
From: sountharyan
To:   unassigned &lt;<unassigned>&gt;
CC:   yi1.li

Last updated: 2024-06-03T00:03:22+00:00

tianocore-issues, 2024-12-20:
## Comment 22821
Date: 2024-03-27 06:12:03 +0000
From: sountharyan

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): unassigned &lt;<unassigned>&gt;

Hi @yi1.li@intel.com, 
@emcmillan@microsoft.com, 
@mikuback@linux.microsoft.com,
@guomin.jiang@intel.com,
@jian.j.wang@intel.com,
@jiewen.yao@intel.com,
@xiaoyu1.lu@intel.com

We are facing handshake error in CryptoPkg from edk2-stable-202308 release tag, while communicating with KMIP server. We are currently using Openssl 3.x series. Below is the error, which we are currently facing,

TlsDoHandshake SSL_HANDSHAKE_ERROR State=0x10 SSL_ERROR_SSL
TlsDoHandshake ERROR 0x308010C=L6:R8010C
TlsDoHandshake ERROR 0xA0C0103=L14:RC0103

Kindly provide your comments ASAP to resolve this error.

cc: @karthikar@ami.com, @gayathrit@ami.com, @srinivasanm@ami.com

Thanks,
Sountharya

tianocore-issues, 2024-12-20:
## Comment 22823
Date: 2024-03-27 21:54:59 +0000
From: @liyi77

Is it a regression? have you tested it with edk2-stable-202305？
And please share you test code or detailed Tls handshake messages.

tianocore-issues, 2024-12-20:
## Comment 22825
Date: 2024-03-27 22:57:57 +0000
From: sountharyan

Hi @yi1.li@intel.com,

Yes, we tested with edk2-stable-202305. We are not facing this issue.

TlsDoHandshake SSL_HANDSHAKE_ERROR State=0x10 SSL_ERROR_SSL
TlsDoHandshake ERROR 0x308010C=L6:R8010C
TlsDoHandshake ERROR 0xA0C0103=L14:RC0103

L6 - EVP library
R8010C - ERR_R_UNSUPPORTED

L14 - ERR_LIB_SSL from ssl Lib
RC0103 - SSL_R_UNSUPPORTED_SSL_VERSION

After checking code found that error might have occurred from below code (evp_fetch.c). edk2-stable-202305, CryptoPkg is not consuming this file, so we couldn&apos;t face this error. 

if ((name_id != 0 || name != NULL) &amp;&amp; method == NULL) {
int code = unsupported ? ERR_R_UNSUPPORTED : ERR_R_FETCH_FAILED;

if (name == NULL)
name = ossl_namemap_num2name(namemap, name_id, 0);
ERR_raise_data(ERR_LIB_EVP, code,
&quot;%s, Algorithm (%s : %d), Properties (%s)&quot;,
ossl_lib_ctx_get_descriptor(methdata-&gt;libctx),
name == NULL ? &quot;&lt;null&gt;&quot; : name, name_id,
properties == NULL ? &quot;&lt;null&gt;&quot; : properties);
}

Thanks,
Sountharya

tianocore-issues, 2024-12-20:
## Comment 22826
Date: 2024-03-27 23:11:29 +0000
From: @liyi77

Currently I don&apos;t get enough information from your comments.
Please share more detailed info of your tls config, such as:
Tls version, cipher suite, complete Tls handshake message...

I have some suspicion that this problem is related to the usage of an unsupported crypto algorithm. You can try to apply this patch:

and check if there are some log related to &apos;&apos;skipping CipherId=&apos;&apos;.
Please note it is DEBUG_VERBOSE log, you can change it to DEBUG_INFO if you cannot see VERBOSE log.

TlsSetCipherList {
...
    if (StackIdx == sk_SSL_CIPHER_num (OpensslCipherStack)) {
      DEBUG ((
        DEBUG_VERBOSE,
        &quot;%a:%a: skipping CipherId=0x%04x\n&quot;,

tianocore-issues, 2024-12-20:
## Comment 22827
Date: 2024-03-27 23:35:18 +0000
From: @liyi77

(In reply to yi1.li@intel.com from comment #3)
&gt; Currently I don&apos;t get enough information from your comments.
&gt; Please share more detailed info of your tls config, such as:
&gt; Tls version, cipher suite, complete Tls handshake message...
&gt; 
&gt; I have some suspicion that this problem is related to the usage of an
&gt; unsupported crypto algorithm. You can try to apply this patch:

Missed patch link here:
https://github.com/tianocore/edk2/commit/4ddd8ac3a29d9c5974a19f36c1dc5896d813dc6e

&gt; and check if there are some log related to &apos;&apos;skipping CipherId=&apos;&apos;.
&gt; Please note it is DEBUG_VERBOSE log, you can change it to DEBUG_INFO if you
&gt; cannot see VERBOSE log.
&gt; 
&gt; TlsSetCipherList {
&gt; ...
&gt;     if (StackIdx == sk_SSL_CIPHER_num (OpensslCipherStack)) {
&gt;       DEBUG ((
&gt;         DEBUG_VERBOSE,
&gt;         &quot;%a:%a: skipping CipherId=0x%04x\n&quot;,

tianocore-issues, 2024-12-20:
## Comment 22828
Date: 2024-03-28 00:37:59 +0000
From: sountharyan

Created attachment 1486
Tls debug messages Log

Hi @yi1.li@intel.com,

I added few debug messages and here I have updated. 

I used DEBUG_ERROR, still I couldn&apos;t see any messages.

if (StackIdx == sk_SSL_CIPHER_num (OpensslCipherStack)) {
      DEBUG ((
        DEBUG_ERROR,
        &quot;%a:%a: skipping CipherId=0x%04x\n&quot;,

Thanks,
Sountharya

Attachment: [BZ-1486-CryptoPkg_Log.txt](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-1486-CryptoPkg_Log.txt)

tianocore-issues, 2024-12-20:
## Comment 22829
Date: 2024-03-28 00:42:45 +0000
From: sountharyan

(In reply to sountharyan from comment #5)
&gt; Created attachment 1486 [details]
&gt; Tls debug messages Log
&gt; 
&gt; Hi @yi1.li@intel.com,
&gt; 
&gt; I added few debug messages and here I have updated. 
&gt; Both server and client are using TLS 1.2 version. 
  Cipher suite: TLS_RSA_WITH_AES_256_CBC_SHA256

&gt; I used DEBUG_ERROR, still I couldn&apos;t see any messages.
&gt; 
&gt; if (StackIdx == sk_SSL_CIPHER_num (OpensslCipherStack)) {
&gt;       DEBUG ((
&gt;         DEBUG_ERROR,
&gt;         &quot;%a:%a: skipping CipherId=0x%04x\n&quot;,
&gt; 
&gt; 
&gt; Thanks,
&gt; Sountharya

tianocore-issues, 2024-12-20:
## Comment 22830
Date: 2024-03-28 00:58:37 +0000
From: @liyi77

Thanks for your log, it is clear and useful.
Please try to add aes256-sha256 cipher to uefiprov.c L127, I think this will solve your problem.

static const OSSL_ALGORITHM_CAPABLE deflt_ciphers[] = {
...

    ALG(PROV_NAMES_AES_256_GCM, ossl_aes256gcm_functions),
    ALG(PROV_NAMES_AES_192_GCM, ossl_aes192gcm_functions),
    ALG(PROV_NAMES_AES_128_GCM, ossl_aes128gcm_functions),

    +ALGC(PROV_NAMES_AES_128_CBC_HMAC_SHA256, ossl_aes128cbc_hmac_sha256_functions,
        ossl_cipher_capable_aes_cbc_hmac_sha256),
    +ALGC(PROV_NAMES_AES_256_CBC_HMAC_SHA256, ossl_aes256cbc_hmac_sha256_functions,
         ossl_cipher_capable_aes_cbc_hmac_sha256),
    { { NULL, NULL, NULL }, NULL }
};

tianocore-issues, 2024-12-20:
## Comment 22831
Date: 2024-03-28 03:25:34 +0000
From: sountharyan

Hi @yi1.li@intel.com,

    //ALGC(PROV_NAMES_AES_256_CBC_HMAC_SHA256, ossl_aes256cbc_hmac_sha256_functions,
      //    ossl_cipher_capable_aes_cbc_hmac_sha256),
    ALG(PROV_NAMES_AES_256_CBC_HMAC_SHA256, ossl_aes256cbc_hmac_sha256_functions),

Tried both ways, but still, it returns same error.

Thanks,
Sountharya

tianocore-issues, 2024-12-20:
## Comment 22832
Date: 2024-03-28 03:37:45 +0000
From: @liyi77

Sounds strange, have you tried OpensslLibAccel.inf?

Anyway, I think this is indeed a problem because AES256-SHA256 is a Tls1.2 suite so we should continue to support it, change status to CONFIRMED.

tianocore-issues, 2024-12-20:
## Comment 22833
Date: 2024-03-28 04:05:17 +0000
From: @liyi77

I don&apos;t have a good way to reproduce this problem locally, so please help capture the details of SSL_SESSION-&gt;cipher when the error occurs:
ssl_cipher_get_evp()
s-&gt;cipher-&gt;
     const char *name; /* text name */
     uint32_t id; /* id, 4 bytes, first is version */
     uint32_t algorithm_mkey; /* key exchange algorithm */
     uint32_t algorithm_auth; /* server authentication */
     uint32_t algorithm_enc; /* symmetric encryption */
     uint32_t algorithm_mac; /* symmetric authentication */

tianocore-issues, 2024-12-20:
## Comment 22834
Date: 2024-03-28 04:32:25 +0000
From: @liyi77

(In reply to yi1.li@intel.com from comment #7)
&gt; Thanks for your log, it is clear and useful.
&gt; Please try to add aes256-sha256 cipher to uefiprov.c L127, I think this will
&gt; solve your problem.
&gt; 
&gt; static const OSSL_ALGORITHM_CAPABLE deflt_ciphers[] = {
&gt; ...
&gt; 
&gt;     ALG(PROV_NAMES_AES_256_GCM, ossl_aes256gcm_functions),
&gt;     ALG(PROV_NAMES_AES_192_GCM, ossl_aes192gcm_functions),
&gt;     ALG(PROV_NAMES_AES_128_GCM, ossl_aes128gcm_functions),
&gt; 
&gt;     +ALGC(PROV_NAMES_AES_128_CBC_HMAC_SHA256,
&gt; ossl_aes128cbc_hmac_sha256_functions,
&gt;         ossl_cipher_capable_aes_cbc_hmac_sha256),
&gt;     +ALGC(PROV_NAMES_AES_256_CBC_HMAC_SHA256,
&gt; ossl_aes256cbc_hmac_sha256_functions,
&gt;          ossl_cipher_capable_aes_cbc_hmac_sha256),
&gt;     { { NULL, NULL, NULL }, NULL }
&gt; };

also add -DAESNI_ASM to OpensslLibAccel/AccelFull.inf:
  DEFINE OPENSSL_FLAGS_IA32      = +-DAESNI_ASM -DAES_ASM
  DEFINE OPENSSL_FLAGS_X64       = +-DAESNI_ASM -DAES_ASM

tianocore-issues, 2024-12-20:
## Comment 22847
Date: 2024-04-02 02:34:00 +0000
From: sountharyan

Hi @yi1.li@intel.com,

 ssl_cipher_get_evp Cipher id:50331709 [TlsDxe]: 
 ssl_cipher_get_evp Cipher name:AES256-SHA256 [TlsDxe]: 
 ssl_cipher_get_evp Cipher mkey:1 [TlsDxe]: 
 ssl_cipher_get_evp Cipher auth:1[TlsDxe]: 
 ssl_cipher_get_evp Cipher enc:128 [TlsDxe]: 
 ssl_cipher_get_evp Cipher mac:16 [TlsDxe]:

Thanks,
Sountharya

tianocore-issues, 2024-12-20:
## Comment 22848
Date: 2024-04-02 02:43:46 +0000
From: sountharyan

Hi @yi1.li@intel.com,

  DEFINE OPENSSL_FLAGS_IA32      = +-DAESNI_ASM -DAES_ASM
  DEFINE OPENSSL_FLAGS_X64       = +-DAESNI_ASM -DAES_ASM

Added above flags and also, we added asm related files in Openssllibfull and still we are getting same error.

Thanks,
Sountharya

tianocore-issues, 2024-12-20:
## Comment 22849
Date: 2024-04-02 02:50:14 +0000
From: @liyi77

(In reply to sountharyan from comment #13)
&gt; Hi @yi1.li@intel.com,
&gt; 
&gt;   DEFINE OPENSSL_FLAGS_IA32      = +-DAESNI_ASM -DAES_ASM
&gt;   DEFINE OPENSSL_FLAGS_X64       = +-DAESNI_ASM -DAES_ASM
&gt; 
&gt; 
&gt; Added above flags and also, we added asm related files in Openssllibfull and
&gt; still we are getting same error.
&gt; 
&gt; Thanks,
&gt; Sountharya

Openssllibfull? AES256-SHA256 is a hw feature so it only supported by OpensslLibAccel and OpensslLibFullAccel lib. Could you check if using the correct Accel lib.
I don&apos;t recommend modifying OpensslLibFull to support AES ASM (from your description I assume you do this), since all openssl INFs are automatically generated by scripts and you might miss something.

tianocore-issues, 2024-12-20:
## Comment 22852
Date: 2024-04-03 03:44:11 +0000
From: sountharyan

Hi @yi1.li@intel.com, 

Consuming OpensslLibFullAccel resolved the handshake error. Thanks for your support.

But below warning messages are printed.

OpensslLibFullAccel.lib(x86cpuid.obj) : warning LNK4210: .CRT section exists; there may be unhandled static initializers or terminators

Thanks,
Sountharya

tianocore-issues, 2024-12-20:
## Comment 22853
Date: 2024-04-03 03:46:18 +0000
From: sountharyan

Created attachment 1487
warning_messages_Build_log

Attachment: [BZ-1487-Warning_log.txt](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-1487-Warning_log.txt)

tianocore-issues, 2024-12-20:
## Comment 22854
Date: 2024-04-03 04:11:55 +0000
From: @liyi77

(In reply to yi1.li@intel.com from comment #7)

&gt; static const OSSL_ALGORITHM_CAPABLE deflt_ciphers[] = {
&gt; ...
&gt; 
&gt;     ALG(PROV_NAMES_AES_256_GCM, ossl_aes256gcm_functions),
&gt;     ALG(PROV_NAMES_AES_192_GCM, ossl_aes192gcm_functions),
&gt;     ALG(PROV_NAMES_AES_128_GCM, ossl_aes128gcm_functions),
&gt; 
&gt;     +ALGC(PROV_NAMES_AES_128_CBC_HMAC_SHA256,
&gt; ossl_aes128cbc_hmac_sha256_functions,
&gt;         ossl_cipher_capable_aes_cbc_hmac_sha256),
&gt;     +ALGC(PROV_NAMES_AES_256_CBC_HMAC_SHA256,
&gt; ossl_aes256cbc_hmac_sha256_functions,
&gt;          ossl_cipher_capable_aes_cbc_hmac_sha256),
&gt;     { { NULL, NULL, NULL }, NULL }
&gt; };

I think Accel lib still need this change to solve handshake issue, right?

tianocore-issues, 2024-12-20:
## Comment 22855
Date: 2024-04-03 04:14:09 +0000
From: @liyi77

(In reply to sountharyan from comment #15)
&gt; Hi @yi1.li@intel.com, 
&gt; 
&gt; Consuming OpensslLibFullAccel resolved the handshake error. Thanks for your
&gt; support.
&gt; 
&gt; But below warning messages are printed.
&gt; 
&gt; OpensslLibFullAccel.lib(x86cpuid.obj) : warning LNK4210: .CRT section
&gt; exists; there may be unhandled static initializers or terminators
&gt; 
&gt; Thanks,
&gt; Sountharya

This is a harmless warning, and if I remember correctly it has existed since openssl1.1. Just ignore it

tianocore-issues, 2024-12-20:
## Comment 22856
Date: 2024-04-03 04:24:22 +0000
From: sountharyan

(In reply to yi1.li@intel.com from comment #17)
&gt; (In reply to yi1.li@intel.com from comment #7)
&gt; 
&gt; &gt; static const OSSL_ALGORITHM_CAPABLE deflt_ciphers[] = {
&gt; &gt; ...
&gt; &gt; 
&gt; &gt;     ALG(PROV_NAMES_AES_256_GCM, ossl_aes256gcm_functions),
&gt; &gt;     ALG(PROV_NAMES_AES_192_GCM, ossl_aes192gcm_functions),
&gt; &gt;     ALG(PROV_NAMES_AES_128_GCM, ossl_aes128gcm_functions),
&gt; &gt; 
&gt; &gt;     +ALGC(PROV_NAMES_AES_128_CBC_HMAC_SHA256,
&gt; &gt; ossl_aes128cbc_hmac_sha256_functions,
&gt; &gt;         ossl_cipher_capable_aes_cbc_hmac_sha256),
&gt; &gt;     +ALGC(PROV_NAMES_AES_256_CBC_HMAC_SHA256,
&gt; &gt; ossl_aes256cbc_hmac_sha256_functions,
&gt; &gt;          ossl_cipher_capable_aes_cbc_hmac_sha256),
&gt; &gt;     { { NULL, NULL, NULL }, NULL }
&gt; &gt; };
&gt; 
&gt; I think Accel lib still need this change to solve handshake issue, right?

 yes, we added these changes too.

tianocore-issues, 2024-12-20:
## Comment 22857
Date: 2024-04-03 04:26:54 +0000
From: sountharyan

(In reply to yi1.li@intel.com from comment #18)
&gt; (In reply to sountharyan from comment #15)
&gt; &gt; Hi @yi1.li@intel.com, 
&gt; &gt; 
&gt; &gt; Consuming OpensslLibFullAccel resolved the handshake error. Thanks for your
&gt; &gt; support.
&gt; &gt; 
&gt; &gt; But below warning messages are printed.
&gt; &gt; 
&gt; &gt; OpensslLibFullAccel.lib(x86cpuid.obj) : warning LNK4210: .CRT section
&gt; &gt; exists; there may be unhandled static initializers or terminators
&gt; &gt; 
&gt; &gt; Thanks,
&gt; &gt; Sountharya
&gt; 
&gt; This is a harmless warning, and if I remember correctly it has existed since
&gt; openssl1.1. Just ignore it

Okay. Thanks for your comments.

tianocore-issues, 2024-12-20:
## Comment 22858
Date: 2024-04-03 04:32:02 +0000
From: sountharyan

(In reply to sountharyan from comment #19)
&gt; (In reply to yi1.li@intel.com from comment #17)
&gt; &gt; (In reply to yi1.li@intel.com from comment #7)
&gt; &gt; 
&gt; &gt; &gt; static const OSSL_ALGORITHM_CAPABLE deflt_ciphers[] = {
&gt; &gt; &gt; ...
&gt; &gt; &gt; 
&gt; &gt; &gt;     ALG(PROV_NAMES_AES_256_GCM, ossl_aes256gcm_functions),
&gt; &gt; &gt;     ALG(PROV_NAMES_AES_192_GCM, ossl_aes192gcm_functions),
&gt; &gt; &gt;     ALG(PROV_NAMES_AES_128_GCM, ossl_aes128gcm_functions),
&gt; &gt; &gt; 
&gt; &gt; &gt;     +ALGC(PROV_NAMES_AES_128_CBC_HMAC_SHA256,
&gt; &gt; &gt; ossl_aes128cbc_hmac_sha256_functions,
&gt; &gt; &gt;         ossl_cipher_capable_aes_cbc_hmac_sha256),
&gt; &gt; &gt;     +ALGC(PROV_NAMES_AES_256_CBC_HMAC_SHA256,
&gt; &gt; &gt; ossl_aes256cbc_hmac_sha256_functions,
&gt; &gt; &gt;          ossl_cipher_capable_aes_cbc_hmac_sha256),
&gt; &gt; &gt;     { { NULL, NULL, NULL }, NULL }
&gt; &gt; &gt; };
&gt; &gt; 
&gt; &gt; I think Accel lib still need this change to solve handshake issue, right?
&gt; 
&gt;  yes, we added these changes too and also added -DAESNI_ASM flags.

tianocore-issues, 2024-12-20:
## Comment 22860
Date: 2024-04-03 04:49:26 +0000
From: sountharyan

(In reply to yi1.li@intel.com from comment #22)
&gt; (In reply to sountharyan from comment #21)
&gt; &gt; (In reply to sountharyan from comment #19)
&gt; &gt; &gt; (In reply to yi1.li@intel.com from comment #17)
&gt; &gt; &gt; &gt; (In reply to yi1.li@intel.com from comment #7)
&gt; &gt; &gt; &gt; 
&gt; &gt; &gt; &gt; &gt; static const OSSL_ALGORITHM_CAPABLE deflt_ciphers[] = {
&gt; &gt; &gt; &gt; &gt; ...
&gt; &gt; &gt; &gt; &gt; 
&gt; &gt; &gt; &gt; &gt;     ALG(PROV_NAMES_AES_256_GCM, ossl_aes256gcm_functions),
&gt; &gt; &gt; &gt; &gt;     ALG(PROV_NAMES_AES_192_GCM, ossl_aes192gcm_functions),
&gt; &gt; &gt; &gt; &gt;     ALG(PROV_NAMES_AES_128_GCM, ossl_aes128gcm_functions),
&gt; &gt; &gt; &gt; &gt; 
&gt; &gt; &gt; &gt; &gt;     +ALGC(PROV_NAMES_AES_128_CBC_HMAC_SHA256,
&gt; &gt; &gt; &gt; &gt; ossl_aes128cbc_hmac_sha256_functions,
&gt; &gt; &gt; &gt; &gt;         ossl_cipher_capable_aes_cbc_hmac_sha256),
&gt; &gt; &gt; &gt; &gt;     +ALGC(PROV_NAMES_AES_256_CBC_HMAC_SHA256,
&gt; &gt; &gt; &gt; &gt; ossl_aes256cbc_hmac_sha256_functions,
&gt; &gt; &gt; &gt; &gt;          ossl_cipher_capable_aes_cbc_hmac_sha256),
&gt; &gt; &gt; &gt; &gt;     { { NULL, NULL, NULL }, NULL }
&gt; &gt; &gt; &gt; &gt; };
&gt; &gt; &gt; &gt; 
&gt; &gt; &gt; &gt; I think Accel lib still need this change to solve handshake issue, right?
&gt; &gt; &gt; 
&gt; &gt; &gt;  yes, we added these changes too and also added -DAESNI_ASM flags.
&gt; 
&gt; I checked code again -DAESNI_ASM is unnecessary, my bad please ignore it.
&gt; Would you like send patch to edk2 community to fix this issue by email?

Sure, I will send the patch to edk2 community.

tianocore-issues, 2024-12-20:
## Comment 22859
Date: 2024-04-03 04:44:54 +0000
From: @liyi77

(In reply to sountharyan from comment #21)
&gt; (In reply to sountharyan from comment #19)
&gt; &gt; (In reply to yi1.li@intel.com from comment #17)
&gt; &gt; &gt; (In reply to yi1.li@intel.com from comment #7)
&gt; &gt; &gt; 
&gt; &gt; &gt; &gt; static const OSSL_ALGORITHM_CAPABLE deflt_ciphers[] = {
&gt; &gt; &gt; &gt; ...
&gt; &gt; &gt; &gt; 
&gt; &gt; &gt; &gt;     ALG(PROV_NAMES_AES_256_GCM, ossl_aes256gcm_functions),
&gt; &gt; &gt; &gt;     ALG(PROV_NAMES_AES_192_GCM, ossl_aes192gcm_functions),
&gt; &gt; &gt; &gt;     ALG(PROV_NAMES_AES_128_GCM, ossl_aes128gcm_functions),
&gt; &gt; &gt; &gt; 
&gt; &gt; &gt; &gt;     +ALGC(PROV_NAMES_AES_128_CBC_HMAC_SHA256,
&gt; &gt; &gt; &gt; ossl_aes128cbc_hmac_sha256_functions,
&gt; &gt; &gt; &gt;         ossl_cipher_capable_aes_cbc_hmac_sha256),
&gt; &gt; &gt; &gt;     +ALGC(PROV_NAMES_AES_256_CBC_HMAC_SHA256,
&gt; &gt; &gt; &gt; ossl_aes256cbc_hmac_sha256_functions,
&gt; &gt; &gt; &gt;          ossl_cipher_capable_aes_cbc_hmac_sha256),
&gt; &gt; &gt; &gt;     { { NULL, NULL, NULL }, NULL }
&gt; &gt; &gt; &gt; };
&gt; &gt; &gt; 
&gt; &gt; &gt; I think Accel lib still need this change to solve handshake issue, right?
&gt; &gt; 
&gt; &gt;  yes, we added these changes too and also added -DAESNI_ASM flags.

I checked code again -DAESNI_ASM is unnecessary, my bad please ignore it.
Would you like send patch to edk2 community to fix this issue by email?

tianocore-issues, 2024-12-20:
## Comment 22861
Date: 2024-04-03 10:45:18 +0000
From: sountharyan

Hi @yi1.li@intel.com,

While consuming OpensslLibFullAccel.inf, met an error like to include .inc files.
.inc files consumes .pl (perl files). perl files are not supported with our side. Whether .inc files are necessary or not. kindly provide your comments

&quot;providers/implementations/ciphers/cipher_aes_hw_aesni.inc&quot;
&quot;providers/implementations/ciphers/cipher_aes_ccm_hw_aesni.inc&quot;
&quot;providers/implementations/ciphers/cipher_aes_gcm_hw_aesni.inc&quot;
&quot;providers/implementations/ciphers/cipher_aes_gcm_hw_vaes_avx512.inc&quot;

Thanks,
Sountharya

tianocore-issues, 2024-12-20:
## Comment 22865
Date: 2024-04-04 01:30:37 +0000
From: sountharyan

(In reply to sountharyan from comment #24)
&gt; Hi @yi1.li@intel.com,
&gt; 
&gt; While consuming OpensslLibFullAccel.inf, met an error like to include .inc
&gt; files.
&gt; .inc files consumes .pl (perl files). perl files are not supported with our side. Whether .inc files are necessary or not. without adding this files handshake error is resolved.
kindly provide your comments ASAP.
&gt; 
&gt; &quot;providers/implementations/ciphers/cipher_aes_hw_aesni.inc&quot;
&gt; &quot;providers/implementations/ciphers/cipher_aes_ccm_hw_aesni.inc&quot;
&gt; &quot;providers/implementations/ciphers/cipher_aes_gcm_hw_aesni.inc&quot;
&gt; &quot;providers/implementations/ciphers/cipher_aes_gcm_hw_vaes_avx512.inc&quot;
&gt; 
&gt; Thanks,
&gt; Sountharya

tianocore-issues, 2024-12-20:
## Comment 22881
Date: 2024-04-08 00:58:38 +0000
From: sountharyan

Hi @yi1.li@intel.com,

Below build error we faced while consuming OpensslLibFullAccel lib. To avoid this error we commented out few .inc files. We have a doubt on, we really need .inc files? 

OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved external symbol ossl_vaes_vpclmulqdq_capable
OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved external symbol ossl_aes_gcm_finalize_avx512
OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved external symbol ossl_aes_gcm_decrypt_avx512
OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved external symbol ossl_aes_gcm_update_aad_avx512
OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved external symbol ossl_aes_gcm_setiv_avx512
OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved external symbol ossl_aes_gcm_init_avx512
OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved external symbol ossl_aes_gcm_encrypt_avx512
OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved external symbol ossl_gcm_gmult_avx512

Thanks,
Sountharya

tianocore-issues, 2024-12-20:
## Comment 22882
Date: 2024-04-08 01:00:30 +0000
From: sountharyan

(In reply to sountharyan from comment #27)
&gt; Hi @yi1.li@intel.com,
&gt; 
&gt; Below build error we faced while consuming OpensslLibFullAccel lib. To avoid
&gt; this error we commented out few .inc files. We have a doubt on, Is inc files are really needed.
&gt; 
&gt; 
&gt; OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved
&gt; external symbol ossl_vaes_vpclmulqdq_capable
&gt; OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved
&gt; external symbol ossl_aes_gcm_finalize_avx512
&gt; OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved
&gt; external symbol ossl_aes_gcm_decrypt_avx512
&gt; OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved
&gt; external symbol ossl_aes_gcm_update_aad_avx512
&gt; OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved
&gt; external symbol ossl_aes_gcm_setiv_avx512
&gt; OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved
&gt; external symbol ossl_aes_gcm_init_avx512
&gt; OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved
&gt; external symbol ossl_aes_gcm_encrypt_avx512
&gt; OpensslLibFullAccel.lib(cipher_aes_gcm_hw.obj) : error LNK2001: unresolved
&gt; external symbol ossl_gcm_gmult_avx512
&gt; 
&gt; Thanks,
&gt; Sountharya

tianocore-issues, 2024-12-20:
## Comment 22883
Date: 2024-04-08 01:09:42 +0000
From: @liyi77

Are you using right openssl version 3.0.9?
AVX512 introduced in openssl3.2, but edk2 only compatible with 3.0.9 currently. and 3.0.9 dont have avx512 related files.

Please goto &apos;&apos;edk2\CryptoPkg\Library\OpensslLib\openssl&apos;&apos; and run &apos;&apos;git log&apos;&apos;, the correct commit id should be: de90e54bbe82e5be4fb9608b6f5c308bb837d355   Prepare for release of 3.0.9

tianocore-issues, 2024-12-20:
## Comment 22878
Date: 2024-04-06 21:16:18 +0000
From: @liyi77

Sorry I just come back from PRC holidays.

inc file is similar to .h file, it does not depend on any pl script.
I don&apos;t quite understand your question, I don&apos;t think you need to make any changes to the OpensslLibFullAccel.inf like &apos;adding inc files&apos;.

tianocore-issues, 2024-12-20:
## Comment 22884
Date: 2024-04-08 01:14:58 +0000
From: sountharyan

(In reply to yi1.li@intel.com from comment #29)
&gt; Are you using right openssl version 3.0.9?
&gt; AVX512 introduced in openssl3.2, but edk2 only compatible with 3.0.9
&gt; currently. and 3.0.9 dont have avx512 related files.
&gt; 
&gt; Please goto &apos;&apos;edk2\CryptoPkg\Library\OpensslLib\openssl&apos;&apos; and run &apos;&apos;git
&gt; log&apos;&apos;, the correct commit id should be:
&gt; de90e54bbe82e5be4fb9608b6f5c308bb837d355   Prepare for release of 3.0.9

We are using openssl 3.1.4 version

tianocore-issues, 2024-12-20:
## Comment 22885
Date: 2024-04-08 01:19:49 +0000
From: @liyi77

I see. The default version in EDK2 is openssl3.0.9, any other higher version has not been verified. I believe this problem will go away if continue to use default version.
If you need to upgrade the openssl version, please submit another BZ.

tianocore-issues, 2024-12-20:
## Comment 22913
Date: 2024-04-16 02:19:15 +0000
From: sountharyan

Hi yi1.li@intel.com,

Is CryptoPkg maximum supported version is TLS 1.2? 
TLS 1.3 is not supported in EDK2 CryptoPkg. Is our understanding correct.

#define TLS10_PROTOCOL_VERSION_MAJOR  0x03
#define TLS10_PROTOCOL_VERSION_MINOR  0x01
#define TLS11_PROTOCOL_VERSION_MAJOR  0x03
#define TLS11_PROTOCOL_VERSION_MINOR  0x02
#define TLS12_PROTOCOL_VERSION_MAJOR  0x03
#define TLS12_PROTOCOL_VERSION_MINOR  0x03

Thanks,
Sountharya

tianocore-issues, 2024-12-20:
## Comment 22921
Date: 2024-04-17 05:21:59 +0000
From: sountharyan

(In reply to sountharyan from comment #32)
&gt; Hi yi1.li@intel.com, Laszlo Ersek &lt;lersek@redhat.com&gt;
&gt; 
&gt; Is CryptoPkg maximum supported version is TLS 1.2? 
&gt; TLS 1.3 is not supported in EDK2 CryptoPkg. Is our understanding correct.
&gt; 
&gt; #define TLS10_PROTOCOL_VERSION_MAJOR  0x03
&gt; #define TLS10_PROTOCOL_VERSION_MINOR  0x01
&gt; #define TLS11_PROTOCOL_VERSION_MAJOR  0x03
&gt; #define TLS11_PROTOCOL_VERSION_MINOR  0x02
&gt; #define TLS12_PROTOCOL_VERSION_MAJOR  0x03
&gt; #define TLS12_PROTOCOL_VERSION_MINOR  0x03
&gt; 
&gt; Thanks,
&gt; Sountharya

tianocore-issues, 2024-12-20:
## Comment 23087
Date: 2024-06-03 00:03:22 +0000
From: @liyi77

pushed: https://github.com/tianocore/edk2/commit/746cc5cc40bef22d606cd22d1feb10d73a7b3d11

PrarthanaSagarV, 2025-04-24:
Hi @lersek @liyi77 ,

In Edk2-stable-202408, these changes are included. Please confirm if OpensslLibFull will also support AES-256-SHA256 suite

Thanks
Prarthana