tianocore/edk2 #9583: Add support for PE/COFF resource sections to XCODE5 tool chain in MacOS (Bugzilla Bug 557)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 557
Date: 2017-05-18T19:53:36+00:00
From: @mdkinney
To:   nobody &lt;<nobody>&gt;
CC:   @ajfish, jian.j.wang, mhaeuser, ye.y.wu

See also:     https://bugzilla.tianocore.org/show_bug.cgi?id=1355
Last updated: 2021-08-16T05:15:19+00:00

tianocore-issues, 2024-12-20:
## Comment 2170
Date: 2017-05-24 21:06:19 +0000
From: Chris &lt;<ye.y.wu>&gt;

Liming: communicate with Andrew to get apple feedback

tianocore-issues, 2024-12-20:
## Comment 2076
Date: 2017-05-18 19:53:36 +0000
From: @mdkinney

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): nobody &lt;<nobody>&gt;

The MSFT and GCC tool chains support adding HII packages to a PE/COFF resource section using the UEFI_HII_RESOURCE_SECTION      = TRUE define statement in the [Defines] section of an INF.

The XCODE5 tool chain does not support this feature yet.

tianocore-issues, 2024-12-20:
## Comment 3345
Date: 2018-01-04 20:20:22 +0000
From: nobody &lt;<nobody>&gt;

Andrew gives another option to let GenFw tool adds HII resource data into the last section of PE image. I will investigate how to let it work.

tianocore-issues, 2024-12-20:
## Comment 12682
Date: 2020-05-24 12:59:08 +0000
From: @ajfish

I investigated and the biggest issue a library is used to inject the resource section. For Xcode the libs are mach-O so there is no tool to create the section. The 2nd problem is mtoc hard codes what PE/COFF sections get generated would also need to change. Thus lets inject from GenFv. 

Here are the issues:
1) XCODE does not produce a $(MODULE_NAME)hii.lib, it does produce a $(MODULE_NAME)hii.rc but the makefile dependency is on $(MODULE_NAME)hii.lib. Thus we need to update [Hii-Binary-Package.UEFI_HII]to produced a NULL $(MODULE_NAME)hii.lib
2) We need to patch the PE/COFF image and for XCODE that only comes into existence in [Dynamic-Library-File] after mtoc is run. There is $(GENFW) -e call that is in the correct location so we can leverage that by adding a new argument, --rc, to GenFw. 
3) The $(MODULE_NAME)hii.rc is not generated for every module, so have GenFv ignore the flag if the file does not exists, just process the -e. 
4) Since GenFw is operating on the binary we can&apos;t relocate code as that will break source level debugging:
  a) If there is an empty section entry we can reuse that to add a .rsrc section. 
  b) If there is space in the image to add a section entry add a .rsrc section.
  c) If we can not inject a section append to the last section. 

The RC data is found via a directory entry so what section it is in does not impact the data getting extracted via the BasePeCoffLib. All sections in the PE/COFF are loaded by the BasePeCoffLib. 

Thus the only potential issue I see for adding the .rc to the last section is a resource section aware tool may not be able to edit it, but that is not a requirement for our build system.

tianocore-issues, 2024-12-20:
## Comment 12869
Date: 2020-06-13 18:11:03 +0000
From: @ajfish

Patches posted to mailing list

tianocore-issues, 2024-12-20:
## Comment 16912
Date: 2021-08-16 05:15:19 +0000
From: Marvin Häuser &lt;<mhaeuser>&gt;

It seems like said patches have not been merged yet? Was there any evaluation/discussion I could read on different approaches? Was it ever discussed to not have a PE/COFF resource section at all, but to e.g. embed the HII data as a constant byte array into AutoGen C?

1. This would universally work with every toolchain and every file format.
2. This would not need extra tools support.
3. This would not need extra runtime parsing work.
4. This would save protocol install/locate/uninstall work.
5. It looks to be like all HII data is byte-packed, so alignment should not matter, but even so it probably can be forced by compiler attributes.

Thoughts?

mhaeuser, 2025-02-11:
I implemented a middle ground solution (C byte array in a dedicated section, yielding a parsable format) in AUDK a while ago and we've been using this successfully with all the prior workarounds removed: https://github.com/acidanthera/audk/commit/047d1b2f8899466ef87c1d21d81c6fed0295ffc7