tianocore/edk2 #9731: Decode 'SPE Overflow Interrupt' field in MADT acpiview parser (Bugzilla Bug 1820)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 1820
Date: 2019-05-16T11:03:06+00:00
From: Krzysztof Koch &lt;<krzysztof.koch>&gt;
To:   Krzysztof Koch &lt;<krzysztof.koch>&gt;
CC:   yonghong.zhu

Last updated: 2019-05-29T04:38:58+00:00

tianocore-issues, 2024-12-20:
## Comment 7581
Date: 2019-05-16 11:03:06 +0000
From: Krzysztof Koch &lt;<krzysztof.koch>&gt;

- Industry Specification: ---
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): Krzysztof Koch &lt;<krzysztof.koch>&gt;

ACPI 6.3 update for MADT (APIC) parser in acpiview.

The ACPI 6.3 specification introduces a &apos;SPE overflow Interrupt&apos; field as part of the GICC structure.

Update the MADT parser to decode this field and validate the interrupt ID used.

tianocore-issues, 2024-12-20:
## Comment 7586
Date: 2019-05-16 20:08:27 +0000
From: Yonghong Zhu &lt;<yonghong.zhu>&gt;

Krzysztof Koch， could you help to provide patch for it ?

tianocore-issues, 2024-12-20:
## Comment 7849
Date: 2019-05-29 04:38:58 +0000
From: Krzysztof Koch &lt;<krzysztof.koch>&gt;

Yonghong Zhu, I already did. It can be seen at:

https://edk2.groups.io/g/devel/message/40815?p=,,,20,0,0,0::relevance,,acpiview+spe,20,2,0,31642384

pierregondois, 2025-06-19:
Resolved by commit:
8a08dc5486f1("ShellPkg: acpiview: ACPI 6.3 update for MADT parser")