tianocore/edk2 #9789: ArmSoftFloatLib: compiler error on VS2019 (Bugzilla Bug 2140)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 2140
Date: 2019-08-29T11:27:21+00:00
From: Baptiste Gerondeau &lt;<baptiste.gerondeau>&gt;
To:   Baptiste Gerondeau &lt;<baptiste.gerondeau>&gt;
CC:   nobody, @cetola

Last updated: 2019-11-03T21:18:10+00:00

tianocore-issues, 2024-12-20:
## Comment 9864
Date: 2019-09-28 21:27:00 +0000
From: nobody &lt;<nobody>&gt;

VS2019 tool chain has been added. You can try VS2019 tool chain.

tianocore-issues, 2024-12-20:
## Comment 10080
Date: 2019-10-17 20:42:11 +0000
From: @cetola

Baptiste, please try this fix and report back if you see additional issues.

tianocore-issues, 2024-12-20:
## Comment 10192
Date: 2019-10-23 11:55:52 +0000
From: Baptiste Gerondeau &lt;<baptiste.gerondeau>&gt;

I can confirm that the issue still exists despite explicit support for VS2019 being upstream (and it also appears when using VS2017 and the VS2017 target)
Note: the errors appear on a specific opt-GCC.h file, and berkeley-softfloat-3 seems to be specifically tailored for GCC and doesn&apos;t support non GCC builds.

I do have a workaround to allow for a build of the ARM target without softfloat (by introducing a SOFTFLOAT_ENABLE define set to true by default), I will send it out as a patch once I have cleaned it up and tested the tree on a GCC setup.

tianocore-issues, 2024-12-20:
## Comment 9399
Date: 2019-08-29 11:27:21 +0000
From: Baptiste Gerondeau &lt;<baptiste.gerondeau>&gt;

- Industry Specification: ---
- Target OS: ---
- Bugzilla Assignee(s): Baptiste Gerondeau &lt;<baptiste.gerondeau>&gt;

Created attachment 389
VS2019 Errors building berkeley-softfloat-3 ARM

Hi,

I am currently investigating building ArmVirtPkg/ArmVirtQemu.dsc using Microsoft&apos;s latest toolchain, VS2019.
To achieve this, the OpensslLibCrypto module is built, and in the process ArmSoftFloatLib also, which is a module based on the berkeley-softfloat project (3rd release).
Sadly it seems berkeley-softload-3 cannot be built with VS2019.

Attached is the log of errors I have had with VS2019 building using 
the following command :
&quot;build -a ARM -t VS2017 -p ArmVirtPkg\ArmVirtQemu.dsc&quot;

Will follow up with a workaround once it is cleaned up.

Best regards,
Baptiste GERONDEAU

Attachment: [BZ-389-berkeleysoftfloatlibvserrrors.txt](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-389-berkeleysoftfloatlibvserrrors.txt)

tianocore-issues, 2024-12-20:
## Comment 10096
Date: 2019-10-18 02:41:29 +0000
From: Baptiste Gerondeau &lt;<baptiste.gerondeau>&gt;

Will do,
Thank you

tianocore-issues, 2024-12-20:
## Comment 10295
Date: 2019-11-03 21:18:10 +0000
From: nobody &lt;<nobody>&gt;

Baptiste Gerondeau will provide the fix.

pierregondois, 2025-07-30:
The ArmSoftFloatLib was removed by:
- https://github.com/tianocore/edk2/commit/e54794bcc63e87e2405745306708949b54754b6e
- https://github.com/tianocore/edk2/commit/31ea376b587578897cdeff7b085286668cf9a7a1

Consider the build failure is not relevant anymore.