tianocore/edk2 #9811: New POODLE vulnerability in TLS v1.2 (Bugzilla Bug 2251)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 2251
Date: 2019-10-08T23:23:32+00:00
From: Wang, Jian J &lt;<jian.j.wang>&gt;
To:   guomin.jiang
CC:   BrianM, ericj, gsuckevi, @jkmathews, kevinj, rschiron, @vincent-j-zimmer

Last updated: 2022-03-10T19:24:01+00:00

tianocore-issues, 2024-12-20:
## Comment 9967
Date: 2019-10-08 23:23:32 +0000
From: Wang, Jian J &lt;<jian.j.wang>&gt;

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): guomin.jiang

A researcher reported a vulnerability in multiple load balancer/middleware/VPN products and will be presenting the findings at the Black Hat Asia conference in March.

https://www.blackhat.com/asia-19/briefings/schedule/index.html#zombie-poodle-goldendoodle-and-how-tlsv13-can-save-us-all-13741

Recommendations:

- Evaluate possible customer exposure considering version of OpenSSL used, SSL support implemented or customer specific modifications.
- Update to OpenSSL-1.1.1 or later (done, see BZ#1089)
- Migrate TlsLib to TLS v1.3

tianocore-issues, 2024-12-20:
## Comment 9968
Date: 2019-10-08 23:30:36 +0000
From: Wang, Jian J &lt;<jian.j.wang>&gt;

CVE-2019-14555

tianocore-issues, 2024-12-20:
## Comment 10612
Date: 2019-12-04 12:49:24 +0000
From: @vincent-j-zimmer

from discussion in 12/4/2019

does the UEFI spec call out the TLS level? (if yes, do uefi errata in the open, too)

two questions-

agree on bug?
yes (not a feature request)

if yes, need an embargo?
since bug open in BH talk and can tell from the wire protocol that we&apos;re only 1.2, no reason to keep our fixing closed.  recommend opening bug and fixing publicly

tianocore-issues, 2024-12-20:
## Comment 11075
Date: 2020-02-05 12:28:53 +0000
From: @vincent-j-zimmer

2/5/2020 infosec meeting - we already moved to openssl 1.1.1.  still need to update the TlsLib?  If latter is true then need patch prior to triggering embargo period.  Check to see if Jian right owner or someone in network pkg (a.r. vz)

tianocore-issues, 2024-12-20:
## Comment 13000
Date: 2020-06-30 23:06:32 +0000
From: Wang, Jian J &lt;<jian.j.wang>&gt;

Need to upgrade to TLS1.3 along with openssl1.1.1g or later

tianocore-issues, 2024-12-20:
## Comment 14239
Date: 2020-11-05 05:18:42 +0000
From: Riccardo Schirone &lt;<rschiron>&gt;

What is CVE-2019-14555 that I see in https://bugzilla.tianocore.org/show_bug.cgi?id=2251#c1 ?

tianocore-issues, 2024-12-20:
## Comment 13408
Date: 2020-08-14 14:23:14 +0000
From: Eric Johnson &lt;<ericj>&gt;

(In reply to Vincent Zimmer from comment #3)
&gt; 2/5/2020 infosec meeting - we already moved to openssl 1.1.1.  still need to
&gt; update the TlsLib?  If latter is true then need patch prior to triggering
&gt; embargo period.  Check to see if Jian right owner or someone in network pkg
&gt; (a.r. vz)

This is already public, why embargo?

tianocore-issues, 2024-12-20:
## Comment 14230
Date: 2020-11-04 13:21:40 +0000
From: @jkmathews

Moving this to public, per discussion in 11/4 Infosec meeting.  
This openssl issue is already public under CVE-2019-6593. No need for a new CVE. No need to keep this a secret.

tianocore-issues, 2024-12-20:
## Comment 14995
Date: 2021-02-03 06:30:02 +0000
From: Riccardo Schirone &lt;<rschiron>&gt;

(In reply to Riccardo Schirone from comment #7)
&gt; What is CVE-2019-14555 that I see in
&gt; https://bugzilla.tianocore.org/show_bug.cgi?id=2251#c1 ?

Any reply about this? It is very confusing to see an alias with a CVE and then another CVE number in the comments. Could someone clarify this issue? Thanks.

tianocore-issues, 2024-12-20:
## Comment 15000
Date: 2021-02-03 11:42:00 +0000
From: @jkmathews

Initially Infosec planned to assigning a new CVE to this ticket, and CVE-2019-14555 was allocated.  Later we changed direction.  

This POODLE issue was already publicly disclosed under CVE-2019-6593 by 3rd party.  That is the official CVE for this POODLE issue.

tianocore-issues, 2024-12-20:
## Comment 16580
Date: 2021-07-07 10:41:40 +0000
From: kevinj

Created attachment 763
v1 MITRE .json file

I have attached the .json file for CVE classification. Please review and provide feedback. Please assign a release this issue is observed in, so I can update the .json file.

Attachment: [BZ-763-2251.json](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-763-2251.json)

tianocore-issues, 2024-12-20:
## Comment 17125
Date: 2021-09-01 13:07:58 +0000
From: kevinj

I have attached the .json file for CVE classification. Please review and provide feedback. and assign a release this issue is observed in, so I can update the .json file and submit it back to MITRE.

tianocore-issues, 2024-12-20:
## Comment 18339
Date: 2022-02-11 15:12:09 +0000
From: Brian Mullen &lt;<BrianM>&gt;

CVSS score of 5.9. 1.5 years since last comment. Is TLS1.3 in the pipeline?

tianocore-issues, 2024-12-20:
## Comment 18429
Date: 2022-02-22 20:55:18 +0000
From: guomin.jiang

Yes, it&apos;s in my pipeline.

tianocore-issues, 2024-12-20:
## Comment 18595
Date: 2022-03-10 19:24:01 +0000
From: guomin.jiang

I am studying code.

It seem that only NetworkPkg consume it but NetworkPkg haven&apos;t support TLS1.3

Guomin

KongoHuster, 2025-01-17:
When will this bug be fixed？This issue has been published for a long time.