tianocore/edk2 #9850: TLSv1.3 support (Bugzilla Bug 2424)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 2424
Date: 2019-12-17T04:08:47+00:00
From: prarthanasv
To:   Walon Li &lt;<walon.li>&gt;
CC:   chao-jui.huang, jian.j.wang, @jyao1, macarl, Michael.Turner, nobody, prarthanasv, rajeshms, @spbrogan, timothy_hoppen, vasudevans, walon.li, yonghong.zhu

Depends on:   #3466
Last updated: 2024-05-28T00:54:49+00:00

tianocore-issues, 2024-12-20:
## Comment 10715
Date: 2019-12-17 04:08:47 +0000
From: prarthanasv

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): Walon Li &lt;<walon.li>&gt;

1. Is there any plan to support the below chiper in CryptoPkg?

Cipher Suite ID	 -   Cipher Suite String
0xC030	         -   TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
0xC02F	         -   TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256
0xCCA8	         -   TLS_ECDHE_RSA_WITH_CHACHA20_POLY1305_SHA256

2. When will a complete implementation of TLS_1.3 in CryptoPkg will be available?

tianocore-issues, 2024-12-20:
## Comment 10781
Date: 2020-01-03 13:39:53 +0000
From: @lersek

(In reply to prarthanasv from comment #0)
&gt; 1. Is there any plan to support the below chiper in CryptoPkg?
&gt; 
&gt; Cipher Suite ID	 -   Cipher Suite String
&gt; 0xC030	         -   TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
&gt; 0xC02F	         -   TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256
&gt; 0xCCA8	         -   TLS_ECDHE_RSA_WITH_CHACHA20_POLY1305_SHA256

Can these be &quot;simply&quot; added to &quot;TlsCipherMappingTable&quot; in &quot;CryptoPkg/Library/TlsLib/TlsConfig.c&quot;, perhaps?

&gt; 2. When will a complete implementation of TLS_1.3 in CryptoPkg will be
&gt; available?

TLSv1.3 lets the TLS consumer module control more crypto aspects than TLSv1.2 does. This means that CryptoPkg will have to expose additional interfaces -- some of those new interfaces may even require UEFI spec changes. For (some) details, please see:

  https://bugzilla.redhat.com/show_bug.cgi?id=1559564#c17

tianocore-issues, 2024-12-20:
## Comment 11199
Date: 2020-02-11 02:46:46 +0000
From: Wang, Jian J &lt;<jian.j.wang>&gt;

(In reply to Laszlo Ersek from comment #1)
&gt; (In reply to prarthanasv from comment #0)
&gt; &gt; 1. Is there any plan to support the below chiper in CryptoPkg?
&gt; &gt; 
&gt; &gt; Cipher Suite ID	 -   Cipher Suite String
&gt; &gt; 0xC030	         -   TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
&gt; &gt; 0xC02F	         -   TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256
&gt; &gt; 0xCCA8	         -   TLS_ECDHE_RSA_WITH_CHACHA20_POLY1305_SHA256
&gt; 
&gt; Can these be &quot;simply&quot; added to &quot;TlsCipherMappingTable&quot; in
&gt; &quot;CryptoPkg/Library/TlsLib/TlsConfig.c&quot;, perhaps?
&gt; 

For first two cipher, yes;
For third one, no. You need to remove following two macro definitions.

#ifndef OPENSSL_NO_CHACHA
# define OPENSSL_NO_CHACHA
#endif

#ifndef OPENSSL_NO_POLY1305
# define OPENSSL_NO_POLY1305
#endif

tianocore-issues, 2024-12-20:
## Comment 11223
Date: 2020-02-12 01:21:15 +0000
From: Wang, Jian J &lt;<jian.j.wang>&gt;

(In reply to Wang, Jian J from comment #2)
&gt; (In reply to Laszlo Ersek from comment #1)
&gt; &gt; (In reply to prarthanasv from comment #0)
&gt; &gt; &gt; 1. Is there any plan to support the below chiper in CryptoPkg?
&gt; &gt; &gt; 
&gt; &gt; &gt; Cipher Suite ID	 -   Cipher Suite String
&gt; &gt; &gt; 0xC030	         -   TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
&gt; &gt; &gt; 0xC02F	         -   TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256
&gt; &gt; &gt; 0xCCA8	         -   TLS_ECDHE_RSA_WITH_CHACHA20_POLY1305_SHA256
&gt; &gt; 
&gt; &gt; Can these be &quot;simply&quot; added to &quot;TlsCipherMappingTable&quot; in
&gt; &gt; &quot;CryptoPkg/Library/TlsLib/TlsConfig.c&quot;, perhaps?
&gt; &gt; 
&gt; 
&gt; For first two cipher, yes;
&gt; For third one, no. You need to remove following two macro definitions.
&gt; 
&gt; #ifndef OPENSSL_NO_CHACHA
&gt; # define OPENSSL_NO_CHACHA
&gt; #endif
&gt; 
&gt; #ifndef OPENSSL_NO_POLY1305
&gt; # define OPENSSL_NO_POLY1305
&gt; #endif

Sorry, it&apos;s removing those macro from file CryptoPkg\Library\Include\openssl\opensslconf.h.

In addition, HttpDxe uses UEFI variable &quot;HttpTlsCipherList&quot; to store supported cipher suites and pass the value of it to TlsLib. If this driver is used, the variable should also be updated to contain new cipher suite ids.

tianocore-issues, 2024-12-20:
## Comment 13101
Date: 2020-07-14 14:49:12 +0000
From: Matthew Carlson &lt;<macarl>&gt;

We have this change in Project Mu (https://github.com/microsoft/mu_tiano_plus/tree/release/202005/CryptoPkg/Library/OpensslLib) that adds ECC ciphers. Is anyone actively tackling this? Otherwise, I&apos;ll stage a patch series that fixes this.

tianocore-issues, 2024-12-20:
## Comment 14370
Date: 2020-11-19 03:33:04 +0000
From: @lersek

Related discussion:

[edk2-devel] Propose on enabling TLSv1.3

https://edk2.groups.io/g/devel/message/63676

msgid: &lt;DF4PR8401MB1067E1659BBAA855D904A51CCD4D0@DF4PR8401MB1067.NAMPRD84.PROD.OUTLOOK.COM&gt;

https://www.redhat.com/archives/edk2-devel-archive/2020-August/msg00160.html
https://www.redhat.com/archives/edk2-devel-archive/2020-September/msg00161.html
https://www.redhat.com/archives/edk2-devel-archive/2020-November/msg00842.html

tianocore-issues, 2024-12-20:
## Comment 11346
Date: 2020-02-20 18:52:03 +0000
From: Michael Turner &lt;<Michael.Turner>&gt;

One cannot &quot;Add&quot; ciphers by using the TlsCipherMappingTable or by adding them into the HttpTlsCipherList. To use any of the ECxxx ciphers, you have to reconfigure openssl removing the &quot;no-ec&quot; option in process_file.pl done by TianoCore, and reconfigure Openssl.  Note: This requires a Linux system as none of the Windows perl options like the UEFI selection of &quot;/&quot; characters a file delimiters.

Once doing this, if you expect to pass in these ECxxx ciphers in the HttpTlsCipherList, then the TlsCipherMappingTable needs to be updated.

tianocore-issues, 2024-12-20:
## Comment 13103
Date: 2020-07-14 14:49:54 +0000
From: Matthew Carlson &lt;<macarl>&gt;

*** Bug 2810 has been marked as a duplicate of this bug. ***

tianocore-issues, 2024-12-20:
## Comment 13328
Date: 2020-08-06 06:40:43 +0000
From: Matthew Huang &lt;<chao-jui.huang>&gt;

Created attachment 538
Acessing from UEFI https boot to Apache2.4 https services

We have implemented a workable TLSv1.3 based on configured process_files.pl using ActivePerl Windows + MSYS2 build and several modification in TlsLib.

But there are some minor issues we might need to discuss with the OpenSSL side, most of them are policy ones. We could start to discuss if anyone likes to move foward.

Attachment: [BZ-538-ssl_request.log](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-538-ssl_request.log)

tianocore-issues, 2024-12-20:
## Comment 16406
Date: 2021-06-21 03:35:05 +0000
From: prarthanasv

Please let us know if there is any further update on implementing TLS_1.3

tianocore-issues, 2024-12-20:
## Comment 16432
Date: 2021-06-23 07:44:31 +0000
From: @lersek

A tangential note: my understanding is that OpenSSL 3.0 is about to be
released:

https://www.openssl.org/blog/blog/2021/06/17/OpenSSL3.0ReleaseCandidate/

Perhaps as preparation for TLSv1.3 support, we should consider advancing
edk2 to OpenSSL 3.0, without any functionality changes at first. The idea is
to adapt edk2 (CryptoPkg/BaseCryptLib) to any API-level changes:

&gt;| OpenSSL 3.0 is a major release, which means that the library ABI is
&gt;| changed requiring recompilation of all dependent applications and there
&gt;| are also minor API breaking changes. For most applications that need to
&gt;| be upgraded to work with 3.0 we expect that a simple recompile will be
&gt;| sufficient. However, it is likely that application developers will notice
&gt;| new deprecation warnings when compiling their applications. Many of the
&gt;| “low level” cryptographic API functions have been deprecated in
&gt;| preference to the higher level “EVP” APIs. For detailed guidance on
&gt;| how to migrate applications to work with OpenSSL 3.0, refer to our
&gt;| migration guide.

https://www.openssl.org/docs/manmaster/man7/migration_guide.html

tianocore-issues, 2024-12-20:
## Comment 16474
Date: 2021-06-28 20:50:31 +0000
From: @jyao1

I feel this one includes too many things:

1) ECC enabling. (https://bugzilla.tianocore.org/show_bug.cgi?id=2810)

2) TLS 1.3 enabling. (This one)

3) OpenSSL 3.0 enabling. (Laszlo&apos;s comment)

I prefer we use three different Bugzilla to resolve them one by one.

Maybe 2) depends on 1).
But I do not see a strong dependency for rest.
ECC can be used without enabling TLS 1.3. In case we want ECC for other use case.
TLS 1.3 can be enabled in OpenSSL 1.1.1.

tianocore-issues, 2024-12-20:
## Comment 16480
Date: 2021-06-29 04:30:20 +0000
From: @lersek

Hi Jiewen,

should I file a separate feature request for OpenSSL 3.0 support (with no functional changes in edk2)?

Thanks,
Laszlo

tianocore-issues, 2024-12-20:
## Comment 16481
Date: 2021-06-29 04:46:25 +0000
From: @jyao1

sure. please.

tianocore-issues, 2024-12-20:
## Comment 16483
Date: 2021-06-29 05:11:43 +0000
From: @lersek

Thank you Jiewen, filed this:

CryptoPkg: consume OpenSSL 3.0
https://bugzilla.tianocore.org/show_bug.cgi?id=3466

and also made it block the present bug. (If that dependency is incorrect, then please remove / adjust it as appopriate.) Thanks!

tianocore-issues, 2024-12-20:
## Comment 19337
Date: 2022-07-27 10:01:36 +0000
From: prarthanasv

Please update if there is any progress on TLS_1.3 support changes in CryptoPkg.

tianocore-issues, 2024-12-20:
## Comment 23063
Date: 2024-05-27 23:23:53 +0000
From: Walon Li &lt;<walon.li>&gt;

Created attachment 1514
Walon TLS1.3 patch

Hi all,
I have a TLS 1.3 patch that can work properly with openssl 3.0 + TLS_AES_128_GCM_SHA256 ciphersuite. Do you consider adding this support in edk2?

Attachment: [BZ-1514-0001-CryptoPkg-Support-TLS-1.3-in-UEFI-HTTPS.patch](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-1514-0001-CryptoPkg-Support-TLS-1.3-in-UEFI-HTTPS.patch)

tianocore-issues, 2024-12-20:
## Comment 23064
Date: 2024-05-27 23:25:34 +0000
From: Walon Li &lt;<walon.li>&gt;

Created attachment 1515
Network packet verification(by wireshark) with patch

Attachment: [BZ-1515-tls13_wireshark.pcapng](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-1515-tls13_wireshark.pcapng)

tianocore-issues, 2024-12-20:
## Comment 23065
Date: 2024-05-28 00:54:18 +0000
From: @jyao1

@Walon Li, thanks for the info.

I recommend you just follow EDKII review process, either 1) send the patch to EDKII mailing list for review, or 2) submit PR in tianocore for review.

(We are in the middle of transition)

Please also describe clearly on which test you have done.

kanagavels97, 2026-01-19:
Hi @jyao1 

I checked the edk2-stable202511 release and noticed that CryptoPkg still does not include TLS 1.3 support.
Could you please let us know when TLS 1.3 support is expected to be available in EDK II?

Thanks,
Kanagavel

kanagavels97, 2026-01-19:
Hi @jyao1
According to the update shared in [comment](https://github.com/tianocore/edk2/issues/9850#issuecomment-2556242566) appears that TLS 1.3 support has already been implemented by @WalonLi, and you were requested to submit a PR for integration back on 2024‑05‑28. 

Could you please let us know whether it is acceptable to use the TLS 1.3 implementation changes on top of the edk2‑stable‑202511 CryptoPkg for our development, until the feature becomes officially available upstream?

Hi @WalonLi
Could you please let us know the testing details you have done with the TLS 1.3 support implementation?

Thanks,
Kanagavel