tianocore/edk2 #9854: Remove Framework HII "class\subclass" in Network Pkg VFR (Bugzilla Bug 2437)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 2437
Date: 2019-12-27T03:56:29+00:00
From: sivaraman &lt;<sivaramann>&gt;
To:   @jiaxinwu
CC:   @dandanbi, nobody, sivaramann, weijiwei

Last updated: 2020-10-27T01:25:04+00:00

tianocore-issues, 2024-12-20:
## Comment 10757
Date: 2019-12-27 03:56:29 +0000
From: sivaraman &lt;<sivaramann>&gt;

- Industry Specification: ---
- Target OS: ---
- Bugzilla Assignee(s): @jiaxinwu

Edk2 older branch removed the &quot;class\subclass&quot; in all VFR in the Network Stack modules. (IP4, IP6 and ISCSI) Refer below commit.

https://github.com/tianocore/edk2/commit/b965bd99c533b24bec56647a8c9b1fa2c2ed8bd8#diff-3d9f695b702160a70a74145d12121272

But when we upload the Initial IP4Dxe the changes are reverted. But IP6Config and ISCSI modules are not having those. 

https://github.com/tianocore/edk2/commit/1f6729ffe98095107ce82e67a4a0209674601a90#diff-6788a5f6871902342bd0ea72eb123df4

Please review and update the required modules

tianocore-issues, 2024-12-20:
## Comment 10827
Date: 2020-01-09 20:27:21 +0000
From: nobody &lt;<nobody>&gt;

Confirm, and remove those EDK style in VFR file.

tianocore-issues, 2024-12-20:
## Comment 14097
Date: 2020-10-27 01:25:04 +0000
From: weijiwei

Created attachment 589
remove the &quot;class\subclass&quot; in VFR

Attachment: [BZ-589-0001-NetworkPkg-remove-the-class-subclass-in-VFR.patch](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-589-0001-NetworkPkg-remove-the-class-subclass-in-VFR.patch)

pierregondois, 2025-06-19:
Closed by:
- b965bd99c533 ("Remove Framework HII "class\subclass"  in all VFR")