tianocore/edk2 #9882: [OpensslLib] Add support for EVP interface (Bugzilla Bug 2545)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 2545
Date: 2020-02-21T19:26:15+00:00
From: Christopher Zurcher &lt;<christopher.j.zurcher>&gt;
To:   Christopher Zurcher &lt;<christopher.j.zurcher>&gt;
CC:   nobody

Last updated: 2020-09-16T21:43:18+00:00

tianocore-issues, 2024-12-20:
## Comment 11391
Date: 2020-02-21 19:26:15 +0000
From: Christopher Zurcher &lt;<christopher.j.zurcher>&gt;

- Industry Specification: ---
- Target OS: ---
- Bugzilla Assignee(s): Christopher Zurcher &lt;<christopher.j.zurcher>&gt;

The OpenSSL project is slowly deprecating direct interfaces to crypto operations in favor of the EVP interface. Currently EDK does not have an EVP interface implemented. I have created an initial implementation to submit for review.

tianocore-issues, 2024-12-20:
## Comment 13702
Date: 2020-09-16 21:43:18 +0000
From: Christopher Zurcher &lt;<christopher.j.zurcher>&gt;

All CryptX.c files and HmacSha256 need to be modified to replace low-level function calls with EVP calls.

This could be done transparently if necessary, by returning UINTN from GetContextSize and using the user-supplied &quot;context&quot; to store the OpenSSL-supplied context pointer.

tianocore-issues, 2024-12-20:
## Comment 11432
Date: 2020-02-25 22:45:20 +0000
From: Christopher Zurcher &lt;<christopher.j.zurcher>&gt;

Update: All low-level interfaces have already been deprecated in OpenSSL 3.0.0. We cannot upgrade from 1.1.1 to 3.0.0 until the EVP interface is implemented.

tianocore-issues, 2024-12-20:
## Comment 11574
Date: 2020-03-05 20:19:15 +0000
From: nobody &lt;<nobody>&gt;

Please take it.

zurcher, 2025-01-15:
This can probably be closed