tianocore/edk2 #9928: NvmExpressDxe reconfigures controllers already handled by another driver (Bugzilla Bug 2773)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 2773
Date: 2020-05-28T10:34:52+00:00
From: Nathan Blythe &lt;<nblythe>&gt;
To:   @mdkinney
CC:   @ydong10, hao.a.wu, @niruiyu

Last updated: 2020-06-02T22:25:31+00:00

tianocore-issues, 2024-12-20:
## Comment 12757
Date: 2020-05-28 10:34:52 +0000
From: Nathan Blythe &lt;<nblythe>&gt;

- Industry Specification: ---
- Target OS: ---
- Bugzilla Assignee(s): @mdkinney

It appears that NvmExpressDxe reconfigures a controller&apos;s registers, assigning a new Admin Submission Queue base address (ASQ) and Admin Completion Queue base address (ACQ) and so on, even if the controller is already started and configured by another NVME driver.

One fix would be to return EFI_ALREADY_STARTED if the NVME controller is already running. Another option would be to check whether the controller has an instance of EFI_NVM_EXPRESS_PASS_THRU installed before initializing it.

tianocore-issues, 2024-12-20:
## Comment 12773
Date: 2020-06-02 21:49:27 +0000
From: @ydong10

Can you help to provide the change for this issue? The maintainer can help to review the patch. Appreciate your contribution.

or can you help to provide the reproduce steps, so we can reproduce it and continue the work.

Korenchkin, 2026-02-01:
hello there,sorry to marsh in here as a basic user,but i believe i encountered this problem,sadly i don't know how to help here,but i'll just try to describe what it does:

i have dell pro max t2 (fct2250) and kingston dc3000me(both latest bios/firmware available),and if i connect the drive directly to pcie(whichever pcie - cpu,chipset,m.2 using 'wires' converter boards a.k.a dumb boards),just when starting to boot(boot device selection phase?),bios freezes (ahci mode)

if i switch to intel vmd,or use lsi-9500 hba,system boots (i.e. nvme driver does not ?claim? the drive)

if i'm in nvme mode,i can see the drive in uefi device manager and in dell bios (just before boot/crash)

sorry,not sure what can i do to help identify problem more