tianocore/edk2 #9990: ACPI Spec ECR - Generic Interrupt Mux (Bugzilla Bug 2995)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 2995
Date: 2020-10-05T16:30:06+00:00
From: Samer El-Haj-Mahmoud &lt;<samer.el-haj-mahmoud>&gt;
To:   Samer El-Haj-Mahmoud &lt;<samer.el-haj-mahmoud>&gt;
CC:   @lgao4, harb, ikoskinen, pbrobinson, samer.el-haj-mahmoud

Last updated: 2021-05-11T17:35:10+00:00

tianocore-issues, 2024-12-20:
## Comment 13876
Date: 2020-10-05 16:30:06 +0000
From: Samer El-Haj-Mahmoud &lt;<samer.el-haj-mahmoud>&gt;

- Industry Specification: ACPI Specification
- Target OS: ---
- Bugzilla Assignee(s): Samer El-Haj-Mahmoud &lt;<samer.el-haj-mahmoud>&gt;

This ECR introduces a new ACPI concept known as the “Generic Interrupt Mux Device (GIMD)”, and associated “Generic Interrupt Mux Instances (GIMI)”. A GIMD handles interrupt combining (i.e. group of wired interrupts OR’ed into a single wired interrupt represented by a single Global System Interrupt (GSI). This ECR is intended to enable ACPI firmware to further describe the details of the interrupt combining logic in a standard way so that OSPM can handle these configurations with a new standard software device driver.

tianocore-issues, 2024-12-20:
## Comment 13877
Date: 2020-10-05 16:31:03 +0000
From: Samer El-Haj-Mahmoud &lt;<samer.el-haj-mahmoud>&gt;

Created attachment 570
CodeFirst_ACPI_GIM_ECR.docx

First draft of proposal

Attachment: [BZ-570-CodeFirst_ACPI_GIM_ECR.docx](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-570-CodeFirst_ACPI_GIM_ECR.docx)

tianocore-issues, 2024-12-20:
## Comment 13963
Date: 2020-10-13 21:56:49 +0000
From: @lgao4

Samer: please follow code first process. And, upload the code and doc into the staging branch.

tianocore-issues, 2024-12-20:
## Comment 13964
Date: 2020-10-13 21:58:00 +0000
From: @lgao4

The doc should be markdown format.

tianocore-issues, 2024-12-20:
## Comment 14035
Date: 2020-10-19 17:50:36 +0000
From: Samer El-Haj-Mahmoud &lt;<samer.el-haj-mahmoud>&gt;

OK will update to markdown and put in edk2-staging

tianocore-issues, 2024-12-20:
## Comment 14783
Date: 2021-01-11 19:57:57 +0000
From: Ilkka Koskinen &lt;<ikoskinen>&gt;

I&apos;d have a couple of comments:

1) I feel like interrupt type and polarity bits should be always valid, since they define the summary interrupt routed to a higher level IRQ controller and one often needs to tell the IRQ controller what kind of interrupt it is dealing with.

2) The other comment concerns the in interrupt clearing method and clear register behavior flags. I think, it would be more logical, if value of &apos;1&apos; means &quot;write one&quot;. That is, I&apos;d swap the values in both flags.

tianocore-issues, 2024-12-20:
## Comment 14787
Date: 2021-01-11 23:25:30 +0000
From: harb

(In reply to Ilkka Koskinen from comment #5)
&gt; I&apos;d have a couple of comments:
&gt; 
&gt; 1) I feel like interrupt type and polarity bits should be always valid,
&gt; since they define the summary interrupt routed to a higher level IRQ
&gt; controller and one often needs to tell the IRQ controller what kind of
&gt; interrupt it is dealing with.
&gt; 
&gt; 2) The other comment concerns the in interrupt clearing method and clear
&gt; register behavior flags. I think, it would be more logical, if value of &apos;1&apos;
&gt; means &quot;write one&quot;. That is, I&apos;d swap the values in both flags.

Thanks for the feedback.  Agree, this makes sense.

@Samer, any objections to these changes on your side.

tianocore-issues, 2024-12-20:
## Comment 14810
Date: 2021-01-14 18:13:08 +0000
From: Samer El-Haj-Mahmoud &lt;<samer.el-haj-mahmoud>&gt;

No objections on the two proposed changes

tianocore-issues, 2024-12-20:
## Comment 16005
Date: 2021-05-11 17:35:10 +0000
From: Samer El-Haj-Mahmoud &lt;<samer.el-haj-mahmoud>&gt;

In Progress ASWG ECR: https://mantis.uefi.org/mantis/view.php?id=2153

jmarinho, 2025-09-04:
Hi @leiflindholm,

Whenever we have time, could we set the "code-first" label in this ticket?
Since the ticket is closed in ASWG (please see https://mantis.uefi.org/mantis/view.php?id=2153), could we please close the ticket too?

Thank you,
Jose