tianocore/edk2 #10579: [Bug]: <Mismatch of EFI Conformance Profile Table GUID when UEFI application reads from EFI System Table>

sathishas89, 2025-01-02:
### Is there an existing issue for this?

- [X] I have searched existing issues

### Bug Type

- [X] Firmware
- [ ] Tool
- [ ] Unit Test

### Code first?

- [ ] Yes

### What packages are impacted?

MdeModulePkg, MdePkg

### Which targets are impacted by this bug?

DEBUG, RELEASE

### Current Behavior

When I read EFI Conformance Profile Table GUID, UEFI application returns following GUID's on Buildroot Qemu:
gST->GUID is: {36122546-F7EF-4C8F-BD9B-EB8525B50C0B} -->EFI Conformance Profile Table GUID
gST->GUID is: {B122A263-3661-4F68-9929-78F8B0D62180}
gST->GUID is: {EB66918A-7EEF-402A-842E-931D21C38AE9}
gST->GUID is: {EB9D2D31-2D88-11D3-9A16-0090273FC14D}
gST->GUID is: {B1B621D5-F19C-41A5-830B-D9152C69AAE0}

But as per UEFI 2.10 specification, EFI Conformance Profile Table GUID is 36122546-F7E7-4C8F-BD9B-EB8525B50C0B and when I read EFI System Table it's giving 36122546-F7EF-4C8F-BD9B-EB8525B50C0B guid. 

### Expected Behavior

UEFI edk2 firmware should return 36122546-F7E7-4C8F-BD9B-EB8525B50C0B for EFI Conformance Table.

### Steps To Reproduce

Tested environment: Qemu Arm64
Read guid using gST->ConfigurationTable[TableWalker].VendorGuid.
Compare with gEfiConfProfilesTableGuid

### Build Environment

```markdown
- OS(s): Ubuntu 22.04
- Tool Chain(s): aarch64-linux-gnu-
```

### Version Information

```text
Tag: edk2-stable202411
```

### Urgency

High

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

Maintainer feedback requested

YangGangUEFI, 2025-01-04:
Is this related to u-boot? If yes, please check the following information.

I found that the ECPT GUID in u-boot is `36122546-F7EF-4C8F-BD9B-EB8525B50C0B`, which is incorrect.

And I have sent a patch to u-boot.
Now it's already on the [Pull request efi-next-20250104](https://patchwork.ozlabs.org/project/uboot/patch/5222dd9e-83f6-40ad-adc1-dc3fe1651991@gmx.de/) of u-boot, and the details of the commit can be found at https://source.denx.de/u-boot/custodians/u-boot-efi/-/commits/efi-next-20250104?ref_type=tags (the latest commit:[efi: Correct ECPT table GUID](https://source.denx.de/u-boot/custodians/u-boot-efi/-/commit/ce190e1b33b8c0e1228f4123759dfa1981c202de) )

sathishas89, 2025-01-08:
Hi,
I have used latest uboot tag (v2025.01) but I could see still old ECPT table GUID.

YangGangUEFI, 2025-01-10:
Hi @sathishas89, I received an [email](https://lists.denx.de/pipermail/u-boot/2025-January/577015.html) saying that the [Pull request efi-next-20250105](http://patchwork.ozlabs.org/project/uboot/patch/6c11a4a5-8a38-4665-bd04-3e20a662d714@gmx.de/) has been applied to [u-boot/master](https://source.denx.de/u-boot/u-boot/-/commits/master?ref_type=heads). Please give some feedback again if you have time.

lgao4, 2025-01-16:
This is not the issue in Edk2 project. So, close it.

closed 2025-01-16 by lgao4