tianocore/edk2 #10663: [Bug]: [QEMU_EFI] Not Loading Network Protocol Drivers and http efi command

AndreasBBS, 2025-01-23:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### Code first?

- [ ] Yes

### What packages are impacted?

NetworkPkg, OvmfPkg

### Which targets are impacted by this bug?

DEBUG, NOOPT

### Current Behavior

In the latest stable releases I haven't been able to make the X64 Ovmf build load the networking drivers when launching the Ovmf.fd in QEMU.

This is what my `debug.log` looks like
```
Driver 6490F1C5-EBCC-4665-8892-0075B9BB49B7 was discovered but not loaded!!
Driver A2F436EA-A127-4EF8-957C-8048606FF670 was discovered but not loaded!!
Driver E4F61863-FE2C-4B56-A8F4-08519BC439DF was discovered but not loaded!!
Driver 025BBFC7-E6A9-4B8B-82AD-6815A1AEAF4A was discovered but not loaded!!
Driver 529D3F93-E8E9-4E73-B1E1-BDF6A9D50113 was discovered but not loaded!!
Driver 94734718-0BBC-47FB-96A5-EE7A5AE6A2AD was discovered but not loaded!!
Driver 9FB1A1F3-3B71-4324-B39A-745CBB015FFF was discovered but not loaded!!
Driver 6D6963AB-906D-4A65-A7CA-BD40E5D6AF2B was discovered but not loaded!!
Driver DC3641B8-2FA8-4ED3-BC1F-F9962A03454B was discovered but not loaded!!
Driver 95E3669D-34BE-4775-A651-7EA41B69D89E was discovered but not loaded!!
Driver 5BEDB5CC-D830-4EB2-8742-2D4CC9B54F2C was discovered but not loaded!!
Driver D912C7BC-F098-4367-92BA-E911083C7B0E was discovered but not loaded!!
Driver 99F03B99-98D8-49DD-A8D3-3219D0FFE41E was discovered but not loaded!!
Driver 1A7E4468-2F55-4A56-903C-01265EB7622B was discovered but not loaded!!
Driver B95E9FDA-26DE-48D2-8807-1F9107AC5E3A was discovered but not loaded!!
Driver B219E140-DFFC-11E3-B956-0022681E6906 was discovered but not loaded!!
Driver 2366C20F-E15A-11E3-8BF1-E4115B28BC50 was discovered but not loaded!!
Driver 22EA234F-E72A-11E4-91F9-28D2447C4829 was discovered but not loaded!!
Driver ECEBCB00-D9C8-11E4-AF3D-8CDCD426C973 was discovered but not loaded!!
Driver 86CDDF93-4872-4597-8AF9-A35AE4D3725F was discovered but not loaded!!
Driver A487A478-51EF-48AA-8794-7BEE2A0562F1 was discovered but not loaded!!
Driver 19618BCE-55AE-09C6-37E9-4CE04084C7A1 was discovered but not loaded!!
Driver AD416CE3-A483-45B1-94C2-4B4E4D575562 was discovered but not loaded!!
```

Inside QEMU when I run `drivers` this is what I get:
```
            T   D
D           Y C I
R           P F A
V  VERSION  E G G #D #C DRIVER NAME                         IMAGE NAME
== ======== = = = == == =================================== ==========
58 0000000A B - -  1  6 PCI Bus Driver                      PciBusDxe
5A 00000010 D - -  1  - Virtio PCI Driver                   VirtioPciDeviceDxe
5B 00000010 ? - -  -  - Virtio 1.0 PCI Driver               Virtio10
5C 00000010 ? - -  -  - Virtio Block Driver                 VirtioBlkDxe
5D 00000010 ? - -  -  - Virtio SCSI Host Driver             VirtioScsiDxe
5E 00000010 ? - -  -  - Virtio Serial Driver                VirtioSerialDxe
5F 0000000A D - -  3  - Platform Console Management Driver  ConPlatformDxe
60 0000000A D - -  3  - Platform Console Management Driver  ConPlatformDxe
61 0000000A B - -  3  3 Console Splitter Driver             ConSplitterDxe
62 0000000A ? - -  -  - Console Splitter Driver             ConSplitterDxe
[... 136 lines cut ...]
- OS(s): Ubuntu
- Tool Chain(s): GCC5
```

### Version Information

```text
Tag: edk2-stable202411
```

### Urgency

Low

### Are you going to fix this?

I will fix it

### Do you need maintainer feedback?

No maintainer feedback needed

### Anything else?

I've been struggling with this bug but I've been keeping my work in the 534021965f6f7c417610add53984f39d6945bbcf commit. I'm at a point at my exploration where I can't seem to get the TLS working in that commit so I'd like to try more recent versions of the `NetworkPkg` as I've noticed that a lot of development has been happening there but for some reason I can't get it to build in a way that QEMU will load the drivers... Any help will be highly appreciated and I really hope it's not something really obvious that I miss that's now wasting someone's time. Thanks for the attention in advance!

lgao4, 2025-02-13:
@Zclarkwilliams , could you review this issue and give your feedback?

vamposdecampos, 2025-06-21:
@AndreasBBS  thanks for the pointer to a "good" commit (as well as the "discovered but not loaded" breadcrumb).  I had tried to bisect this before, but got nowhere.  Turns out, once a bad commit is built, it will break all future incremental builds, even after checking out a good commit.  I re-bisected with nuking the `Build/` directory between steps, and found first bad commit: 4c4ceb2ceb80c42fd5545b2a4bd80321f07f4345 "NetworkPkg: SECURITY PATCH CVE-2023-45237"

vamposdecampos, 2025-06-21:
At the tip of `master` looks like `RngDxe` is almost already included, but is forced to use a `RngLib` that explicitly avoids the necessary implementation for `gEfiRngProtocolGuid`.  Worksforme like 

```diff
diff --git a/OvmfPkg/Include/Dsc/OvmfRngComponents.dsc.inc b/OvmfPkg/Include/Dsc/OvmfRngComponents.dsc.inc
index 68839a0caa..c26e846468 100644
--- a/OvmfPkg/Include/Dsc/OvmfRngComponents.dsc.inc
+++ b/OvmfPkg/Include/Dsc/OvmfRngComponents.dsc.inc
@@ -2,8 +2,5 @@
 #    SPDX-License-Identifier: BSD-2-Clause-Patent
 ##
 
-  SecurityPkg/RandomNumberGenerator/RngDxe/RngDxe.inf {
-    <LibraryClasses>
-      RngLib|MdePkg/Library/BaseRngLib/BaseRngLib.inf
-  }
+  SecurityPkg/RandomNumberGenerator/RngDxe/RngDxe.inf
   OvmfPkg/VirtioRngDxe/VirtioRng.inf
```

but I'm not sure of all the implications. Commit 712797cf19acd292bf203522a79e40e7e13d268b says the override was done for `rdrand`.

AntonioPMCS, 2025-08-04:
I believe this other issue contains valuable information to understand this problem: https://github.com/acidanthera/bugtracker/issues/2421

lersek, 2025-08-07:
I've been thinking about this problem on-and-off for a while now; I might as well write it up.

The **first problem** was CVE-2023-45237 -- predictable TCP sequence numbers. The idea for fixing it was to consume proper randomness in the network stack; hence commit 4c4ceb2ceb80 ("NetworkPkg: SECURITY PATCH CVE-2023-45237", 2024-05-24), first released in `edk2-stable202405`.

The **second problem** was that this commit (unfortunately) introduced a layering violation. The commit changed the `DxeNetLib` instance to depend on `gEfiRngProtocolGuid`, via depex. (With the idea being that the library instance would forward all requests for networking-oriented randomness to `EFI_RNG_PROTOCOL`.) Consequently, any driver linked against the `DxeNetLib` instance would inherit a dependency on `gEfiRngProtocolGuid`. Let me explain why I think this is a layering/sequencing violation:

1. `DxeNetLib` is a `NetLib` instance that is explicitly permitted for / restricted to the following module types: `DXE_CORE`, `DXE_DRIVER`, `DXE_RUNTIME_DRIVER`, `DXE_SMM_DRIVER`, `UEFI_APPLICATION`, `UEFI_DRIVER`. The blanket depex on `gEfiRngProtocolGuid` is wrong for at least three module types from those listed (and I'm assuming _at once_ that the depex does not even apply to `DXE_CORE`):
   1. `DXE_DRIVER`. According to the [UEFI spec](https://uefi.org/specs/UEFI/2.11/37_Secure_Technologies.html#random-number-generator-protocol), `EFI_RNG_PROTOCOL` *is used to provide random numbers for use in **applications**, or entropy for seeding other random number generators* (emphasis mine). Making the protocol a _dispatch pre-requisite_ for DXE drivers (i.e., platform drivers) is therefore improper, as the platform is at liberty **not** to expose the RNG protocol before **applications** may be expected to run.
   2. `UEFI_DRIVER`. As far as I understand, on a PI-based UEFI platform, `UEFI_DRIVER`s are supposed to have a fixed, implicitly set depex on all the UEFI architectural protocols together, _and on nothing else_. A UEFI driver may of course consume other protocols, regardless of whether it follows the UEFI driver model, but placing such protocols in the depex is dubious.
   3. `UEFI_APPLICATION`. Same argument applies as to `UEFI_DRIVER`, only more strongly (if possible). There's no way an application shouldn't be _loadable_ because a protocol is missing.
2. In commit 4c4ceb2ceb80, `TcpDxe` -- a `UEFI_DRIVER` --  received an *additional* depex explicitly, namely on `gEfiHash2ServiceBindingProtocolGuid`, for no discernible reason. That again breaks the principle that `UEFI_DRIVER` modules should depend exactly on the architectural protocols.
3. The networking drivers modified by the commit: `TcpDxe`, `IScsiDxe`, `Ip6Dxe`, `Dhcp4Dxe`, `Dhcp6Dxe`, `DnsDxe`, `HttpBootDxe`, `Ip4Dxe`, `Udp4Dxe`, `Udp6Dxe`, and `UefiPxeBcDxe`, are *all* `UEFI_DRIVER` modules that are supposed to follow the UEFI driver model. Therefore, they should not depend on any protocol except the architectural ones via (hidden, implicit) depex. The `DxeNetLib` change made them *unloadable* on all platforms that lacked an `EFI_RNG_PROTOCOL` provider. Note that the UEFI spec [does not unconditionally require](https://uefi.org/specs/UEFI/2.11/02_Overview.html#platform-specific-elements) an instance of this protocol: *If a platform includes the ability to use a hardware feature to create high quality random numbers, this capability should be exposed by instance of `EFI_RNG_PROTOCOL` with at least one EFI RNG Algorithm supported*. The same section also notes, *If a platform supports UEFI general purpose network applications, then [a bunch of networking protocols are required]*. The problem is that the edk2 change does not allow a platform to satisfy one optional requirement (networking) without also satisfying the other optional requirement (rng).

Instead, the networking drivers should have been rendered dependent on secure randomness as follows:
1. `PseudoRandom()` should only cleanly propagate `EFI_NOT_FOUND` from `LocateProtocol()` outwards, and not assert that `EFI_RNG_PROTOCOL` is found.
2. Either state in the UEFI spec that networking services require secure randomness (= RNG protocol), or enable `PseudoRandom()` to fall back to weak randomness if a Feature PCD is set to a non-default override value. The first alternative allows for a strict-only implementation; the second alternative allows platforms, for sake of compatibility, to opt-in to weak security.
3. No `[depex]` additions anywhere.
4. `UEFI_DRIVER` modules should never call `PseudoRandom()` or `PseudoRandomU32()` in their entry point functions -- examples: `TcpDriverEntryPoint`, `Udp4DriverEntryPoint`, `Udp6DriverEntryPoint` --; instead, they should call these APIs -- i.e., consume the RNG protocol -- right when they need randomness. At the earliest, call these APIs in `DriverBindingStart`, or (in service binding protocol implementations) in `CreateChild`. Failures should be propagated cleanly all the way up.
5. Make the platform responsible for exposing the RNG protocol *in time*:
   - A platform can satisfy this by producing the RNG protocol in a **DXE driver** (example: `SecurityPkg/RandomNumberGenerator/RngDxe/RngDxe.inf`). Such a DXE driver installs the RNG protocol in its entry point function (= during the DXE phase), whereas said networking drivers, being UEFI drivers following the UEFI driver model, bind devices no earlier than Platform BDS connects controllers (= during the BDS phase). This prov [... cut ...]
Interfering with depexes / dispatch order of UEFI drivers is both unnecessary and in violation of Platform Init principles.

... Commit 712797cf19ac ("OvmfPkg: wire up RngDxe", 2024-06-13) should also be reverted; there's no need to implement the RNG protocol with an `RDRAND`-based DXE driver in OVMF. First, `RDRAND` is widely untrusted, to my knowledge, as an *exclusive* source of entropy; second, `VirtioRngDxe` is sufficient, once the faulty depex is remedied.

@kraxel @osteffenrh This is just my opinion; thanks if you have read it till the end.

lersek, 2025-08-07:
@osteffenrh 

> About RDRAND: I fear we need a way to fall back to it; it might not be able to provide a virtio-rng device in all cases.

I certainly agree that a "dumb" platform device that can produce secure randomness, without complex enumeration, is great; it's the most flexible solution. For example, Platform Init specifies `RNG_PPI` (see `MdePkg/Include/Ppi/Rng.h`) which is semantically identical to the RNG protocol. In the PEI phase, you'd mostly want to / have to rely on a platform device, if you wanted to expose secure randomness. It's just that `RDRAND` is opaque.