tianocore/edk2 #10719: [Bug]: distutils is deprecated

makubacki, 2025-02-04:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [ ] Firmware
- [x] Tool
- [ ] Unit Test

### Code first?

- [ ] Yes

### What packages are impacted?

BaseTools

### Current Behavior

The `pythonx.x-distutils` package provided the standard `distutils` module.

We are using an Ubuntu 22.04 container and found `distutils` is deprecated. In the deadsnakes PPA python3.12.7-1+jammy1 to python3.12.8-1+jammy1 update on 12/04/2024, the following diff was made:

https://launchpadlibrarian.net/761821028/python3.12_3.12.7-1+jammy1_3.12.8-1+jammy1.diff.gz

In that, `python3.12-distutils` is no longer a built package per [PEP-0632](https://peps.python.org/pep-0632/).

---

The only code found to be using `distutils` in edk2 is:

- `PyEfiCompressor`
  - BaseTools/Source/C/PyEfiCompressor/setup.py
- `Brotli`
  - BaseTools/Source/C\BrotliCompres/brotli/python/tests/_test_utils.py
  - MdeModulePkg/Library/BrotliCustomDecompressLib/brotli/setup.py

### Expected Behavior

The codebase needs to no longer depend on `distutils`.

Migration advice is here: https://peps.python.org/pep-0632/#migration-advice

### Steps To Reproduce

Search the edk2 codebase `*.py` files for `distutils`.

### Build Environment

```markdown
- OS(s): Ubuntu 22.04
- python3.12.8-1+jammy1
```

### Version Information

```text
Commit: 258f2d1563c366bf73932f80737776baa71b7ab5 (master)
```

### Urgency

Medium

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

Maintainer feedback requested

makubacki, 2025-02-04:
@bexcran & @mdkinney - It looks like [BaseTool/Source/C/PyEfiCompressor/setup.py](https://github.com/tianocore/edk2/blob/master/BaseTools/Source/C/PyEfiCompressor/setup.py) has not been used in a while. Do you know its current status?

lgao4, 2025-02-13:
@YuweiChen1110 , please help check this issue.