tianocore/edk2 #10723: [Bug]: ShellPkg: ARMV9 support should be added to SmbioView command based on Type 4 processor information.

praveensankarn, 2025-02-05:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### Code first?

- [ ] Yes

### What packages are impacted?

ShellPkg

### Current Behavior

If the processor family is ArmV9 based processor, then shell shows "**_Undefined processor family and type_**" against the field "**_Processor Family 2 :_** ", whenever the below command is passed in shell.
smbiosview -t 4.

### Expected Behavior

The processor family should be shown as ARMV9 while executing "smbiosview -t 4" command in shell.

### Steps To Reproduce

1. Build the code.
2. Flash the image and boot to shell
3. Execute smbiosview -t 4 command in shell prompt.
4. See the error "Undefined processor family" against Processor Family 2 : field.

### Build Environment

```markdown
- OS(s):Windows or Ubuntu
- Tool Chain(s):GCC5 or VS2019 or VS2015.
```

### Version Information

```text
edk2-stable202411
```

### Urgency

Medium

### Are you going to fix this?

I will fix it

### Do you need maintainer feedback?

Maintainer feedback requested

### Anything else?

Hi @lgao4 @mdkinney @pierregondois  
      Added switch case for ArmV9 processor family inside function DisplayProcessorFamily2 in PrintInfo.c to avoid inappropriate data shown for ArmV9 processors. ArmV9 processor family is added in Smbios 3.8 specification, which should be noted as well. I will push the local branch here to review the changes.

Thanks,
Praveen Sankar

praveensankarn, 2025-02-05:
Hi @lgao4 @mdkinney, 
     I got permission denied error while pushing local branch (which has changes made for the fix). Could you please help me with this ?

Thanks,
Praveen Sankar

pierregondois, 2025-02-05:
Hello Praveen,
if you are trying to push a branch on the tianocore edk2 github repository, I think this is normal to have a permission denied error. You should create a branch on a fork of the edk2 repository and then make a pull request of that branch on the tianocore edk2 repository.
For more information, the following page should help: https://github.com/tianocore/tianocore.github.io/wiki/EDK-II-Development-Process

praveensankarn, 2025-02-06:
Hi @pierregondois,
     Created the pull requested, as you said. URL for PR has been given below for your reference.
[https://github.com/tianocore/edk2/pull/10726](url)

Thanks,
Praveen Sankar

pierregondois, 2025-06-11:
The PR is now merged

closed 2025-06-11 by pierregondois