tianocore/edk2 #10727: [Bug]: <Unclosed BlockIo2 Protocol during DiskIoDriverBindingStart Error Handling>

kanagavels97, 2025-02-06:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### Code first?

- [ ] Yes

### What packages are impacted?

MdeModulePkg

### Current Behavior

During the DiskIoDriverBindingStart, both the BlockIo and BlockIo2 protocols are opened for the specified controller handle. The Instance->SharedWorkingBuffer is then allocated based on the BlockSize. If the BlockSize is zero, indicating that the handle does not support Read/Write operations, the buffer will not be allocated. A NULL check is performed on the SharedWorkingBuffer, and if it is NULL, the process jumps to the ErrorExit. In the ErrorExit section, only the BlockIo protocol is closed, while the BlockIo2 protocol remains open.

We are facing an issue with Uninstalling the BlockIo2 protocol during the DriverBindingStop process because it remains open.

**MdeModulePkg\Universal\DiskIoDxe\DiskIo.c**

_ErrorExit:
  if (EFI_ERROR (Status)) {
    if ((Instance != NULL) && (Instance->SharedWorkingBuffer != NULL)) {
      FreeAlignedPages (
        Instance->SharedWorkingBuffer,
        EFI_SIZE_TO_PAGES (PcdGet32 (PcdDiskIoDataBufferBlockNum) * Instance->BlockIo->Media->BlockSize)
        );
    }
    if (Instance != NULL) {
      FreePool (Instance);
    }
    gBS->CloseProtocol (
           ControllerHandle,
           &gEfiBlockIoProtocolGuid,
           This->DriverBindingHandle,
           ControllerHandle
           );
  }_

### Expected Behavior

The check needs to be added for the BlockIo2 protocol in the ErrorExit, and it should be closed properly if the BlockIo2 protocol is opened.

### Steps To Reproduce

1. Connect a UFS device that supports WLUN. This WLUN will be detected as a separate handle under the UFS host controller.
2. In my case, the UFS device supports WLUN (44h), which is used for the RPMB feature. According to the UFS specification, the RPMB 
    LUN does not support Read/Write operations.
3. Add UFS module to the bios
4. Flash the bios on SUT and boot to the shell.
5. Run the devtree command to identify the controller handle numbers.
6. In my case, the UFS Host controller number is 165.
7. Run the disconnect 165 command.
8. The command returns "Result Not Found." Upon analyzing the issue, we discovered that the BlockIo2 protocol is opened for the 
    UFS WLUN handle and not closed.

### Build Environment

```markdown
Shell environment
```

### Version Information

```text
Tag: edk2-stable202411
```

### Urgency

Medium

### Are you going to fix this?

I will fix it

### Do you need maintainer feedback?

Maintainer feedback requested

### Anything else?

Hi @niruiyu 

Please verify the issue on your end and provide confirmation.
I have a fix ready and will create a PR with the necessary changes.

CC: @lgao4 @mdkinney 

Thanks,
Kanagavel

kanagavels97, 2025-03-04:
Hi @niruiyu

Is there any update on this issue?

CC: @lgao4 @mdkinney 

Thanks,
Kanagavel

mdkinney, 2025-03-04:
I have left a comment in the PR.  I think some additional code changes are required.

kanagavels97, 2025-03-25:
Hi @mdkinney 

I have a UFS with LUN 0 enabled and a WLUN 44h supported device. According to the JEDEC UFS specification, the WLUN 44h does not support read/write operations, and its BlockSize is 0.

This issue can be resolved by restricting the installation of the BlockIo and BlockIo2 protocols on handles that do not support read/write operations or have a block size of 0.

Could you explain why the BlockIo and BlockIo2 protocols are installed on handles that do not support read/write operations or have a BlockSize of 0?

Thanks,
Kanagavel

kanagavels97, 2025-04-21:
Hi @mdkinney 

Could you please provide your suggestion regarding my previous comment?

Thanks,
Kanagavel

mdkinney, 2025-04-21:
I do not think we expected to see devices with those attributes.  CD devices are RO, so devices with only read access with a non-zero block size are supported.

I think it would be good for drivers that produce the Block I/O protocols to check if the target supports a minimum of read operations and a block size > 0.  If a device does not support reads or has a block size of 0, then the device should be ignored and no Block I/O protocols should be produced because the device is not bootable.

One challenge may be a Block I/O device that requires configuration to support a minimum of read operations with a non-zero block size.  If that type of device exists, then a handle with Block I/O may be required for the configuration operation to be performed.  If that condition does not exist, then adding the check above would be appropriate.