tianocore/edk2 #10733: [Feature]: Additional Stack Cookie Features

os-d, 2025-02-07:
### Feature Overview

This feature issue is being used to track additional stack cookie work.

### Solution Overview

- AARCH64 Standalone MM Core Dynamic Cookie Support
- Ensure calling convention from ASM code to C code is EFIAPI, document it in comments what the expectation is

### What packages are impacted?

Other

### Urgency

Medium

### Are you going to implement the feature request?

I will implement the feature

### Do you need maintainer feedback?

No maintainer feedback needed

### Code first?

- [ ] Yes

lgao4, 2025-02-13:
@os-d , please take it.

os-d, 2025-02-13:
Thanks, I had meant to assign myself