tianocore/edk2 #10765: [Bug]: After edk2-ovmf update(s), VM boots slowly and spends a lot of time in edk2, until a NVRAM reset

adam900710, 2025-02-14:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### Code first?

- [ ] Yes

### What packages are impacted?

OvmfPkg

### Which targets are impacted by this bug?

RELEASE

### Current Behavior

The original VM booted pretty fast, spent less than 2s in the firmware, then into the bootloader (systemd-boot).

But after one or more updates of edk2-ovmf (x86_64), the same VM boots very slowly, spends over 5s in the firmware then into the bootloader.

After a NVRAM reset, the slow boot is completely gone.

### Expected Behavior

Even after several updates of the edk2-ovmf package, the VM should boot as fast as the initial boot, without the need to reset the NVRAM file.

### Steps To Reproduce

1. Setup a libvirt with qemu, kvm, and native edk2-ovmf firmware (release, no debug output)
2. Install a VM, record the time spent the firmware.
3. Update the edk2-ovmf several times, and after each update, boot the VM at least once
4. Record the time spent in the firmware after all the edk2-ovmf updates

### Build Environment

```markdown
- OS(s): Archlinux
- Tool Chain(s): GCC5 according to the PKGBUILD (https://gitlab.archlinux.org/archlinux/packaging/packages/edk2/-/blob/main/PKGBUILD?ref_type=heads)
```

### Version Information

```text
Tag: edk2-stable202411
```

### Urgency

Low

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

No maintainer feedback needed

lgao4, 2025-03-13:
@ard,  @kraxel , could you help look this issue?

wanglmb, 2025-04-07:
https://github.com/tianocore/edk2/discussions/6518

kraxel, 2025-04-07:
Most likely this is simply writing to emulated flash being slow.  It's a vmexit (round-trip to the host) for every mmio write operation, so flash block updates are quite costy in VMs.

adam900710, 2025-04-08:
> Most likely this is simply writing to emulated flash being slow. It's a vmexit (round-trip to the host) for every mmio write operation, so flash block updates are quite costy in VMs.

I'm also wondering under which cases we need so many flash updates?

Is it related to the way how libvirt uses it (template with local copy? and the extra readonly loader?)
If it's related, is there any workaround?

kraxel, 2025-04-09:
> I'm also wondering under which cases we need so many flash updates?

One offender is iscsidxe, which got a runtime switch in 2025-02 stable tag (off by default).

> Is it related to the way how libvirt uses it (template with local copy? and the extra readonly loader?) If it's related, is there any workaround?

qemu got support for managing efi variables in 10.0 (will be released in a week or so).
edk2 side of that is https://github.com/tianocore/edk2/pull/10695 which hopefully lands in 2025-05 stable tag.
With that in place variable reads + writes are down to one or two context switches to qemu.

wanglmb, 2025-04-16:
> qemu got support for managing efi variables in 10.0 (will be released in a week or so). edk2 side of that is [#10695](https://github.com/tianocore/edk2/pull/10695) which hopefully lands in 2025-05 [...]

I am trying test with your edk2 branch [kraxel:devel/secure-boot-external-vars] and qemu-10.0.0-rc3, but my VM can't boot . seems it run into  qemu_main_loop and cpu utils of qemu-system-aarch64  is 100% 

I build edk2 with -D QEMU_PV_VARS=TRUE and start qemu with -device uefi-vars-sysbus,jsonfile=~/vars.json
Is there anything else I am missing ?   

following list how I built edk2:
-----------------------------------
$ git clone --recurse-submodule  https://github.com/kraxel/edk2.git kraxel
$ git checkout  devel/secure-boot-external-vars
$ make -C BaseTools -j128
$ . edksetup.sh
$ build -t GCC5 -n 128 -b DEBUG -a AARCH64 -p ArmVirtPkg/ArmVirtQemu.dsc --cmd-len=65536 -D NETWORK_IP6_ENABLE -D QEMU_PV_VARS=TRUE

following is how I sart qemu:
-------------------------------
$ qemu-system-aarch64 --version
QEMU emulator version 9.2.93 (v10.0.0-rc3)
Copyright (c) 2003-2025 Fabrice Bellard and the QEMU Project developers

$ qemu-system-aarch64 \
  -machine virt -accel kvm \
  -cpu host -m 8192 -smp 4 \
  -cdrom /home/iso/linux_aarch64.iso -boot order=c \
  -vnc :0 -daemonize\
  -device uefi-vars-sysbus,jsonfile=/home/vars.json \
  -bios /usr/share/edk2/aarch64/QEMU_EFI.fd \
  -device virtio-gpu-pci \
  -device qemu-xhci \
  -device usb-kbd \
  -device usb-mouse

kraxel, 2025-04-23:
Looks sane on a quick glance.  Any output on the serial console?

wanglmb, 2025-04-24:
> Looks sane on a quick glance. Any output on the serial console?
There's not and console output.

wanglmb, 2025-04-24:
> Looks sane on a quick glance. Any output on the serial console?

I built edk2 with DEBUG option, but there is not any console output.

wanglmb, 2025-04-24:
> Looks sane on a quick glance. Any output on the serial console?

the searial log stay at 
```
[Bds]BdsWait ...Zzzzzzzzzzzz...
[Bds]BdsWait(5)..Zzzz...
```

There are also several Error reported such as:
```
 "Error: Image at 0023F5D2000 start failed: 00000001"
```
See the full console output:  [console out](https://gist.github.com/xTire/c688f2a7b2e8d5a83aabc4bc7fcb6c55#file-gistfile1-txt)

kraxel, 2025-04-24:
> ```
> [Bds]BdsWait ...Zzzzzzzzzzzz...
> [Bds]BdsWait(5)..Zzzz...
> ```

Hmm, that doesn't look variable store related.  Does this only happen with `QEMU_PV_VARS=TRUE`?
The PR landed yesterday btw, so you can simply use the master branch now.

libvirt support is TBD, for now you have to use this...
```
  <qemu:commandline>
    <qemu:arg value='-device'/>
    <qemu:arg value='uefi-vars-x64,jsonfile=/home/kraxel/libvirt/qemuvars/fedora.json'/>
  </qemu:commandline>
```
... to add custom args to the qemu command line.

wanglmb, 2025-04-24:
> Hmm, that doesn't look variable store related. Does this only happen with `QEMU_PV_VARS=TRUE`? The PR landed yesterday btw, so you can simply use the master branch now.
>
> libvirt support is TBD, for [...]

Great!  
It works well after updating master branch.
Thank you very much~

wanglmb, 2025-06-05:
@kraxel  Is it possible to boot