tianocore/edk2 #10771: [Bug]: DXE core memory protection feature might cause inconsistency between GCD internal data structure with CPU page table

qhuang8, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### Code first?

- [ ] Yes

### What packages are impacted?

MdeModulePkg

### Which targets are impacted by this bug?

DEBUG, RELEASE

### Current Behavior

When we enable DXE core memory protection feature, if we set some GCD irrelevent attributes like EFI_MEMORY_SP or EFI_MEMORY_CPU_CRYPTO via GCD SetMemorySpaceAttributes (), it will cause page fault except.

### Expected Behavior

We can still invoke SetMemorySpaceAttributes () successfully to **append** EFI_MEMORY_SP or EFI_MEMORY_CPU_CRYPTO 

### Steps To Reproduce

1. Set the following two platform PCD to enable memory protection.
PcdSetNxForStack| TRUE
PcdDxeNxMemoryProtectionPolicy| 0x7FD5

2. Invoke GCD SetMemorySpaceAttributes () API in later DXE phase to append EFI_MEMORY_SP or EFI_MEMORY_CPU_CRYPTO (keep the original attributes) 
3. System will have page fault exception

### Build Environment

```markdown
- OS(s):Windows 11
- Tool Chain(s): VS2019
```

### Version Information

```text
Commit: f979f51157c5560c658537ea2a488b7a8439e463
```

### Urgency

High

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

No maintainer feedback needed

### Anything else?

The root cause of this issue is that Edk2\MdeModulePkg\Core\Dxe\Misc\MemoryProtection.c (line 1294) directly uses CPU arch protocol to clear XP in page table when loading DXE driver for execution, but it fails to update the GCD internal data structure according (the memory range still makes as XP attribute):
gCpu->SetMemoryAttributes (gCpu, Memory, Length, NewAttributes);

Later invoking of GCD SetMemorySpaceAttributes () will incorrectly re-apply the incorrect XP attribute and make it take effect in the final page table. When CPU executes the memory, it triggers the page fault exception.

makubacki, 2025-02-20:
@os-d, you've looked at this recently. Would you like to follow up?

os-d, 2025-02-24:
I agree this looks like a correct analysis of the bug. I can take a look at this when I get a chance. In general, the GCD gets very out of sync with the page table, leading to issues like this. I have a lot of fixes and design changes to upstream that fix a number of these issues that I will be working to upstream now that the hard freeze is lifted.

os-d, 2025-02-24:
@qhuang8 I was actually unable to repro this on OVMF IA32X64. I updated the PlatformDxe driver to add `EFI_MEMORY_SP` to the region of CpuDxe covering the `CpuSetMemoryAttributes` function and keep the other GCD attributes. However, the GCD correctly showed this page was mapped `EFI_MEMORY_RO` and the driver was able to set SP and CpuDxe could continue to execute.

Can you share a repro on OVMF or ArmVirtPkg? Or a bit more clearly describe the chain of calls you believe is the issue (the line number you cite doesn't exist in that file, I'm guessing you mean 1109?

qhuang8, 2025-02-25:
Sorry that I am not familiar with OVMF or ArmVirtPkg... 

Our use case of GCD SetMemoryAttribute happens in very later DXE phase. We set `EFI_MEMORY_CPU_CRYPTO` to all availabe DDR memory *after* PCI enumeration has completed. Before that, I see DXE core has updated CPU page table to drop XP attribute for some code region, but the correspondent GCD internal data structure has not been updated.

os-d, 2025-02-25:
@qhuang8 can you share what driver corresponds to that code region? Like I said, when I tried updating a code region of CpuDxe, it had the correct attributes set in the GCD. Perhaps your case is a driver loaded earlier? Or DXE Core itself? I can update my repro attempt accordingly.

qhuang8, 2025-02-26:
In my platform it is Edk2\NetworkPkg\DnsDxe\DnsDxe.inf driver.

After this driver is loaded, its memory attributes in cpu page table has been changed to drop XP

lgao4, 2025-04-10:
@os-d , have you any other input for this problem?

qhuang8, 2025-04-10:
@os-d can you reproduce this issue locally with real platform or ovmf?

os-d, 2025-04-14:
@lgao4 @qhuang8 , I am unable to reproduce this issue. If you are able to share a repro on OVMF, I'm happy to debug further. Otherwise I am still working towards upstreaming some memory management improvements that should help fix issues like this, but I don't have a timeline for that.

qhuang8, 2025-04-15:
@os-d I have figured out how to reproduce this issue on open source OVMF X64 platform. 

Step1. Apply the attached patch to the latest EDK2 trunk (mine is 0f1c0d2d2d9e567746ce4fedea23f833fa0d3abd)
[0001-update-to-add-gcd-in-later-dxe-phase-to-trigger-cpu-.patch](https://github.com/user-attachments/files/19752282/0001-update-to-add-gcd-in-later-dxe-phase-to-trigger-cpu-.patch)

Step2. Build OVMF X64 platform with serial log turned on. 
I am using VS2019  `build -t VS2019 -a X64 -p OvmfPkg\OvmfPkgX64.dsc -D DEBUG_ON_SERIAL_PORT`

Step3. Run qemu with the generated FD image.  It will trigger the cpu exception. See my local run log file.
[ovmf_Apply_Patch.log](https://github.com/user-attachments/files/19752297/ovmf_Apply_Patch.log)

If we try to run without UpdateMemoryMap() in my patch. Everything goes smoothly. 
[image]
See log file below
[ovmf_Apply_Patch_Without_UpdateMemoryMap.log](https://github.com/user-attachments/files/19752314/ovmf_Apply_Patch_Without_UpdateMemoryMap.log)

UpdateMemoryMap() function simulates what we are trying to do in our real platform. It just tries to append `EFI_MEMORY_CPU_CRYPTO `to all available memory and it should not impact hardware configure.

os-d, 2025-04-15:
@qhuang8 @lgao4 , I was able to repro the issue and posted a PR to resolve it here: https://github.com/tianocore/edk2/pull/10967. @qhuang8 can you please verify this solves your issue?

qhuang8, 2025-04-16:
@os-d Actually, I tried this fix, but it will hang in much eariler phase in my platform....

os-d, 2025-04-16:
@qhuang8 thanks for testing, can you pull the latest of my branch in the PR and try again? I did not add the capabilities if required, which can lead to an assert depending on platform, but I have now updated this.

qhuang8, 2025-04-17:
@os-d I tried the updated PR, but I still get the exception when setting the attributes via GCD at the same invokingin the real platform...

Even if I apply the core change in OVMF platform, it can still reproduce the cpu exception. Can you confirm?  Attach the OVMF log file.

[ovmf_Apply_Patch_WithPR.log](https://github.com/user-attachments/files/19790054/ovmf_Apply_Patch_WithPR.log)

os-d, 2025-04-17:
@qhuang8 I have resolved the issue. I had a typo and was missing one case where there are no cache attributes set, so we still need to call `gCpu->SetMemoryAttributes()` ourselves after updating the GCD. I now have the CI cases passing and your repro is passing for me.

Here is my passing log with your patch: 

[success.log](https://github.com/user-attachments/files/19800969/success.log)

I will confirm that all the CI is now passing and if so, would you try again on your side?

qhuang8, 2025-04-18:
@os-d  thanks for the update. It is strange that I can still reproduce this exception on my real platform. I tried appied the change in DXE core and tried in OVMF platform. Still no luck...

I even tried to download your repo with the fix and applied my patch. 

[image]

Here attach the log for your reference

[ovmf.log](https://github.com/user-attachments/files/19810287/ovmf.log)

os-d, 2025-04-18:
@qhuang8 hmm, thanks for trying again.

Ah, I am able to reproduce your issue. With VS2022 (my normal toolchain and what CI uses) this works just fine. However, with VS2019, it is broken. I suspect this is just because the pattern of allocations changes ever so slightly with the toolchain difference, but I will debug and see. I would be curious if you build with VS2022 if you still see the issue on your physical platform (presumably on OVMF you would repro my success), but let me root cause why VS2019 is broken.

os-d, 2025-04-18:
@qhuang8 , okay, the reason VS2022 was working with my changes and VS2019 was not is being VS2019 in tools_def.template is not configured to set image section alignment to 4k. So this bails out of trying to apply image memory protections and never hit my new code which resolves the issue when image sections are 4k aligned. So, in a way its good you were using VS2019 and I was using VS2022, because it allowed me to cover both cases.

I have updated my PR to set the GCD to reflect RWX in the case where image memory protections cannot be applied (when image section alignment is not a multiple of 4k). With this change, now both VS2022 and VS2019 boot with your changes.

Here is the VS2019 log: [2019_success.log](https://github.com/user-attachments/files/19815940/2019_success.log)

VS2019's configuration could be updated to 4k align images but I don't think this is worthwhile, it is not run in CI and so any issues changing that would not be caught. I would encourage you to move to VS2022 so that you can apply image memory protections.

Hopefully now you can recreate the success boot in OVMF and it works on your physical platform, but please let me know what happens :)

qhuang8, 2025-04-21:
@os-d thanks for the update. I tried the latest PR can confim that it can address the issue in OVMF in VS2019. However, I cannot validate my real platform as its edk2 core is based on **release/edk2_stable_202402** and the source code fragment below

[image]
conflicts with source code in that snaphsot.  Can you please provide a hot fix in that stable tag so that I can try on the real platform?

os-d, 2025-04-21:
@qhuang8 thanks for confirming on OVMF! I have posted a branch here that is edk2-stable202402 with my commits cherry-picked on top: https://github.com/os-d/edk2/tree/2402_with_gcd_sync

Can you please give that a try on your physical platform? It works with VS2019 on OVMF and your patch: 

[2019_success_202402.log](https://github.com/user-attachments/files/19837326/2019_success_202402.log)

qhuang8, 2025-04-22:
@os-d  thanks for providing a patch on 202402 snapshot. I tried this patch in my real platform, but it hangs in the entrypoint of CPU DXE driver. i have not yet have the extra bandwidth to dig out the root cause.  But it should not prevent you to close this open source   sighting as it has confirmed to address the issue I reported in OVMF.

PR #10967 (2025-04-27): MdeModulePkg: DxeCore: Set Image Memory Protections Through GCD
# Description

Today, SetUefiImageMemoryAttributes calls directly to the CPU Arch protocol to set EFI_MEMORY_XP or EFI_MEMORY_RO on image memory. However, this bypasses the GCD and so the GCD is out of sync with the actual state of memory.

This can cause an issue in the scenario where a new attribute is being set (whether a virtual attribute or a real HW attribute), if the GCD attributes are queried for a region and the new attribute is appended to the existing GCD attributes (which are incorrect), then the incorrect attributes c [... cut]
closed 2025-04-27 by mergify[bot]