tianocore/edk2 #10812: [Feature]:  Add function for extract ACPI table from ACPI Silicon HOB.

GeorgeLiao012, 2025-03-03:
### Feature Overview

Add function for extract ACPI table from ACPI Silicon HOB. 

### Solution Overview

1. Got RSDP table which installed during the FSP phase from ACPI Silicon Hob, then pass it to the DXE AcpiTableInstance.
2. Got XSDT from RSDP and extract necessary data from XSDT, according to the old XSDT to Initialize a new XSDT. 
3.Re-install ACPI table from old XSDT to the new XSDT.
    a. If ACPI silicon Hob has DSDT table then re-install DSDT table in the new XSDT. if not, then skip it.
    b. If ACPI silicon Hob has FACS table then re-install FACS table in the new XSDT. if not, then skip it. 

### What packages are impacted?

MdeModulePkg

### Urgency

Low

### Are you going to implement the feature request?

I will implement the feature

### Do you need maintainer feedback?

No maintainer feedback needed

### Code first?

- [x] Yes

lgao4, 2025-03-13:
@GeorgeLiao012 , please work on it.

GeorgeLiao012, 2025-03-13:
> [@GeorgeLiao012](https://github.com/GeorgeLiao012) , please work on it.

I have completed draft version, once I finish whole verification I will send PR soon.

mdkinney, 2025-04-02:
This does not look like a code-first issue.  This looks like a new feature in the edk2 implementation.

GeorgeLiao012, 2025-04-07:
> This does not look like a code-first issue. This looks like a new feature in the edk2 implementation.
It is a new feature, I have created a PR : https://github.com/tianocore/edk2/pull/10904
Because of I don't familiar the EDKII processing, not sure the each steps I did is it correct ?