tianocore/edk2 #10834: [Bug]: compile edk2 on Ubuntu 24.04 using stable/202011 fails due to

chaos-prevails, 2025-03-09:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

BaseTools

### Current Behavior

Hello,

I try to compile the i915ovmfPkg (https://github.com/x78x79x82x79/i915ovmfPkg) which requires edk2 and edk2-platforms. I have a build error within edk2 (still, please redirect me to the correct issue tracker if this is not the correct one)

I have installed the following packages (also according to the old wiki, see below):
```
ii automake 1:1.16.5-1.3ubuntu1
ii build-essential 12.10ubuntu1
ii clang 1:18.0-59~exp2
ii clang-18 1:18.1.3-1ubuntu1
ii cmake 3.28.3-1build7
ii cmake-data 3.28.3-1build7
ii gcc 4:13.2.0-7ubuntu1
ii gcc-13 13.3.0-6ubuntu2~24.04
ii gcc-13-base:amd64 13.3.0-6ubuntu2~24.04
ii gcc-13-x86-64-linux-gnu 13.3.0-6ubuntu2~24.04
ii gcc-14-base:amd64 14.2.0-4ubuntu2~24.04
ii gcc-x86-64-linux-gnu 4:13.2.0-7ubuntu1
ii git 1:2.43.0-1ubuntu7.2
ii git-man 1:2.43.0-1ubuntu7.2
ii nasm 2.16.01-1build1
ii python-is-python3 3.11.4-1
ii python3-pip 24.0+dfsg-1ubuntu1.1
ii python3-setuptools 68.1.2-2ubuntu1.1
ii uuid-dev:amd64 2.39.3-9ubuntu6.2
```

As (partly) documented, I:

1. clone side-by-side i915ovmfPkg, edk2 and edk2-platforms
2. according to old wiki (https://github.com/patmagauran/i915ovmfPkg/wiki/Compiling) I then change
3. edk2 branch to stable/202011 (because I think BaseMemEncryptSevLib.inf is changed to DxeMemEncryptSevLib.inf afterwards, see https://github.com/tianocore/edk2/pull/1317) 
4. update all submodules in edk2 and edk2-platforms
5. change ./i915ovmfPkg/config according to my environment
6. set `TARGET_ARCH` in ./edk2/Conf/target.txt to X64
7. try to build it with ./i915ovmfPkg/build.sh

However, the building fails with:
```
root@coss:/home/gen/i915dev/i915ovmfPkg# ./build.sh 
Loading previous configuration from /home/gen/i915dev/edk2/Conf/BuildEnv.sh
WORKSPACE: /home/gen/i915dev
EDK_TOOLS_PATH: /home/gen/i915dev/edk2/BaseTools
CONF_PATH: /home/gen/i915dev/edk2/Conf
make: Entering directory '/home/gen/i915dev/edk2/BaseTools'
make -C Source/C
make[1]: Entering directory '/home/gen/i915dev/edk2/BaseTools/Source/C'
Attempting to detect HOST_ARCH from 'uname -m': x86_64
Detected HOST_ARCH of X64 using uname.
mkdir -p .
make -C Common
make[2]: Entering directory '/home/gen/i915dev/edk2/BaseTools/Source/C/Common'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/gen/i915dev/edk2/BaseTools/Source/C/Common'
[... 287 lines cut ...]
ii git-man 1:2.43.0-1ubuntu7.2
ii nasm 2.16.01-1build1
ii python-is-python3 3.11.4-1
ii python3-pip 24.0+dfsg-1ubuntu1.1
ii python3-setuptools 68.1.2-2ubuntu1.1
ii uuid-dev:amd64 2.39.3-9ubuntu6.2
```

### Version Information

```text
stable/202011
```

### Urgency

Low

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

Maintainer feedback requested

lgao4, 2025-03-13:
please check submodule pull correctly, and please give gcc version to be used for build the tools? can you try the latest edk2 version?

chaos-prevails, 2025-03-15:
hello,

thanks:

gcc: 
```
gcc --version
gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I tried to re-check out submodules with `git submodule update --init` in ./edk2

I still get the same error:
```
root@coss:/home/gen/i915dev/i915ovmfPkg# ./build.sh 
Loading previous configuration from /home/gen/i915dev/edk2/Conf/BuildEnv.sh
WORKSPACE: /home/gen/i915dev
EDK_TOOLS_PATH: /home/gen/i915dev/edk2/BaseTools
CONF_PATH: /home/gen/i915dev/edk2/Conf
make: Entering directory '/home/gen/i915dev/edk2/BaseTools'
make -C Source/C
make[1]: Entering directory '/home/gen/i915dev/edk2/BaseTools/Source/C'
Attempting to detect HOST_ARCH from 'uname -m': x86_64
Detected HOST_ARCH of X64 using uname.
mkdir -p .
make -C Common
make[2]: Entering directory '/home/gen/i915dev/edk2/BaseTools/Source/C/Common'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/gen/i915dev/edk2/BaseTools/Source/C/Common'
make -C BrotliCompress
make[2]: Entering directory '/home/gen/i915dev/edk2/BaseTools/Source/C/BrotliCompress'
make[2]: *** No rule to make target 'brotli/c/include/brotli/shared_dictionary.h', needed by 'BrotliCompress.o'.  Stop.
make[2]: Leaving directory '/home/gen/i915dev/edk2/BaseTools/Source/C/BrotliCompress'
make[1]: *** [GNUmakefile:77: BrotliCompress] Error 2
make[1]: Leaving directory '/home/gen/i915dev/edk2/BaseTools/Source/C'
make: *** [GNUmakefile:19: Source/C] Error 2
make: Leaving directory '/home/gen/i915dev/edk2/BaseTools'
Build environment: Linux-6.8.0-54-generic-x86_64-with-glibc2.39
Build start time: 13:15:56, Mar.15 2025

WORKSPACE        = /home/gen/i915dev
PACKAGES_PATH    = /home/gen/i915dev/edk2:/home/gen/i915dev/edk2-platforms
EDK_TOOLS_PATH   = /home/gen/i915dev/edk2/BaseTools
CONF_PATH        = /home/gen/i915dev/edk2/Conf
PYTHON_COMMAND   = /usr/bin/python3

Processing meta-data .
Architecture(s)  = X64
Build target     = DEBUG
Toolchain        = VS2022

Active Platform          = /home/gen/i915dev/i915ovmfPkg/i915ovmf.dsc

FLASH_DEFINITION = None

Library instances of module [/home/gen/i915dev/i915ovmfPkg/i915ovmf.inf] [X64]:
        BaseLib : /home/gen/i915dev/edk2/MdePkg/Library/BaseLib/BaseLib.inf
        BaseMemoryLib : /home/gen/i915dev/edk2/MdePkg/Library/BaseMemoryLib/BaseMemoryLib.inf
        DebugLib : /home/gen/i915dev/edk2/MdePkg/Library/BaseDebugLibSerialPort/BaseDebugLibSerialPort.inf
        DevicePathLib : /home/gen/i915dev/edk2/MdePkg/Library/UefiDevicePathLib/UefiDevicePathLibOptionalDevicePathProtocol.inf
        FrameBufferBltLib : /home/gen/i915dev/edk2/MdeModulePkg/Library/FrameBufferBltLib/FrameBufferBltLib.inf
        MemoryAllocationLib : /home/gen/i915dev/edk2/MdePkg/Library/UefiMemoryAllocationLib/UefiMemoryAllocationLib.inf
        PcdLib : /home/gen/i915dev/edk2/MdePkg/Library/BasePcdLibNull/BasePcdLibNull.inf
        PciLib : /home/gen/i915dev/edk2/OvmfPkg/Library/DxePciLibI440FxQ35/DxePciLibI440FxQ35.inf
        UefiBootServicesTableLib : /home/gen/i915dev/edk2/MdePkg/Library/UefiBootServicesTableLib/UefiBootServicesTableLib.inf
        UefiDriverEntryPoint : /home/gen/i915dev/edk2/MdePkg/Library/UefiDriverEntryPoint/UefiDriverEntryPoint.inf
        UefiLib : /home/gen/i915dev/edk2/MdePkg/Library/UefiLib/UefiLib.inf
        MemEncryptSevLib : /home/gen/i915dev/edk2/OvmfPkg/Library/BaseMemEncryptSevLib/BaseMemEncryptSevLib.inf
        CacheMaintenanceLib : /home/gen/i915dev/edk2/MdePkg/Library/BaseCacheMaintenanceLib/BaseCacheMaintenanceLib.inf
        CpuLib : /home/gen/i915dev/edk2/MdePkg/Library/BaseCpuLib/BaseCpuLib.inf
        PrintLib : /home/gen/i915dev/edk2/MdePkg/Library/BasePrintLib/BasePrintLib.inf
        UefiRuntimeServicesTableLib : /home/gen/i915dev/edk2/MdePkg/Library/UefiRuntimeServicesTableLib/UefiRuntimeServicesTableLib.inf
        PciCf8Lib : /home/gen/i915dev/edk2/MdePkg/Library/BasePciCf8Lib/BasePciCf8Lib.inf
        PciExpressLib : /home/gen/i915dev/edk2/MdePkg/Library/BasePciExpressLib/BasePciExpressLib.inf
        IoLib : /home/gen/i915dev/edk2/MdePkg/Library/BaseIoLibIntrinsic/BaseIoLibIntrinsic.inf
        SerialPortLib : /home/gen/i915dev/edk2/PcAtChipsetPkg/Library/SerialIoLib/SerialIoLib.inf
        DebugPrintErrorLevelLib : /home/gen/i915dev/edk2/MdePkg/Library/BaseDebugPrintErrorLevelLib/BaseDebugPrintErrorLevelLib.inf
 

- Failed -
Build end time: 13:15:57, Mar.15 2025
Build total time: 00:00:01
```

I think I cannot use the latest edk2 version because for the i915ovmf package I need stable/202011 for `BaseMemEncryptSevLib.inf` instead of `DxeMemEncryptSevLib.inf`

If I re-checkout edk2 master branch, and update all submodules, I end up (again) with:
```
....
Ran 303 tests in 3.352s

OK
make[1]: Leaving directory '/home/gen/i915dev/edk2/BaseTools/Tests'
make: Leaving directory '/home/gen/i915dev/edk2/BaseTools'
Build environment: Linux-6.8.0-54-generic-x86_64-with-glibc2.39
Build start time: 13:28:24, Mar.15 2025

WORKSPACE        = /home/gen/i915dev
PACKAGES_PATH    = /home/gen/i915dev/edk2:/home/gen/i915dev/edk2-platforms
EDK_TOOLS_PATH   = /home/gen/i915dev/edk2/BaseTools
CONF_PATH        = /home/gen/i915dev/edk2/Conf
PYTHON_COMMAND   = python3

Processing meta-data .
Architecture(s)  = X64
Build target     = DEBUG
Toolchain        = VS2022

Active Platform          = /home/gen/i915dev/i915ovmfPkg/i915ovmf.dsc

FLASH_DEFINITION = None

build.py...
/home/gen/i915dev/i915ovmfPkg/i915ovmf.dsc(133): error 000E: File/directory not found in workspace
        OvmfPkg/Library/BaseMemEncryptSevLib/BaseMemEncryptSevLib.inf is not found in packages path:
        /home/gen/i915dev/edk2
        /home/gen/i915dev/edk2-platforms
 

- Failed -
Build end time: 13:28:24, Mar.15 2025
Build total time: 00:00:00
```

lgao4, 2025-04-10:
Liming will check what issue here.