tianocore/edk2 #10838: [Code First]: Clarify the Energy Performance Priority Register

souvikkc, 2025-03-10:
### Code First Item Overview

[CodeFirst_ACPI_EPP-Clarifications.docx](https://github.com/user-attachments/files/19168457/CodeFirst_ACPI_EPP-Clarifications.docx)

### What specification(s) are directly related?

ACPI

tapanansel, 2025-03-10:
We are comfortable with the 8-bit register size but have some concerns regarding the newly added **8.4.6.1.8 Maximum Energy Performance Preference Value Register.** The issue arises because Windows is designed with the assumption that higher-revision CPPC implementations are backward compatible with older revision OS. In cases where hardware supports the new EPP max value register (e.g., for 4 bits) and the old OSPM scaling from 0-100%, there may be incorrect mappings to the hardware.

To address this problem, we recommend implementing a protocol for OSPM to communicate CPPCv5 support. This would allow the platform to revert to the existing EPP behavior on OSPM without CPPCv5 support. If implementing this protocol is too complex, we suggest dropping this new register.

souvikkc, 2025-03-19:
[CodeFirst_ACPI_EPP-Clarifications-v2.docx](https://github.com/user-attachments/files/19344089/CodeFirst_ACPI_EPP-Clarifications-v2.docx)

souvikkc, 2025-03-24:
[CodeFirst_ACPI_EPP-Clarifications-v3.docx](https://github.com/user-attachments/files/19430021/CodeFirst_ACPI_EPP-Clarifications-v3.docx)

leiflindholm, 2025-04-03:
A couple of comments/questions:
1) What codebases are expected to need updates because of this change?
2) ACPI is now maintained in .rst format on github: https://github.com/UEFI/ACPI_spec/ (although you need to request access), could the spec update portion be represented as a patch against that rather than a word doc?

closed 2025-05-20 by souvikkc