tianocore/edk2 #10839: [Bug]: Build fails due to insufficient FV size in edk2-stable202502

r1chard-lyu, 2025-03-11:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

OvmfPkg

### Which targets are impacted by this bug?

DEBUG

### Current Behavior

I built OVMF with the following options
```shell
build -D TPM2_ENABLE -D TPM2_CONFIG_ENABLE -D NETWORK_IP6_ENABLE -D NETWORK_HTTP_BOOT_ENABLE -D BUILD_SHELL=FALSE -a X64 -b DEBUG -t GCC5 -p OvmfPkg/OvmfPkgX64.dsc -D FD_SIZE_2MB -D SECURE_BOOT_ENABLE
```

However, I encountered an insufficient FV size issue. This issue did not occur in edk2-stable202411, but it appears in edk2-stable202502.
```shell
GenFv: ERROR 3000: Invalid
  the required fv image size 0x1affc8 exceeds the set fv image size 0x1ac000
```

Is there any way to resolve this?

### Expected Behavior

Build OVMF successfully

```shell
FV Space Information
SECFV [29%Full] 212992 (0x34000) total, 63536 (0xf830) used, 149456 (0x247d0) free
PEIFV [47%Full] 917504 (0xe0000) total, 431672 (0x69638) used, 485832 (0x769c8) free
DXEFV [41%Full] 15204352 (0xe80000) total, 6370728 (0x6135a8) used, 8833624 (0x86ca58) free
FVMAIN_COMPACT [98%Full] 1753088 (0x1ac000) total, 1729808 (0x1a6510) used, 23280 (0x5af0) free

- Done -
Build end time: 14:18:06, Mar.11 2025
Build total time: 00:01:00

```

### Steps To Reproduce

```shell
build -D TPM2_ENABLE -D TPM2_CONFIG_ENABLE -D NETWORK_IP6_ENABLE -D NETWORK_HTTP_BOOT_ENABLE -D BUILD_SHELL=FALSE -a X64 -b DEBUG -t GCC5 -p OvmfPkg/OvmfPkgX64.dsc -D FD_SIZE_2MB -D SECURE_BOOT_ENABLE
```

### Build Environment

```markdown
- OS(s): OpenSUSE Leap15.6
- Tool Chain(s): GCC5
```

### Version Information

```text
Tag: edk2-stable202502
```

### Urgency

Medium

### Are you going to fix this?

I will fix it

### Do you need maintainer feedback?

Maintainer feedback requested

### Anything else?

I can work on fixing this issue, but I need some suggestions from upstream.

lgao4, 2025-03-13:
[r1chard-lyu](https://github.com/r1chard-lyu): please work with Ard.

lgao4, 2025-03-13:
@ardbiesheuvel , please give some suggestion for this issue.

closed 2025-03-13 by ardbiesheuvel