tianocore/edk2 #10848: [Code First]: Fix Op0 in ARM Processor Context Information CPER.

jmarinho, 2025-03-12:
### Code First Item Overview

Fix CPER Arm Misc Context System Register definition. The "Op0" field was incorrectly named "O0" and spanned a single bit.

[ECR_MISC_Context_fix.docx](https://github.com/user-attachments/files/19206848/ECR_MISC_Context_fix.docx)

### What specification(s) are directly related?

UEFI

leiflindholm, 2025-04-03:
The UEFI spec is now maintained in .rst format in https://github.com/UEFI/UEFI_spec (access request needed) - can the change be described as a diff against that instead of in a word doc?

leiflindholm, 2025-04-03:
As for the proposed change:
The existing text in UEFI 2.11 is ... ambitious and incorrect in more ways than this issue is resolving.

So, we're intending to use parts from the instruction encodings used to read certain system registers, regardless of referring to we're in AArch32 or AArch64 state. This means for AArch64 that we're looking at MRS instruction encoding, and for AArch32 we're looking at MRC instruction encoding. (The Description text says it is about MRS.)

Then we have the o0 1-bit field this change is about.
That field
1) Is indeed called o0, and is 1 bit, in the AArch64 MRS instruction encoding.
2) Does not exist in the AArch32 MRC instruction encoding.

*However*, the pseudocode of the MRS instruction (ARM DDI 0487 L.a) *expands* the o0 encoding bit into a sys_op0 field by oring it with b'10.

I don't know what to about the AArch32 thing, but neither the existing version nor the proposed update aligns with reality.

For AArch64 - is the implication that records may be generated based on instructions other than MRS, which end up with different encoding for the sys_op0 field?
That would appear to be the case: the MRRS/SYS/SYSL/SYSP instructions (not all of which actually have an o0 in their encoding, but has sys_op0 hardwired to 1 in their pseudocode).
Which opens up a further can of worms in that some of these return 128-bit values, whereas the Type 8 structure holds only 8 bits of Value.

I think someone in Arm needs to go off and untangle this. As it stands, I feel the change as proposed is incorrect.

jmarinho, 2025-04-03:
For system registers accessible by MRS, Op0 is defined but always takes values in {10b, 11b}.

The move o0 --> Op0 is not technically wrong.
The field Op0 is used in the System Register definition in the Arm ARM.
See this example for HDBSSBR_EL2:

[image]

Agree that the AArch32 name of the register is not fully captured in table 8.
Could we clarify that table 8 applies to AArch64 only?

leiflindholm, 2025-04-03:
> For system registers accessible by MRS, Op0 is defined but always takes values in {10b, 11b}.
> The move o0 --> Op0 is not technically wrong. The field Op0 is used in the System Register definition in the Arm ARM. See this example for HDBSSBR_EL2:

OK, I see what you're saying. But the result is still technically wrong. If we describe this as viewed through MRS instruction encoding, it should match the MRS instruction encoding. MRS has o0. It does not have Op0.

[image]

If the text is about something else, it should say that.
(And it is, it is about how to unambiguously reference a specific register in System register encoding space.) The text above the table is actually pretty clear that it is referring to system registers rather than MRS instructions as such. But the table confuses things by trying to be helpful and describe it through an MRS encoding view.

Someone really should try to figure out a credible way to retcon this corner of the Arm ARM.

> Agree that the AArch32 name of the register is not fully captured in table 8. Could we clarify that table 8 applies to AArch64 only?

I think it's a call for Arm to make whether an alternative table is needed for AArch32, we can drop AArch32 support completely for this table, or whether we do something where we say "op0 exists only for AArch64 and must be WAZ for AArch32", but I think it's clear the current description has never been correct.

I think we also need to consider what we need to do for FEAT_SYSREG128, since the current Type 8 has a fixed Value size.

jmarinho, 2025-04-10:
Thank you @leiflindholm.
We'll discuss the alternatives internally to decide how to progress.
Please bear with us.

jmarinho, 2025-04-22:
Correct patch for mantis 2508, applied on top of 2.11 release:
[0001-CPER-fix-MRS-encoding-Op0.patch](https://github.com/user-attachments/files/19849569/0001-CPER-fix-MRS-encoding-Op0.patch)

edhay, 2026-11-05:
The PR is raised as below:
https://github.com/UEFI/UEFI_spec/pull/155