tianocore/edk2 #10872: [Bug]: Remove Visual Studio ARM 32-bit Host Support

makubacki, 2025-03-18:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [ ] Firmware
- [x] Tool
- [ ] Unit Test

### What packages are impacted?

BaseTools, Build or CI Code

### Which targets are impacted by this bug?

DEBUG, NO-TARGET, NOOPT, RELEASE

### Current Behavior

Visual Studio 2022 is not supported on ARM32 operating systems per https://learn.microsoft.com/visualstudio/releases/2022/system-requirements#supported-operating-systems and https://learn.microsoft.com/visualstudio/install/visual-studio-on-arm-devices?view=vs-2022#native-developers.

Visual Studio 2019 does not support ARM operating systems either as described in https://learn.microsoft.com/en-us/visualstudio/releases/2019/system-requirements#visual-studio-2019-system-requirements.

Therefore, `ARM` architecture support for Visual Studio can be dropped from files such as the BaseTools makefile https://github.com/tianocore/edk2/blob/master/BaseTools/Source/C/Makefiles/ms.common#L60.

### Expected Behavior

Support in build files is not present for unsupported configurations.

### Steps To Reproduce

Review Visual Studio system requirements and build files checked into the repo.

### Build Environment

```markdown
- OS(s): Windows ARM32
- Tool Chain(s): VSxx
```

### Version Information

```text
Commit: 9bceb16000056f31119c79014788bc99d5cfdc3d (current `master` branch head commit)
```

### Urgency

Low

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

No maintainer feedback needed

Damien-Chen, 2025-03-22:
Hi @makubacki , could I take this issue ?

kristibektashi, 2025-10-04:
I assume this can be closed as per https://github.com/tianocore/edk2/pull/11494