tianocore/edk2 #10876: [Bug]: PRM package build fails with GCC compiler for X64

abdattar, 2025-03-19:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [ ] Firmware
- [x] Tool
- [ ] Unit Test

### What packages are impacted?

PrmPkg

### Current Behavior

1) Applied the recommended GCC flags mentioned in readme of PrmPkg
abdattar@BLR-L-ABDATTAR:~/tianocore/edk2$ git show
commit a762df2503f064f7f7a7c45538fab75c12cdcf2f (HEAD -> PrmGccd)
Author: Abdul Lateef Attar <abdattar@amd.com>
Date:   Tue Mar 18 08:51:14 2025 +0000

    PrmPkg: Correct the flags for GCC compiler

diff --git a/PrmPkg/Samples/PrmSampleContextBufferModule/PrmSampleContextBufferModule.inf b/PrmPkg/Samples/PrmSampleContextBufferModule/PrmSampleContextBufferModule.inf
index ff72a23366..8cfa358f82 100644
--- a/PrmPkg/Samples/PrmSampleContextBufferModule/PrmSampleContextBufferModule.inf
+++ b/PrmPkg/Samples/PrmSampleContextBufferModule/PrmSampleContextBufferModule.inf
@@ -44,5 +44,5 @@
   MSFT:*_*_*_DLINK_FLAGS  = /DLL /SUBSYSTEM:CONSOLE /VERSION:1.0
   MSFT:*_*_*_GENFW_FLAGS = --keepoptionalheader

-  GCC:*_*_AARCH64_GENFW_FLAGS = --keepoptionalheader --prm
-  GCC:*_*_AARCH64_DLINK_FLAGS = -Wl,--no-gc-sections -Wl,--require-defined=PrmModuleExportDescriptor -Wl,--require-defined=CheckStaticDataBufferPrmHandler
+  GCC:*_*_*_GENFW_FLAGS = --keepoptionalheader --prm^M
+  GCC:*_*_*_DLINK_FLAGS = -Wl,--no-gc-sections -Wl,--require-defined=PrmModuleExportDescriptor -Wl,--require-defined=CheckStaticDataBufferPrmHandler^M
abdattar@BLR-L-ABDATTAR:~/tianocore/edk2$

2) compilation fails with the above option
$ build -t GCC -a X64 -p PrmPkg/PrmPkg.dsc
cp -p -f /home/abdattar/tianocore/edk2/Build/Prm/DEBUG_GCC/X64/PrmPkg/Samples/PrmSampleHardwareAccessModule/PrmSampleHardwareAccessModule/OUTPUT/PrmSampleHardwareAccessModule.efi /home/abdattar/tianocore/edk2/Build/Prm/DEBUG_GCC/X64/PrmSampleHardwareAccessModule.efi
cp -p -f /home/abdattar/tianocore/edk2/Build/Prm/DEBUG_GCC/X64/PrmPkg/Samples/PrmSampleHardwareAccessModule/PrmSampleHardwareAccessModule/DEBUG/*.map /home/abdattar/tianocore/edk2/Build/Prm/DEBUG_GCC/X64/PrmPkg/Samples/PrmSampleHardwareAccessModule/PrmSampleHardwareAccessModule/OUTPUT
cp -p -f /home/abdattar/tianocore/edk2/Build/Prm/DEBUG_GCC/X64/PrmPkg/Samples/PrmSampleAcpiParameterBufferModule/PrmSampleAcpiParameterBufferModule/DEBUG/*.map /home/abdattar/tianocore/edk2/Build/Prm/DEBUG_GCC/X64/PrmPkg/Samples/PrmSampleAcpiParameterBufferModule/PrmSampleAcpiParameterBufferModule/OUTPUT
cp -p -f /home/abdattar/tianocore/edk2/Build/Prm/DEBUG_GCC/X64/PrmPkg/Samples/PrmSampleContextBufferModule/PrmSampleContextBufferModule/DEBUG/PrmSampleContextBufferModule.debug /home/abdattar/tianocore/edk2/Build/Prm/DEBUG_GCC/X64/PrmSampleContextBufferModule.debug
"GenFw" -e DXE_RUNTIME_DRIVER -o /home/abdattar/tianocore/edk2/Build/Prm/DEBUG_GCC/X64/PrmPkg/Samples/PrmSampleContextBufferModule/PrmSampleContextBufferModule/OUTPUT/PrmSampleContextBufferModule.efi /home/abdattar/tianocore/edk2/Build/Prm/DEBUG_GCC/X64/PrmPkg/Samples/PrmSampleContextBufferModule/PrmSampleContextBufferModule/DEBUG/PrmSampleContextBufferModule.dll --keepoptionalheader --prm
Segmentation fault (core dumped)
make: *** [GNUmakefile:349: /home/abdattar/tianocore/edk2/Build/Prm/DEBUG_GCC/X64/PrmPkg/Samples/PrmSampleContextBufferModule/PrmSampleContextBufferModule/OUTPUT/PrmSampleContextBufferModule.efi] Error 139

build.py...
 : error 7000: Failed to execute command
        make tbuild [/home/abdattar/tianocore/edk2/Build/Prm/DEBUG_GCC/X64/PrmPkg/Samples/PrmSampleContextBufferModule/PrmSampleContextBufferModule]

build.py...
 : error F002: Failed to build module
        /home/abdattar/tianocore/edk2/PrmPkg/Samples/PrmSampleContextBufferModule/PrmSampleContextBufferModule.inf [X64, GCC, DEBUG]

- Failed -
Build end time: 07:23:38, Mar.19 2025
Build total time: 00:00:06

abdattar@BLR-L-ABDATTAR:~/tianocore/edk2$

### Expected Behavior

Build should be passed.

### Steps To Reproduce

1) on linux build environment (in this case i am using WSL ubuntu 24.04.01)
2) $ gcc --version
gcc (Ubuntu 12.3.0-17ubuntu1) 12.3.0
3) $ ld --version
GNU ld (GNU Binutils for Ubuntu) 2.42

4) make the following changes according to readme of PrmPkg
abdattar@BLR-L-ABDATTAR:~/tianocore/edk2$ git show
commit a762df2503f064f7f7a7c45538fab75c12cdcf2f (HEAD -> PrmGccd)
Author: Abdul Lateef Attar <abdattar@amd.com>
Date:   Tue Mar 18 08:51:14 2025 +0000

    PrmPkg: Correct the flags for GCC compiler

diff --git a/PrmPkg/Samples/PrmSampleContextBufferModule/PrmSampleContextBufferModule.inf b/PrmPkg/Samples/PrmSampleContextBufferModule/PrmSampleContextBufferModule.inf
index ff72a23366..8cfa358f82 100644
--- a/PrmPkg/Samples/PrmSampleContextBufferModule/PrmSampleContextBufferModule.inf
+++ b/PrmPkg/Samples/PrmSampleContextBufferModule/PrmSampleContextBufferModule.inf
@@ -44,5 +44,5 @@
   MSFT:*_*_*_DLINK_FLAGS  = /DLL /SUBSYSTEM:CONSOLE /VERSION:1.0
   MSFT:*_*_*_GENFW_FLAGS = --keepoptionalheader

-  GCC:*_*_AARCH64_GENFW_FLAGS = --keepoptionalheader --prm
-  GCC:*_*_AARCH64_DLINK_FLAGS = -Wl,--no-gc-sections -Wl,--require-defined=PrmModuleExportDescriptor -Wl,--require-defined=CheckStaticDataBufferPrmHandler
+  GCC:*_*_*_GENFW_FLAGS = --keepoptionalheader --prm^M
+  GCC:*_*_*_DLINK_FLAGS = -Wl,--no-gc-sections -Wl,--require-defined=PrmModuleExportDescriptor -Wl,--require-defined=CheckStaticDataBufferPrmHandler^M
abdattar@BLR-L-ABDATTAR:~/tianocore/edk2$

5) build the Prm package
$ build -t GCC -a X64 -p PrmPkg/PrmPkg.dsc

### Build Environment

```markdown
- OS(s):Ubuntu 24.04.1
- Tool Chain(s): GCC
```

### Version Information

```text
https://github.com/tianocore/edk2/pull/10875
```

### Urgency

High

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

No maintainer feedback needed

makubacki, 2025-03-19:
@abdattar, you selected "Someone else needs to fix it" in this issue but you also created the PR to address the issue. My understanding is that you are planning to fix it in that PR and this issue can be assigned to you. Correct?

abdattar, 2025-03-20:
Hi @makubacki,
I wanted to submit the PR (https://github.com/tianocore/edk2/pull/10875),
but the this compilation failure/CI failure which is in BaseTools.
Hence need someone who knows the tool to fix the issue.
here is the RCA:
A segmentation fault is occurring in the basetool of EDK2.edk2/BaseTools/Source/C/GenFw/Elf64Convert.c
      //
      // First Get PrmModuleExportDescriptor
      //
      for (SymIndex = 0; SymIndex < SymNum; SymIndex++) {
        Sym = (Elf_Sym *)(Symtab + SymIndex * shdr->sh_entsize);
        SymName = GetSymName(Sym);
        if (SymName == NULL) {
            continue;
        }

        if (strcmp((CHAR8*)SymName, PRM_MODULE_EXPORT_DESCRIPTOR_NAME) == 0) { **/// symbol table do not have this symbol despite providing “--require-defined=PrmModuleExportDescriptor” option**

          //
          // Find PrmHandler Number and Name
          //
          FindPrmHandler(Sym->st_value);

          strcpy(mExportSymName[mExportSymNum], (CHAR8*)SymName);
          mExportRVA[mExportSymNum] = (UINT32)(Sym->st_value);
          mExportSize += 2 * EFI_IMAGE_EXPORT_ADDR_SIZE + EFI_IMAGE_EXPORT_ORDINAL_SIZE + strlen((CHAR8 *)SymName) + 1;
          mExportSymNum ++;  /// control didn’t reach here because its not able to find the symbol
          break;
        }
      }

      //
      // Second Get PrmHandler
      //
      for (SymIndex = 0; SymIndex < SymNum; SymIndex++) {
        UINT32   ExpIndex;
        Sym = (Elf_Sym *)(Symtab + SymIndex * shdr->sh_entsize);
        SymName = GetSymName(Sym);
        if (SymName == NULL) {
            continue;
        }

        for (ExpIndex = 0; ExpIndex < (mExportSymNum -1); ExpIndex++) {
          if (strcmp((CHAR8*)SymName, mExportSymName[ExpIndex]) != 0) { **// Segmentation fault happening here, because mExportSymNum is zero**
            **/// trying to access array as 0 -1 = -1 (round-off to MAX_UINT32 value**
            continue;
          }

makubacki, 2025-03-20:
The BaseTools change to support PRM with GCC was added by @huanglx729 in https://github.com/tianocore/edk2/commit/414cd2a4d536b444acee3fcdbe33bd0c8ef9819b. If they are still working on the project and able to investigate, we can assign it to them. Otherwise, if an existing BaseTools maintainer is interested in doing so they can take it. I can also look into it, but it will at least be a few days before I have time.

abdattar, 2025-03-26:
Looks like during objcopy symbols are getting stripped, by adding below options it solves the build failure.
GCC:*_*_X64_OBJCOPY_STRIPFLAG = --keep-symbol=<symbol1> --keep-symbol=<symbol2>

lgao4, 2025-04-10:
this issue has been fixed in https://github.com/tianocore/edk2/pull/10875

closed 2025-04-10 by lgao4