tianocore/edk2 #10881: [Bug]: AAVMF direct kernel boot broken

kukrimate, 2025-03-20:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [ ] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

ArmVirtPkg

### Which targets are impacted by this bug?

RELEASE

### Current Behavior

Direct kernel boot (with the qemu -kernel argument) seems to be broken on recent version.

As in AAVMF just ignored the -kernel argument.

### Expected Behavior

AAVMF boots the provided kernel.

### Steps To Reproduce

Boot recent AAVMF in qemu with -kernel argument.

### Build Environment

```markdown
- OS(s): Ubuntu 25.04
- Tool Chain(s): GCC 14
```

### Version Information

```text
Reverting the following commits fixes the problem:
"OvmfPkg/QemuKernelLoaderFsDxe: rework direct kernel boot filesystem"
"OvmfPkg/QemuKernelLoaderFsDxe: add support for named blobs"
"OvmfPkg/QemuKernelLoaderFsDxe: allow longer file names"
"OvmfPkg/QemuKernelLoaderFsDxe: drop bogus assert"
"OvmfPkg/QemuKernelLoaderFsDxe: accept absolute paths"
"OvmfPkg/QemuKernelLoaderFsDxe: don't quit when named blobs are present"
"OvmfPkg/QemuKernelLoaderFsDxe: root directory name should be """
"OvmfPkg/QemuKernelLoaderFsDxe: use SIZE_OF_EFI_FILE_INFO"
```

### Urgency

Medium

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

No maintainer feedback needed

dannf, 2025-03-23:
I can reproduce and, annoyingly, it works fine with a DEBUG build.

rhythm16, 2025-03-29:
I suspect I am hitting the same issue, however I can reproduce with DEBUG as well.
Here are logs:
https://pastebin.ubuntu.com/p/YYFYvhzPDr/
I bisected the history and found the breaking commit is: `459f5ffa24ae8574 OvmfPkg/QemuKernelLoaderFsDxe: rework direct kernel boot filesystem`.

I was trying to run this: https://linaro.atlassian.net/wiki/spaces/QEMU/pages/29051027459/Building+an+RME+stack+for+QEMU

dannf, 2025-04-08:
Would you be able to paste the exact command and kernel image you are using? I'm doing this:

```
qemu-system-aarch64 -cpu cortex-a57 -no-user-config -nodefaults -m 2048 -smp 1,sockets=1,cores=1,threads=1 -display none -serial stdio -machine virt -drive file=$(pwd)/debian/aavmf-install/AAVMF_CODE.secboot.fd,if=pflash,format=raw,unit=0,readonly=on -kernel vmlinux
```

Where `vmlinux` is the kernel from `debian-12.10.0-arm64-netinst.iso`

And it works in DEBUG mode:
```
PlatformBootManagerAfterConsole: not uninstalling EFI memory protocol
FindKernelBlob: BLOB Name is "cmdline"
FindKernelBlob: BLOB Name is "kernel"
QemuKernelStubFileOpen: file not found: "shim"
FindKernelBlob: BLOB Name is "cmdline"
FindKernelBlob: BLOB Name is "kernel"
QemuKernelStubFileOpen: file opened: "kernel"
QemuKernelBlobTypeToFileInfo: file info: "kernel"
QemuKernelBlobTypeToFileInfo: file info: "kernel"
QemuKernelStubFileRead: file read: "kernel", 61571584 bytes
[Security] 3rd party image[0] can be loaded after EndOfDxe: VenMedia(1428F772-B64A-441E-B8C3-9EBDD7F893C7)/kernel.
None of Tcg2Protocol/CcMeasurementProtocol is installed.
InstallProtocolInterface: 5B1B31A1-9562-11D2-8E3F-00A0C969723B BADFB0C0
Loading driver at 0x000B36C0000 EntryPoint=0x000B6035118
Loading driver at 0x000B36C0000 EntryPoint=0x000B6035118 
InstallProtocolInterface: BC62157E-3E33-4FEC-9920-2D3B36D750DF BADFBD98
ProtectUefiImageCommon - 0xBADFB0C0
  - 0x00000000B36C0000 - 0x0000000003BF0000
SetUefiImageMemoryAttributes - 0x00000000B36C0000 - 0x0000000000010000 (0x0000000000004008)
SetUefiImageMemoryAttributes - 0x00000000B36D0000 - 0x00000000029E0000 (0x0000000000020008)
SetUefiImageMemoryAttributes - 0x00000000B60B0000 - 0x0000000001200000 (0x0000000000004008)
FindKernelBlob: BLOB Name is "cmdline"
QemuKernelStubFileOpen: file opened: "cmdline"
QemuKernelBlobTypeToFileInfo: file info: "cmdline"
QemuKernelBlobTypeToFileInfo: file info: "cmdline"
FindKernelBlob: BLOB Name is "cmdline"
QemuKernelStubFileOpen: file opened: "cmdline"
QemuKernelStubFileRead: file read: "cmdline", 1 bytes
FindKernelBlob: BLOB Name is "cmdline"
FindKernelBlob: BLOB Name is "kernel"
QemuKernelStubFileOpen: file not found: "initrd"
[Bds]Stop Hotkey Service!
[Bds]UnregisterKeyNotify: 000C/0000 Success
[Bds]UnregisterKeyNotify: 0017/0000 Success
[Bds]UnregisterKeyNotify: 0000/000D Success
EFI stub: Booting Linux Kernel...
EFI stub: EFI_RNG_PROTOCOL unavailable
SetMemoryAttributes: BaseAddress == 0xAFA00000, Length == 0x29F0000, Attributes == 0x20000
SetMemoryAttributes: BaseAddress == 0xB23F0000, Length == 0x1200000, Attributes == 0x4000
EFI stub: Generating empty DTB
EFI stub: Exiting boot services...
SetUefiImageMemoryAttributes - 0x00000000BFD40000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000BFCF0000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000BFCA0000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000BFC50000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000BEC40000 - 0x0000000000100000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000BFC10000 - 0x0000000000030000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000BEAA0000 - 0x0000000000030000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000BEA60000 - 0x0000000000030000 (0x0000000000000008)
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd070]
[    0.000000] Linux version 6.14.0-11-generic (buildd@bos03-arm64-071) (aarch64-linux-gnu-gcc-14 (Ubuntu 14.2.0-19ubuntu1) 14.2.0, GNU ld (GNU Binutils for Ubuntu) 2.44) #11-Ubuntu SMP PREEMPT_DYNAMIC Mon Mar 17 12:19:09 UTC 2025 (Ubuntu 6.14.0-11.11-generic 6.14.0-rc7)
```

bdrung, 2025-04-08:
This bug caused the initramfs-tools autopkgtest to fail on arm64 and armhf. Example build failure logs:
* https://autopkgtest.ubuntu.com/results/autopkgtest-plucky/plucky/arm64/i/initramfs-tools/20250320_160021_9d652@/log.gz
* https://autopkgtest.ubuntu.com/results/autopkgtest-plucky/plucky/armhf/i/initramfs-tools/20250320_153427_a4152@/log.gz

You can see the arguments there:
> run-qemu: qemu-system-arm64 -machine virt,gic-version=max -cpu max,pauth-impdef=on -m 1G -drive file=/usr/share/AAVMF/AAVMF_CODE.fd,if=pflash,format=raw,read-only=on -drive file=/tmp/autopkgtest.SviTuI/autopkgtest_tmp/AAVMF_VARS.fd.HYqrce3e9X,if=pflash,format=raw -device virtio-rng-pci,rng=rng0 -object rng-random,filename=/dev/urandom,id=rng0 -nodefaults -no-reboot -kernel /boot/vmlinuz-6.14.0-11-generic -initrd /tmp/autopkgtest.SviTuI/autopkgtest_tmp/initramfs-test.DClam64v1R/initrd.img -nographic -chardev stdio,id=char0 -serial chardev:char0 -drive file=/tmp/autopkgtest.SviTuI/autopkgtest_tmp/initramfs-test.DClam64v1R/rootdisk.raw,if=virtio,media=disk,format=raw -device virtio-net-pci,netdev=lan0,mac=52:54:00:65:43:21 -netdev user,id=lan0,net=10.0.3.0/24,ipv6-net=fec7::/48,hostname=pizza,dnssearch=test,domainname=example.com,bootfile=/path/to/bootfile2 -device virtio-net-pci,netdev=lan1,mac=52:54:00:12:34:56 -netdev user,id=lan1,hostname=goulash,dnssearch=example,dnssearch=example.net,domainname=test,bootfile=/path/to/bootfile -append console=ttyS0,115200 ro root=/dev/vda panic=-1

The exact package version can be seen in the logs as well. The failure should probably happen on amd64 as well (the initramfs-tools autopkgtest on amd64 does not fail, because it does not use EFI).

rhythm16, 2025-04-08:
I'm basically following the steps provided in the link, which is a rather heavy setup
https://linaro.atlassian.net/wiki/spaces/QEMU/pages/29051027459/Building+an+RME+stack+for+QEMU
I extract the relevant parts here:

EDK2:
```
git clone https://github.com/tianocore/edk2.git
cd edk2/
git submodule update --init --recursive
source edksetup.sh
make -j -C BaseTools
export GCC5_AARCH64_PREFIX=aarch64-linux-gnu-
build -b DEBUG -a AARCH64 -t GCC5 -p ArmVirtPkg/ArmVirtQemuKernel.dsc
```

The QEMU (v10.0.0-rc1) command is
```
qemu-system-aarch64 -M virt,virtualization=on,secure=on,gic-version=3
        -M acpi=off -cpu max,x-rme=on -m 8G -smp 8
        -nographic
        -bios trusted-firmware-a/flash.bin
        -kernel linux-cca/arch/arm64/boot/Image
        -drive format=raw,if=none,file=buildroot/output/images/rootfs.ext4,id=hd0
        -device virtio-blk-pci,drive=hd0
        # The following parameters allow to use separate consoles for Firmware (port 54320),
        # Secure payload (54321), host (54322) and guest (54323).
        -nodefaults
        -serial tcp:localhost:54320
        -serial tcp:localhost:54321
        -chardev socket,mux=on,id=hvc0,port=54322,host=localhost
        -device virtio-serial-device
        -device virtconsole,chardev=hvc0
        -chardev socket,mux=on,id=hvc1,port=54323,host=localhost
        -device virtio-serial-device
        -device virtconsole,chardev=hvc1
        -append "root=/dev/vda console=hvc0 nokaslr"
        -device virtio-net-pci,netdev=net0 -netdev user,id=net0
        # This shares the current directory with the host, providing the files needed
        # to launch the guest.
        -device virtio-9p-device,fsdev=shr0,mount_tag=shr0
        -fsdev local,security_model=none,path=.,id=shr0
```
QEMU will connect to four TCP ports for the different consoles. Create the servers manually with `socat -,rawer TCP-LISTEN:5432x`  (x = 0, 1, 2, 3) 

`trusted-firmware-a/flash.bin` is created by: (I leave out RMM, which is not needed to reproduce i believe)
```
git clone https://git.codelinaro.org/linaro/dcap/tf-a/trusted-firmware-a --branch cca/v4
cd trusted-firmware-a/
# Embed the RMM image and edk2 into the Final Image Package (FIP)
make -j CROSS_COMPILE=aarch64-linux-gnu- PLAT=qemu ENABLE_RME=1 DEBUG=1 LOG_LEVEL=40 \
    QEMU_USE_GIC_DRIVER=QEMU_GICV3 \
    BL33=../edk2/Build/ArmVirtQemuKernel-AARCH64/DEBUG_GCC5/FV/QEMU_EFI.fd all fip
# Pack whole image into flash.bin
dd if=build/qemu/debug/bl1.bin of=flash.bin
dd if=build/qemu/debug/fip.bin of=flash.bin seek=64 bs=4096
```
Linux kernel:
```
git clone https://gitlab.arm.com/linux-arm/linux-cca --branch cca-host/v7
cd linux-cca/
make CROSS_COMPILE=aarch64-linux-gnu- ARCH=arm64 defconfig
# Enable the configfs-tsm driver that provides the attestation interface
scripts/config -e VIRT_DRIVERS -e ARM_CCA_GUEST -e CONFIG_HZ_100 \
 -d CONFIG_HZ_250 -e CONFIG_MACVLAN -e CONFIG_MACVTAP
make CROSS_COMPILE=aarch64-linux-gnu- ARCH=arm64 -j8 Image
```

kraxel, 2025-04-10:
Reproduces on aa64, RELEASE builds only.  Works for me on x64, both DEBUG + RELEASE builds.

kraxel, 2025-04-15:
Hmm.  Setting `gEfiMdePkgTokenSpaceGuid.PcdDebugPropertyMask` to 0x25 globally (in `ArmVirtPkg/ArmVirt.dsc.inc`) makes things work.  Setting this for OvmfPkg/QemuKernelLoaderFsDxe/QemuKernelLoaderFsDxe.inf only (via override) does not work.  /me looks puzzled.

kraxel, 2025-04-16:
https://github.com/tianocore/edk2/pull/10971