tianocore/edk2 #10892: [Bug]: SEV and SEV-ES VM boot is broken due to faulty check in AmdSvsmIsSvsmPresent()

jxu023, 2025-03-25:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

OvmfPkg

### Current Behavior

https://github.com/tianocore/edk2/pull/6527 merged in e-VTPM SVSM support. All previous uses of AmdSvsmLib happened to occur within sections of code that first check if SNP is enabled, until that PR which has DTpm code directly check for SVSM. AmdSvsmIsSvsmPresent() then checks if SNP is enabled by seeing if the PCD value for OvmfSnpSecretsBase is NULL. However, the Ovmf Fdf file defines a non-null value for secrets base which is then overwritten later in case of Snp.

This can result in false positive determinations of being on Svsm depending on the unitialized memory at OvmfSnpSecretsBase (address defined in the Ovmf fdf). This breaks SEV and SEV-ES VM boot on our internal edk2 codebase as the DTpm SVSM shim code will continue to try making SVSM calls despite not being on SVSM.

This can be fixed a few different ways. We could fix the SNP enabled check in AmdSvsmIsSvsmPresent(), or only use AmdSvsmLib functions in conjunction with a separate SNP enabled check.

### Expected Behavior

I expect AmdSvsmIsSvsmPresent() to correctly determine if SNP is enabled.

### Steps To Reproduce

Build OVMF at a version including https://github.com/tianocore/edk2/pull/6527 and attempt to run a SEV or SEV-ES VM using it.

### Build Environment

```markdown
Our internal build environment uses Bazel and clang to build edk2, I haven't replicated the upstream build environment. But i expect it would reproduce readily there.
```

### Version Information

```text
Any version including https://github.com/tianocore/edk2/pull/6527 should repro the problem.
```

### Urgency

High

### Are you going to fix this?

I will fix it

### Do you need maintainer feedback?

Maintainer feedback requested

### Anything else?

@tlendacky @osteffenrh  Hi Tom, Oliver, could you provide a suggestion?

We have a few options:
- Checking if OvmfSnpSecretsPage addr has a well-formed Hob could be "unlucky" if the unitialized mem looks like a HoB though I'm unfamiliar if there's hob validation we could use anyway.
- Or only using AmdSvsmLib API with a separate SNP Enabled check, but future users could make the mistake again.
    - I've written up a quick fix which we can merge to fix the boot breakage faster.
- Adding a CPUID and then MSR sev status check is do-able, but it may be desirable to shift the dependency ordering since since MemEncryptSevLib (containing MemEncryptSnpIsEnabled()) currently depends on AmdSvsmLib.
    - This option seems "most" correct but I'm unsure about how to do it since currently we have multiple implemented SNP enabled checks located in CCExitLib, OvmfPkg/Sec/AmdSev.c, and MemEncryptSevLib.

osteffenrh, 2025-03-25:
I am OK with the quick fix while we can discuss what the best solution is.

osteffenrh, 2025-03-25:
I can reproduce the issue and can confirm that the fix works.

lgao4, 2025-04-10:
@jxu023 , please provide your fix.

osteffenrh, 2025-04-10:
The approved fix currently under discussion is https://github.com/tianocore/edk2/pull/10891

closed 2025-04-17 by jxu023